python-microscopy/python-microscopy #111: difficulty opening large ome.tif series when each frame is saved separately

barentine, 2019-12-09:
Tried to open a 1000 frame series, UserWarnings from failed to read frames started on frame 506
```[1000, 2, 1, 512, 512]
Detected OME TIFF
sh = {'T': 1000, 'C': 2, 'Z': 1, 'Y': 512, 'X': 512}
[512, 512, 1000, 2]
[512, 512, 2000, 1]
/home/smeagol/code/python-microscopy/PYME/DSView/dsviewer.py:134: wxPyDeprecationWarning: Call to deprecated item. Use Append instead.
  tmp_menu.AppendMenu(-1, 'Save &Results', self.save_menu)
Traceback (most recent call last):
  File "/home/smeagol/anaconda2/envs/py36/lib/python3.6/site-packages/wx/core.py", line 3259, in <lambda>
  File "/home/smeagol/code/python-microscopy/PYME/DSView/dsviewer.py", line 433, in LoadData
  File "/home/smeagol/code/python-microscopy/PYME/DSView/dsviewer.py", line 161, in __init__
  File "/home/smeagol/code/python-microscopy/PYME/DSView/modules/__init__.py", line 94, in loadMode
  File "/home/smeagol/code/python-microscopy/PYME/DSView/modules/__init__.py", line 69, in loadModule
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 674, in exec_module
  File "<frozen importlib._bootstrap_external>", line 780, in get_code
  File "<frozen importlib._bootstrap_external>", line 832, in get_data
OSError: [Errno 24] Too many open files: '/home/smeagol/code/python-microscopy/PYME/DSView/modules/filtering.py'
```

Surely there're some workarounds with changing system settings, or converting the series to a single file, but wanted to point out it is at least not as easy as it ideally would be.

David-Baddeley, 2019-12-09:
This is super weird, based on the traceback. What platform and python version?

barentine, 2019-12-09:
Ubuntu 16.04, python 3.6.8.

I ended up opening the series in fiji and saving as a single tif.

David-Baddeley, 2019-12-09:
Just to clarify ...

- you have a folder of 2D ome.tif images
- you have created a metadata file (e.g. `ABC.md`) with a  "SeriesPattern" key
- you then try and load the series by opening `dhview ABC.md`

There's quite a lot in your traceback that doesn't match this, not least that it appears that what you'd expect to be a 2D slice is in fact a 512x512x1x1000x2 volume.

closed 2021-03-06 by barentine