python-microscopy/python-microscopy #126: dependency on python dispatcher package

carandraug, 2020-01-09:
Trying to make PYMEAcquire I was having trouble finding the required dispatch module. On PyPI, there is the [dispatch](https://pypi.org/project/dispatch/) and [python-dispatch](https://pypi.org/project/python-dispatch/) but it's not those. Seems to require [dispatcher](https://pypi.org/project/dispatcher/) which is extracted from django but it's abandoned and does not work in Python 3.

Any chance to use `from django import dispatch`?

barentine, 2020-01-10:
@David-Baddeley , @carandraug and I were having a look and talk about this today while testing his python microscope adaptor. Would you be open to a PR changing the dispatch imports to django, dropping dispatch from conda-recipes, etc.? Not an issue most users will run into, but non-conda users trying to pull dispatcher 1.0 from pypy get a non py3 compatible version, and if we can drop the whole thing and use a dependency we already have it'd be nice from a maintenance perspective. We mentioned this briefly in the issue #69, but I don't think we resolved it.

David-Baddeley, 2020-01-10:
Putting django on the dependency list is a hard no due to the problems it's
caused in the past.

It's not the dispatcher module, but rather the dispatch module (which is
py3 compatible). Look at the pyme conda recipes repo for the github repo
(or the azure-pipeline script which pip-installs it from a github url).

A few options (not mutually exclusive) to make this easier going forward.

1) write some docs for a pip based install
2) pull the dispatch repo into PYME (undesirable) and distribute like our
patched versions of e.g. tiffile
3) clone (or git-link) the dispatch repo into the python-microscopy
organisation to make it easier to find
4) have a working pip requirements.txt

barentine, 2020-01-14:
Hi David, sorry for the confusion but the project on pypy is called dispatcher, though the module is in fact called dispatch. From [the github page we build dispatch from](https://github.com/olivierverdier/dispatch):

> This python library is extracted from django's dispatch module. This very documentation is also adapted from the django documentation on signals.

It's the same thing, and if we to replace it with django we can. I don't know how much historical pain django has caused, but I think it'd be great to decrease our packaging burden - especially if we don't have a specific reason not to.

David-Baddeley, 2020-01-14:
The 3 issues with django are:

1. Django breaks backwards compatibility really fast
2. For security reasons, it's always a good idea to run the latest version of django (this means we can't / shouldn't pin to a fixed version of django)
3. Django is a big package with loads of dependencies (makes satisfying these as well as all the standard PYME dependencies a headache, particularly on conda) 

Having to handle the whole of django just to get a bit of non-core functionality is a pain (which is presumably why it was factored out in the first place). The short of it is, adding django will likely compound our dependency issues rather than alleviate them. As it is the dispatch package has no external dependencies (which is really great :).

an additional option to those in my first reply would be to git link (I don't think the terminology is quite correct) the dispatch folder from django into pyme.contrib - that way the code would track the django head, but we wouldn't have to deal with the whole django package headaches.

David-Baddeley, 2020-01-15:
Having looked a bit further the git-link option I was referring to is subrepos, and probably not what we want.

I've added a `requirements.txt` to PYME (currently not well tested) which should setup most of the dependencies, including dispatch, via pip for those who don't want to use conda. ie new setup instructions if you don't want to use conda are:

```bash
git clone https://github.com/python-microscopy/python-microscopy.git
cd python-microscopy
pip install -r requirements.txt
python setup.py develop
```

David-Baddeley, 2020-10-19:
should now be fixed on py3

closed 2020-10-19 by David-Baddeley