python-microscopy/python-microscopy #129: interface for filterwheel

carandraug, 2020-01-16:
I am trying to implement an adapter for Python Microscope filterwheels
(see related PR #127).

As far as I can figure out from the code, the only filterwheel
supported is Thorlabs FW102.  However, instead of constructing the
hardware object (`fw102.FW102B`) and pass it to the related GUI class
(`FilterWheel.FiltWheel`), which is what seems to happen with other
devices, the filter wheel GUI class instantiates the actual hardware
controlling object.

This situation makes it hard to support other filterwheel.  Without
modifying the `FilterWheel.FiltWheel`, the only solution I see is to
monkey patch the import `fw102.FW102B` which I guess is not desirable.
Or am I missing something?

I would like to have this changed to add more filterwheels but I fear
it won't be backwards compatible.  Some pointers before I start this?

David-Baddeley, 2020-01-16:
Nope, not missing anything. The current filter wheel support is arse about face and not really something I'm proud of. I just never got around to changing it because there was never a pressing need. The filter wheel GUI as it stands is also a bit rough (has some synchronisation issues where it might not accurately reflect the current state of the filter wheel). The whole lot needs a bunch of refactoring.

The `FilterWheel.FiltWheel` class itself is not a GUI class, but rather the (common) filter wheel interface (think of it as a "controler" in a MVC pattern) which acts as a shim between the filter wheel and the GUI. It just happens to be in the same module as the GUI class. 

There's also a bunch of stuff to automatically change the filter wheel in the excitation to automatically match the currently selected dichroic in the Nikon Ti stand. This is pretty specific to one microscope setup and might not generalise well.

Moving forward, there would be two logical approaches:
* make `FilterWheel.FiltWheel` (or the class signature defined therein) the base class for filter wheels, and subclass this for additional filter wheels.
* keep `FiltWheel` as a shim, but pass in an instance of a class whose signature matches the FW102 signature rather than a serial port.

There's arguments for both approaches  - or even a hybrid. Subclassing `FiltWheel` (or a new `FilterWheelBase` with the same method signatures) would make things more consistent with how other hardware is handled, but the `FiltWheel` class manages mappings between filter names and numeric positions so you can, e.g. call `FiltWheel.SetFilterPos('GFP')` and it will translate this into the relevant physical position. This is not something you would want to replicate for each filter wheel (hence the argument for instantiating something which has the same interface as FW102 and passing it in to the constructor).

The hybrid approach transfers the entirety of `FiltWheel` to the base class, but adds a couple of virtual private functions `_set_physical_pos(int pos)` and `_get_physical_pos()` to replace `self.fw.setPos()` and `self.fw.getPos()`.

To maintain backwards compatibility, we could have:

`FilterWheelBase` : the new base class which is essentially `FiltWheel` with the proposed additions above and the fw102 specific stuff removed.

`FiltWheel` : a derived class for fw102 (same name so as to keep backwards compatibility for now)

additional subclasses for other filter wheels.

Ultimately the GUI (`FilterWheel.FiltFrame`) should move out into it's own file.

Things are further complicated by the fact that `ExciterWheel.py` is a subtly modified copy of `FilterWheel.py`

David-Baddeley, 2020-01-16:
Scratch the ExciterWheel comment - it doesn't appear to be used anywhere.

David-Baddeley, 2020-01-16:
I think I'll go ahead and modify the base class using the hybrid approach (having more or less worked out how to do it already).

David-Baddeley, 2020-01-16:
As of 8861b58528a332576e74d053de6bdca8b838d0b1 we now have a filter wheel base class (which should hopefully work).

carandraug, 2020-01-16:
Nice. Thank you.

Can you comment on 6a78c6d51c208c374 (or my "microscope-fw-adapter" branch) as an implementatoin of filterwheel adapter? I can't test them at home but this should add support for Thorlabs FW212, Zaber, and Prior.

David-Baddeley, 2020-02-11:
Guess I can close this now.

closed 2020-02-11 by David-Baddeley