python-microscopy/python-microscopy #153: Will it work for python3?

michalstepniewski, 2020-02-16:
Is it possible to get it to work with python? What are the likely challenges? I would like to write a plugin using latest libraries, that is why i ask...

barentine, 2020-02-16:
Hi Michal, it is possible to use Python 3 - I am currently running a microscope and computer cluster on Python 3.6. I haven't tested py3.7 or 3.8.

We are behind on building/hosting conda packages for Python 3 (though the dependencies are there for mac and windows, and the whole thing is there for linux, all py3.6), but if you clone the [conda-recipes repository](https://github.com/python-microscopy/pyme-conda-recipes) you can build them locally easy enough. 

Numpy wise I am using Numpy 1.16. There are a couple visible deprecation warnings (`arr[tuple(seq)]` vs `arr[seq]`) which I haven't addressed yet, so if you want to push to Numpy 1.17 or 1.18 I would maybe look at 1.16 first and squash those warnings. Would love to see a PR on it if you do!

@David-Baddeley  - what do you recon we have left to do before making py36 numpy1.16 the default install?

michalstepniewski, 2020-02-16:
Oh, that's great to here. Do I really need conda to install it or is it possible with pip also? :)

David-Baddeley, 2020-02-16:
Hi Michael,

The answer to your question is "it depends on what you want to do" - we've got about 95% of stuff working on python3.6, numpy 1.14, but there are a few corner cases which don't work (either our code, or dependencies which haven't been ported).

The most critical roadblock is likely to be that our default localization analysis pipeline no longer works on Py3 due to a dependency which hasn't been ported. We have an alternative which  should represent a significant improvement on the current version, but it's not quite "production ready" and is still a bit of a headache to get configured and running. It's also out of sync with the current documentation.

The other place where you might run into issues is getting all the dependencies installed. Most of our py3 development to date has targeted py3.6 and numpy 1.14 and we have conda- installable packages for most of the dependencies (in the david_baddeley anaconda channel for those which aren't in the default anaconda repositories). I'd strongly recommend  py3.6, numpy 1.14 as a starting point - and at least get everything working there before trying higher versions. Andrew's developing against numpy 1.16, and it seems to work for the subset of the code that he's using, but there is a very high likelihood that there will be some parts of the code that are broken on 1.16 - some of our code dates back to numpy 0.x and there have been several deprecation cycles since then. We've caught all the important stuff up to 1.14 but use anything higher at your peril (bug reports welcome). Although the dependencies are packaged, we don't have packages built for pyme itself on python 3, so you'll need to install that from source (probably not a bad idea anyway if you are developing plugins).

If you choose to use pip rather than conda for dependency management, we now have a `requirements.txt` that you can use with `pip install -r`. Note that this is incomplete, so you will still get the occasional import error which you need to track down and install.

There are additionally some minor issues UI around a dependency, traitsui, on py3 (or more specifically wx4, but as wx4 is the lowest version of wx supported on py3 it amounts to the same thing). These mostly manifest as a whole bunch of deprecation warnings, but might also result in a few cosmetic glitches. These are slowly being addressed in the traitsui upstream, so it might make sense to install traitsui from the repository head rather than using conda packages (although the lag time between conda and the repository doesn't seem to be too high).

In terms of bumping the default install to py3.6, there are currently 4 roadblocks:
- the localization analysis pipeline replacement (which is around 95% done)
- an alternative to Mayavi for iso-surfaces etc ... (probably around 80-90% done)
- changing how IPC is done in our acquisition time drift tracking module
- An internal plugin used for post-analysis drift correction which hasn't been ported 

The latter 2 should only affect my group and a couple of collaborators, and should not be hugely complex once we find time to do them. I'm currently not in any great hurry to bump the default numpy version past 1.14 (in general things will work fine on higher versions, but 1.14 seems to be a good compromise in terms of compatibility), but do plan on starting to test against both 1.14 and the numpy head once we've completed the py3 transition.

David-Baddeley, 2020-02-16:
Posted at the same time - short answer to the pip question, yes it is possible, maybe not as smooth (more details in the blurb above). You will need a c compiler set up to pip-install.

David-Baddeley, 2020-02-16:
Two potential hiccups in the pip install is that there are no pypi modules for either `dispatch` or `pymecompress`, and these need to be installed direct from the github repos. `requirements.txt` should handle that, if you are starting from a partial install (and/or want to try different numpy versions etc ... look in `requirements.txt` for the appropriate repo urls.

David-Baddeley, 2020-10-19:
I think we can comfortably close this one as we've pretty much checked off all the important py3 issues

closed 2020-10-19 by David-Baddeley