python-microscopy/python-microscopy #155: Read noise and electrons per count standardisation / bug in Orca

David-Baddeley, 2020-02-20:
At the moment the different cameras report their read noise in different units.

AndorZyla - read noise is in electrons, electrons per count correct
AndorIXon - read noise in electrons, electrons per count probably correct
HamamatsuOrca - read noise is in ADUs (wrong), electrons per count looks correct?

Haven't checked the UEye cameras (but these will not be well calibrated in any case)

barentine, 2020-02-20:
Looks like Orca read noise 'regressed' when remFitBuf getvarmap was updated last month 7f071cd3ed957defc3c6c6d8e06cd53f0a3dec29.  Thanks for catching this, David.

Orca e-/ADU should be right.

David-Baddeley, 2020-02-20:
This whole mess was likely due to an error on my part in the metadata documentation and the analysing foreign data docs, which propagated across to the Orca class when it was written. 

I've fixed all the docs to specify read noise in photoelectrons, and adjusted the Orca values to what they should hopefully be. 

Last months update to remFitBuf was unwinding a change I'd made in Dec after talking with Andrew about units, which had messed up analysis of data coming from the Andor cameras (the code, as it stood was expecting electrons). 

Moving forward things should be fine, but we will be left with a bunch of data generated by the Orca which has incorrect metadata. I'm guessing that mostly effects you, @barentine - and the severity of the issue will likely depend on whether you are regularly using camera maps for analysis.

Tagging @zacsimile as well as another potential Orca user.

David-Baddeley, 2020-02-20:
Closing now, as should be fixed

closed 2020-02-27 by David-Baddeley