python-microscopy/python-microscopy #160: launchWorkers -> PYMEClusterOfOne migration

David-Baddeley, 2020-02-26:
For full Py3 compatibility we need to replace our task distribution setup. We currently have an alpha quality replacement (PYMEClusterOfOne). This issue summarises the remaining tasks before we can switch off support for the old distribution and push all users onto the Py3 version.

Before recommending ClusterOfOne for new installs/users
-----------------------------------------------------------

- loads and loads of testing
- [ ] nodeserver currently fails to find ruleserver when tested in my windows VM (this might be a race condition, exacerbated by the windows VM being slow)
- [ ] last time I tested on a physical windows machine, had incomplete spooling / analysis. This could be one of 3 things - issues with flushing data to files, spooling not keeping up with data generation, or a bug in the analysis which has since been fixed.
- [ ] components use too much CPU when idle (often spiking to 100% for a minute or so). I think this might only occur on a crowded network, and might have something to do with zeroconf.
- quite a lot of functionality has not been tested at all (incomplete list of potentially problematic cases below)
   + [ ] fits which need a PSF measurement
   + [ ] ROI extraction fits
- [ ] checking and tweaks to the dh5view LMAnalysis module to make sure it uses the cluster of one where appropriate.
- [ ] making the UI around 'spool to h5 on cluster' a bit clearer / more automated and checking that performance of the h5 on cluster backend is adequate
- [ ] checking that things fail gracefully (or at least semi-gracefully) when saving is not keeping up with acquisition.
- [ ] possibly revisit UI (clusterUI might not be that appropriate for most cluster or one use cases) - this could potentially be deferred to step 2
- [ ] update dependencies (django) if we stick with clusterUI for the cluster of one case.

Before dropping support for launchWorkers / making any changes which could break launchWorkers / QueueSpooler
-------------------------------------------------------------------------------------------------------------

- feature parity:
    + [ ] real time localisation display in dh5view (super useful for standard STORM and particularly PAINT imaging where you might
        want to abort part way through if the data you are getting is shit)
    + [ ] free disk space reporting
    + [ ] aborting a running analysis part way through
    + [ ] distributed analysis where the data is not also distributed
- more testing (in practice, this probably means having used it in production on a windows machine with a variety of different workloads for 3-6 months)

There are a few other things which are likely to break (e.g. distributed deconvolution) but are either sufficiently poorly used or already broken that I don't consider them blocking.

barentine, 2020-03-04:
What do you have in mind for django versions, @David-Baddeley? 1.11 looks to be the latest py2/py3 version hosted in the anaconda channel. Bump to that and then django 3 once we finish this checklist?

barentine, 2020-03-04:
After talking to David current plan is push to django 1.11

barentine, 2020-03-08:
How do we want to svgwrite packaging? It is hosted in several conda repositories, but not anaconda nor @David-Baddeley's. I assume we will want to add django and svgwrite both as dependencies for pyme-depends.

zacsimile, 2020-04-24:
Tried to run this from Windows. Had to first type `conda install -c anaconda django=1.11`. Then ran `pymeclusterofone`. The web GUI popped up no problem. I then launched `pymeacquire` and spooled some empty frames to the cluster. The file was generated and saved properly. The console spewed out

```
Traceback (most recent call last):
  File "c:\users\zrc4\code\python-microscopy\PYME\IO\h5rFile.py", line 194, in _pollQueues
    self._appendToTable(tablename, waiting.popleft())
  File "c:\users\zrc4\code\python-microscopy\PYME\IO\h5rFile.py", line 127, in _appendToTable
    expectedrows=500000)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\tables\file.py", line 1055, in create_table
    chunkshape=chunkshape, byteorder=byteorder)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\tables\table.py", line 799, in __init__
    nparray = numpy.rec.array(description)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\numpy\core\records.py", line 850, in array
    return fromstring(obj, dtype, shape=shape, offset=offset, **kwds)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\numpy\core\records.py", line 723, in fromstring
    raise TypeError("fromstring() needs a 'dtype' or 'formats' argument")
TypeError: fromstring() needs a 'dtype' or 'formats' argument

DEBUG:root:H5RFile - closing: C:\Users\zrc4\zrc4\2020_4_24\24_4_series_A.h5
DEBUG:root:H5RFile - closed: C:\Users\zrc4\zrc4\2020_4_24\24_4_series_A.h5
DEBUG:root:pytables open call: C:\Users\zrc4\zrc4\2020_4_24\24_4_series_A.h5
DEBUG:root:pytables file open: C:\Users\zrc4\zrc4\2020_4_24\24_4_series_A.h5
Traceback (most recent call last):
  File "c:\users\zrc4\code\python-microscopy\PYME\IO\h5rFile.py", line 117, in _appendToTable
    table = getattr(self._h5file.root, tablename)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\tables\group.py", line 818, in __getattr__
    return self._f_get_child(name)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\tables\group.py", line 698, in _f_get_child
    self._g_check_has_child(childname)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\tables\group.py", line 395, in _g_check_has_child
    % (self._v_pathname, name))
tables.exceptions.NoSuchNodeError: group ``/`` does not have a child named ``PZFImageData``

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\users\zrc4\code\python-microscopy\PYME\IO\h5rFile.py", line 194, in _pollQueues
    self._appendToTable(tablename, waiting.popleft())
  File "c:\users\zrc4\code\python-microscopy\PYME\IO\h5rFile.py", line 127, in _appendToTable
    expectedrows=500000)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\tables\file.py", line 1055, in create_table
    chunkshape=chunkshape, byteorder=byteorder)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\tables\table.py", line 799, in __init__
    nparray = numpy.rec.array(description)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\numpy\core\records.py", line 850, in array
    return fromstring(obj, dtype, shape=shape, offset=offset, **kwds)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\numpy\core\records.py", line 723, in fromstring
    raise TypeError("fromstring() needs a 'dtype' or 'formats' argument")
TypeError: fromstring() needs a 'dtype' or 'formats' argument
ERROR:root:Traceback (most recent call last):
  File "c:\users\zrc4\code\python-microscopy\PYME\IO\h5rFile.py", line 117, in _appendToTable
    table = getattr(self._h5file.root, tablename)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\tables\group.py", line 818, in __getattr__
    return self._f_get_child(name)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\tables\group.py", line 698, in _f_get_child
    self._g_check_has_child(childname)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\tables\group.py", line 395, in _g_check_has_child
    % (self._v_pathname, name))
tables.exceptions.NoSuchNodeError: group ``/`` does not have a child named ``PZFImageData``

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\users\zrc4\code\python-microscopy\PYME\IO\h5rFile.py", line 194, in _pollQueues
    self._appendToTable(tablename, waiting.popleft())
  File "c:\users\zrc4\code\python-microscopy\PYME\IO\h5rFile.py", line 127, in _appendToTable
    expectedrows=500000)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\tables\file.py", line 1055, in create_table
    chunkshape=chunkshape, byteorder=byteorder)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\tables\table.py", line 799, in __init__
[... 83 lines cut ...]
Traceback (most recent call last):
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\wx\core.py", line 3259, in <lambda>
    lambda event: event.callable(*event.args, **event.kw) )
  File "c:\users\zrc4\code\python-microscopy\PYME\DSView\dsviewer.py", line 428, in LoadData
[... cut ...]
    sl = PZFFormat.loads(clusterIO.get_file(frameName, self.clusterfilter))[0]
  File "c:\users\zrc4\code\python-microscopy\PYME\IO\clusterIO.py", line 788, in get_file
    raise IOError("Specified file could not be found: %s" % filename)
OSError: Specified file could not be found: zrc4/2020_4_24/24_4_series_A.h5/frame00000.pzf
c:\users\zrc4\code\python-microscopy\PYME\DSView\splashScreen.py:117: wxPyDeprecationWarning: Call to deprecated item EmptyBitmap. Use :class:`wx.Bitmap` instead
  MemBitmap = wx.EmptyBitmap(s.GetWidth(), s.GetHeight())
```

barentine, 2020-04-26:
Thanks for giving that a go, Zach! There is [an open PR](https://github.com/python-microscopy/pyme-conda-recipes/pull/3) in the pyme-conda-recipes repository to add django 1.11 as a req. @David-Baddeley any thoughts on that?

I'll have a look at the numpy fromstring error, which I think is something that arrived with np 1.14 and we've hit in other places already, but that append to table would've caused the remaining errors.

barentine, 2020-04-26:
Hmm, actually, what version of numpy and pytables were you using, @zacsimile ?

zacsimile, 2020-04-26:
numpy=1.18.1 and pytables=3.4.2

barentine, 2020-04-26:
Note that this bug also exists on numpy 1.14, tables 3.4.2, and numpy 1.18 tables 3.6.1. Issue is leaving the data as a bytes instance registers as a 'python' dtype flavor, so tables calls `numpy.rec.array(description)` with no further dtype / format parameters.

barentine, 2020-04-26:
Ahhhh, it seems six has failed us and the bytes wasn't caught as a str in H5RFile._appendToTable.
```
type(data)
Out[8]: bytes
isinstance(data, six.string_types)
Out[9]: False
six.string_types
Out[10]: (str,)
```
Will submit a PR shortly

zacsimile, 2020-04-29:
OK, those errors are fixed! I re-ran this on my Windows 10 machine again. No errors on the `pymeclusterofone` side now, but I get the following errors if I spool in `PYMEAcquire` and then press Analyze. I checked in the file system and the `h5` file definitely exists at the right location.

```
DEBUG:PYME.Acquire.SpoolController:Looking for zrc4/2020_4_29/29_4_series_B (.pcs or .h5) on cluster

('currSlide:', [None])
DEBUG:PYME.Acquire.HTTPSpooler:Starting spooling: zrc4/2020_4_29/29_4_series_B.h5

'http://127.0.0.1:57286/__aggregate_h5/zrc4/2020_4_29/29_4_series_B.h5/metadata.json'
Warning: not keeping up with camera, giving up with 1 frames still in buffer
DEBUG:PYME.Acquire.Spooler:Disconnecting from frame source

DEBUG:PYME.Acquire.Spooler:Frame source should be disconnected

DEBUG:PYME.Acquire.HTTPSpooler:Stopping spooling zrc4/2020_4_29/29_4_series_B.h5

'http://127.0.0.1:57286/__aggregate_h5/zrc4/2020_4_29/29_4_series_B.h5/final_metadata.json'
'http://127.0.0.1:57286/__aggregate_h5/zrc4/2020_4_29/29_4_series_B.h5/events.json'
Warning: not keeping up with camera, giving up with 9 frames still in buffer
INFO:numexpr.utils:Note: NumExpr detected 12 cores but "NUMEXPR_MAX_THREADS" not set, so enforcing safe limit of 8.
INFO:numexpr.utils:NumExpr defaulting to 8 threads.
Starting main loop
Loading data
filename == PYME-CLUSTER:///zrc4/2020_4_29/29_4_series_B.h5
ERROR:PYME.IO.clusterIO:Timeout on get file
Traceback (most recent call last):
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\urllib3\connectionpool.py", line 672, in urlopen
    chunked=chunked,
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\urllib3\connectionpool.py", line 421, in _make_request
    six.raise_from(e, None)
  File "<string>", line 3, in raise_from
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\urllib3\connectionpool.py", line 416, in _make_request
    httplib_response = conn.getresponse()
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\http\client.py", line 1331, in getresponse
    response.begin()
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\http\client.py", line 297, in begin
    version, status, reason = self._read_status()
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\http\client.py", line 266, in _read_status
    raise RemoteDisconnected("Remote end closed connection without"
http.client.RemoteDisconnected: Remote end closed connection without response

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\requests\adapters.py", line 449, in send
    timeout=timeout
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\urllib3\connectionpool.py", line 720, in urlopen
    method, url, error=e, _pool=self, _stacktrace=sys.exc_info()[2]
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\urllib3\util\retry.py", line 400, in increment
    raise six.reraise(type(error), error, _stacktrace)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\urllib3\packages\six.py", line 734, in reraise
    raise value.with_traceback(tb)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\urllib3\connectionpool.py", line 672, in urlopen
    chunked=chunked,
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\urllib3\connectionpool.py", line 421, in _make_request
    six.raise_from(e, None)
  File "<string>", line 3, in raise_from
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\urllib3\connectionpool.py", line 416, in _make_request
    httplib_response = conn.getresponse()
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\http\client.py", line 1331, in getresponse
    response.begin()
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\http\client.py", line 297, in begin
    version, status, reason = self._read_status()
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\http\client.py", line 266, in _read_status
    raise RemoteDisconnected("Remote end closed connection without"
urllib3.exceptions.ProtocolError: ('Connection aborted.', RemoteDisconnected('Remote end closed connection without response',))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
[... 4 lines cut ...]
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\requests\sessions.py", line 530, in request
    resp = self.send(prep, **send_kwargs)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\requests\sessions.py", line 643, in send
    r = adapter.send(request, **kwargs)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\requests\adapters.py", line 498, in send
    raise ConnectionError(err, request=request)
requests.exceptions.ConnectionError: ('Connection aborted.', RemoteDisconnected('Remote end closed connection without response',))
INFO:PYME.IO.clusterIO:2 retries left
ERROR:PYME.IO.clusterIO:Timeout on get file
Traceback (most recent call last):
[... cut ...]
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\http\client.py", line 1331, in getresponse
    response.begin()
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\http\client.py", line 297, in begin
    version, status, reason = self._read_status()
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\http\client.py", line 266, in _read_status
    raise RemoteDisconnected("Remote end closed connection without"
http.client.RemoteDisconnected: Remote end closed connection without response
```

David-Baddeley, 2020-04-29:
That looks like it's probably a data server error - maybe check `PYMEDataServer.log`

zacsimile, 2020-04-29:
Where is `PYMEDataServer.log` supposed to be located? I looked in all the usual places and found nothing. I also did a search on the Windows C drive and didn't find this file.

David-Baddeley, 2020-04-29:
should be in `pyme.config.get('dataserver-root')/LOGS/`, although this might be related to the 'fix default directory if none specified' point above - I believe that `dataserver-root` is undefined in a std install - whereas it should probably default to PYMEDATADIR (which in turn defaults to <home>/PYMEData). Likely need to add a default to the config.get call on l55 of HTTPDataServer

David-Baddeley, 2020-04-29:
trying to remember where the defaulting logic for PYMEDATADIR is - maybe nameUtils ?

David-Baddeley, 2020-04-29:
closing was a mistake - hit the wrong button when commenting

zacsimile, 2020-04-29:
I think it may be called something besides `PYMEDataServer.log` because I really cannot find that file.

zacsimile, 2020-04-29:
The data gets stored in `C:\Users\<username>\<username>\<date>\<series>`, but there's no log file.

David-Baddeley, 2020-04-30:
Should have been a bit clearer - PYMEDataServer writes it's log to STDERR if `dataserver-root` is undefined.

David-Baddeley, 2020-04-30:
See lines 56-72 of HTTPDataServer

zacsimile, 2020-04-30:
Aha. You are right. My terminal must have frozen the last time I ran this. This time, I have a log from the `pymeclusterofone` terminal.

```
DEBUG:root:pytables file open: C:\Users\zrc4\zrc4\2020_4_30\30_4_series_A.h5
----------------------------------------
Exception happened during processing of request from ('127.0.0.1', 51909)
Traceback (most recent call last):
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\socketserver.py", line 651, in process_request_thread
    self.finish_request(request, client_address)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\socketserver.py", line 361, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\socketserver.py", line 721, in __init__
    self.handle()
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\http\server.py", line 420, in handle
    self.handle_one_request()
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\http\server.py", line 406, in handle_one_request
    method()
  File "c:\users\zrc4\code\python-microscopy\PYME\cluster\HTTPDataServer.py", line 482, in do_GET
    f = self.send_head()
  File "c:\users\zrc4\code\python-microscopy\PYME\cluster\HTTPDataServer.py", line 569, in send_head
    return self.get_h5_part(path)
  File "c:\users\zrc4\code\python-microscopy\PYME\cluster\HTTPDataServer.py", line 669, in get_h5_part
    f, length = self._string_to_file(h5f.get_file(part))
  File "c:\users\zrc4\code\python-microscopy\PYME\IO\h5File.py", line 123, in get_file
    return self.get_frame(frame_num)
  File "c:\users\zrc4\code\python-microscopy\PYME\IO\h5File.py", line 99, in get_frame
    data = self.image_data[idx]
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\tables\vlarray.py", line 671, in __getitem__
    return self.read(start, stop, step)[0]
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\tables\vlarray.py", line 811, in read
    listarr = self._read_array(start, stop, step)
  File "tables\hdf5extension.pyx", line 2106, in tables.hdf5extension.VLArray._read_array
ValueError: cannot set WRITEABLE flag to True of this array
----------------------------------------
----------------------------------------
Exception happened during processing of request from ('127.0.0.1', 51910)
Traceback (most recent call last):
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\socketserver.py", line 651, in process_request_thread
    self.finish_request(request, client_address)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\socketserver.py", line 361, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\socketserver.py", line 721, in __init__
    self.handle()
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\http\server.py", line 418, in handle
    self.handle_one_request()
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\http\server.py", line 406, in handle_one_request
    method()
  File "c:\users\zrc4\code\python-microscopy\PYME\cluster\HTTPDataServer.py", line 482, in do_GET
    f = self.send_head()
  File "c:\users\zrc4\code\python-microscopy\PYME\cluster\HTTPDataServer.py", line 569, in send_head
    return self.get_h5_part(path)
  File "c:\users\zrc4\code\python-microscopy\PYME\cluster\HTTPDataServer.py", line 669, in get_h5_part
    f, length = self._string_to_file(h5f.get_file(part))
  File "c:\users\zrc4\code\python-microscopy\PYME\IO\h5File.py", line 123, in get_file
    return self.get_frame(frame_num)
  File "c:\users\zrc4\code\python-microscopy\PYME\IO\h5File.py", line 99, in get_frame
    data = self.image_data[idx]
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\tables\vlarray.py", line 671, in __getitem__
    return self.read(start, stop, step)[0]
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\tables\vlarray.py", line 811, in read
    listarr = self._read_array(start, stop, step)
  File "tables\hdf5extension.pyx", line 2106, in tables.hdf5extension.VLArray._read_array
ValueError: cannot set WRITEABLE flag to True of this array
----------------------------------------
----------------------------------------
Exception happened during processing of request from ('127.0.0.1', 51911)
Traceback (most recent call last):
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\socketserver.py", line 651, in process_request_thread
    self.finish_request(request, client_address)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\socketserver.py", line 361, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\socketserver.py", line 721, in __init__
    self.handle()
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\http\server.py", line 418, in handle
    self.handle_one_request()
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\http\server.py", line 406, in handle_one_request
    method()
  File "c:\users\zrc4\code\python-microscopy\PYME\cluster\HTTPDataServer.py", line 482, in do_GET
    f = self.send_head()
[... cut ...]
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\tables\vlarray.py", line 671, in __getitem__
    return self.read(start, stop, step)[0]
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\tables\vlarray.py", line 811, in read
    listarr = self._read_array(start, stop, step)
  File "tables\hdf5extension.pyx", line 2106, in tables.hdf5extension.VLArray._read_array
ValueError: cannot set WRITEABLE flag to True of this array
----------------------------------------
```

barentine, 2020-04-30:
Haha! Just dealt with this yesterday while helping someone. It's because we pin to an old version of pytables. See issue #215 . 

If you update you should be golden @zacsimile . Give >=3.5.2 a spin

barentine, 2020-04-30:
and thanks for the testing!

zacsimile, 2020-04-30:
Yep, ran `pip install tables=3.5.2` and it works now! I can spool, stop spooling, press Analyze, and `dh5view` comes up. The cluster web GUI still doesn't list the files. If I refresh the file list by clicking `Files`, I get the following error in the `pymeclusterofone` terminal.

```
ERROR:root:127.0.0.1 - - [30/Apr/2020 12:05:38] code 404, message No permission to list directory
```

zacsimile, 2020-04-30:
Follow up: both real-time (while spooling) and after-the-fact localization analysis appear to work. And I can see the jobs in the web GUI. The main thing that's missing now is listing the files in the web GUI.

David-Baddeley, 2020-04-30:
The listing problem looks like it might possibly be an `os.sep` issue. Thanks for all the debugging work. Unfortunately for the pytables stuff we probably need to do a bit of a matrix of what pytables, python, and numpy versions work together ... and make sure that the conda-installed pytables also works (the original reason for pinning is about equally likely to be a conda packaging error as an error in pytables itself)

zacsimile, 2020-04-30:
Well, we know it works with `pytables=3.5.2`, `python=3.6.7`, and `numpy=1.18.1` (my setup), which I think is an acceptable configuration. `pytables=3.4` doesn't work, and `pytables=3.6` is Python 3 only. Conda was having a helluva time solving the environment for `pytables=3.5.2`, but `pip` just installed it and then everything worked fine, so it definitely works. There is a `pytables=3.5.1` in Conda, but that's the only other 3.5 package, and I'm not sure it's sufficient.

zacsimile, 2020-05-18:
Did we decide to upgrade to `pytables=3.5.2`? If so, this should now work on Windows, and we can check off box 2 in the original post.

David-Baddeley, 2020-05-18:
I've removed the pin, but upgrading pytables won't allow us to check off number 2. This was with a working pytables on py2 (i.e. not the errors you were getting) and incomplete spooling rather than not spooling at all. The errors were highly likely not pytables (see suggestions in the point itself).

David-Baddeley, 2020-05-18:
Point 2 is a bit fuzzy, unfortunately. To check it off I'd like the following (on both py2 and py3 with numpy=1.14). For a number of repeated spoolings (simulator should be sufficient):

- spool to file. 
- launch analysis while spooling is running.
- run for several thousand frames
- stop (and record number of frames spooled as shown in UI)

Open saved sequence in dh5view (from file on disk, not from cluster URL):
- does the number of frames in the file match the number of frames spooled as shown in ui?

Open the .h5r. 
- Does the maximum frame number in the analysis match the number of frames spooled (or get within a few frames, allowing for the possibility of having no localisations in the last few frames)?
- Are there any gaps in analysis (if you colour by time and look at the histogram)?

zacsimile, 2020-05-18:
Apologies, I was unclear. Upgrading `pytables` fixed the last issue I was having on Windows. After that was fixed, the data seemed to spool correctly and completely, hence my suggesting checking off point 2.

zacsimile, 2020-05-18:
I can't do it with `numpy=1.14`. `pytables=3.5.2` compatible with `numpy=1.16` or later. `numpy=1.16` runs on py2, would that be sufficient?

David-Baddeley, 2020-05-18:
I don't think we need to pin pytables to 3.5.2. When I removed the pin I could not get the pytables issue to reappear, regardless of version, so suspect it was some broken packages which have since been fixed. Our numpy support policy at present is: 
- must work with numpy=1.14, should work with 1.14 < numpy < 2.0 (i.e. if it doesn't work with a higher numpy it's a problem we should fix, but it's unacceptable to require numpy >= 1.15.0). This is so that we can claim a known working set of packages.

zacsimile, 2020-05-18:
Well, huge bummer, I went to test this and I can't get the analysis to run anymore, live or after-the-fact. I tried twice to make sure the error is repeatable.

```
Loading data
filename == PYME-CLUSTER:///zrc4/2020_5_18/18_5_series_B.h5
ERROR:PYME.IO.clusterIO:Timeout on get file
Traceback (most recent call last):
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\urllib3\connectionpool.py", line 677, in urlopen
    chunked=chunked,
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\urllib3\connectionpool.py", line 426, in _make_request
    six.raise_from(e, None)
  File "<string>", line 3, in raise_from
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\urllib3\connectionpool.py", line 421, in _make_request
    httplib_response = conn.getresponse()
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\http\client.py", line 1331, in getresponse
    response.begin()
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\http\client.py", line 297, in begin
    version, status, reason = self._read_status()
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\http\client.py", line 266, in _read_status
    raise RemoteDisconnected("Remote end closed connection without"
http.client.RemoteDisconnected: Remote end closed connection without response

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\requests\adapters.py", line 449, in send
    timeout=timeout
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\urllib3\connectionpool.py", line 725, in urlopen
    method, url, error=e, _pool=self, _stacktrace=sys.exc_info()[2]
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\urllib3\util\retry.py", line 403, in increment
    raise six.reraise(type(error), error, _stacktrace)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\urllib3\packages\six.py", line 734, in reraise
    raise value.with_traceback(tb)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\urllib3\connectionpool.py", line 677, in urlopen
    chunked=chunked,
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\urllib3\connectionpool.py", line 426, in _make_request
    six.raise_from(e, None)
  File "<string>", line 3, in raise_from
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\urllib3\connectionpool.py", line 421, in _make_request
    httplib_response = conn.getresponse()
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\http\client.py", line 1331, in getresponse
    response.begin()
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\http\client.py", line 297, in begin
    version, status, reason = self._read_status()
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\http\client.py", line 266, in _read_status
    raise RemoteDisconnected("Remote end closed connection without"
urllib3.exceptions.ProtocolError: ('Connection aborted.', RemoteDisconnected('Remote end closed connection without response',))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\users\zrc4\code\python-microscopy\PYME\IO\clusterIO.py", line 798, in get_file
    r = s.get(url, timeout=.5)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\requests\sessions.py", line 543, in get
    return self.request('GET', url, **kwargs)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\requests\sessions.py", line 530, in request
    resp = self.send(prep, **send_kwargs)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\requests\sessions.py", line 643, in send
    r = adapter.send(request, **kwargs)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\requests\adapters.py", line 498, in send
    raise ConnectionError(err, request=request)
requests.exceptions.ConnectionError: ('Connection aborted.', RemoteDisconnected('Remote end closed connection without response',))
INFO:PYME.IO.clusterIO:2 retries left
ERROR:PYME.IO.clusterIO:Timeout on get file
Traceback (most recent call last):
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\urllib3\connectionpool.py", line 677, in urlopen
    chunked=chunked,
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\urllib3\connectionpool.py", line 426, in _make_request
    six.raise_from(e, None)
  File "<string>", line 3, in raise_from
[... 159 lines cut ...]
    self.Load(filename, prompt=load_prompt)
  File "c:\users\zrc4\code\python-microscopy\PYME\IO\image.py", line 1019, in Load
    self._loadClusterPZF(filename)
  File "c:\users\zrc4\code\python-microscopy\PYME\IO\image.py", line 498, in _loadClusterPZF
    self.data = BGSDataSource.DataSource(self.dataSource) #this will get replaced with a wrapped version
  File "c:\users\zrc4\code\python-microscopy\PYME\IO\DataSources\BGSDataSource.py", line 225, in __init__
    self.bBufferMn = backgroundBuffer(self.dBuffer)
[... cut ...]
    resp = self.send(prep, **send_kwargs)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\requests\sessions.py", line 643, in send
    r = adapter.send(request, **kwargs)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\requests\adapters.py", line 498, in send
    raise ConnectionError(err, request=request)
requests.exceptions.ConnectionError: ('Connection aborted.', RemoteDisconnected('Remote end closed connection without response',))
```

David-Baddeley, 2020-05-18:
I noticed this coming through in the unit tests lately but had written it off as a quirk with the azure environment - not sure what the issue is as the code hasn't really changed at all.

It looks like the data server is not running (but is somehow still registered). Looking at the data server logs could be useful ... It might be that I broke something whilst trying to fix the default directory to run the dataserver from if no directory was specified ...

zacsimile, 2020-05-18:
Just updated with the server output. Was supposed to be in there the first time. Remote workstation copy+paste issues.

zacsimile, 2020-05-18:
Oh funny. My `pytables` downgraded to `3.4.2`. Going to push it back to `3.5.2` and see if that fixes it.

David-Baddeley, 2020-05-18:
This really doesn't look like a pytables issue (if it's a package version issue it's most likely to be requests).

Try the following in an ipython session (while the dataserver is running)

```python
import clusterIO
clusterIO.get_ns().get_advertised_services()
```

It should list one (and only one) service for the data server.

zacsimile, 2020-05-19:
This is what I get

````
ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-1-71f7982e240e> in <module>
----> 1 import clusterIO
      2 clusterIO.get_ns().get_advertised_services()

ModuleNotFoundError: No module named 'clusterIO'
```

zacsimile, 2020-05-19:
OK, I figured it out. `from PYME.IO import clusterIO`.

```
[('PYMEDataServer [MWD09E8F40]: MWD09E8F40 - PID:5940',
  NSInfo(name='PYMEDataServer [MWD09E8F40]: MWD09E8F40 - PID:5940', address=b'\x7f\x00\x00\x01', port=51486, creation_time=1589846396.8291943, URI=''))]
```

barentine, 2020-05-19:
> * [ ]  update dependencies (django) if we stick with clusterUI for the cluster of one case.

Having merged PR #182, is this now satisfied? What other dependencies do we need to look at _before_ moving on from launchWorkers?

> * [ ] possibly revisit UI (clusterUI might not be that appropriate for most cluster or one use cases) - this could potentially be deferred to step 2

I would like to see this removed from the list of things to do before shifting from launchWorkers. launchWorkers has only a shred of UI (taskMonP) which is more than satisfactorily covered by the clusterUI status/analysis tasks pages. Sure, there's room for improvement, but I think we should leave clusterUI improvements and potential alternate UIs for separate issues downstream. Additionally, splitting the UI for cluster / cluster of one would seem to re-open us up to maintenance/training overheads that we could otherwise finally quash if we get off launchWorkers.

David-Baddeley, 2020-05-19:
@barentine - I'm more and more of the opinion that clusterUI should not be the UI for the single computer case. In my testing so far it's often been the failure point. It's really slow to start up, the performance hit due to it's polling of the dataservers is likely to be significant in the single computer scenario and I think we'll always be chasing our tails with django versions - I think the overhead is worth it for the cluster use case, but would prefer something much lighter weight for ClusterOfOne. As it stands `PYMERuleServer` is already an HTTP server, and already has a status endpoint (albeit in json). We might be able to add a really simple landing page to the `RuleServer` and use that as the UI.

David-Baddeley, 2020-05-19:
@zach - any further luck with your regression? For me the dataserver unit tests pass when you run them in isolation, but fail when run as part of the larger test suite (if other tests using the dataserver have run before them). This suggests the error is related to the server discovery and that you are somehow looking at old records for servers which have been closed previously but are somehow not purged.

zacsimile, 2020-05-19:
I was waiting to see what headway you made with the last error I sent. I can restart my machine and give it a go.

zacsimile, 2020-05-20:
OK, I launched `pymeclusterofone`, then `pymeacquire`, then I generated fluorophores, hit go, turned on the 488 laser, hit start spooling to cluster, started spooling, hit analyze to do it live, and I got the following traceback from the server. I also tried hitting analyze after hitting stop, and I got the same error.

```
Exception happened during processing of request from ('127.0.0.1', 52944)
Traceback (most recent call last):
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\socketserver.py", line 651, in process_request_thread
    self.finish_request(request, client_address)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\socketserver.py", line 361, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\socketserver.py", line 721, in __init__
    self.handle()
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\http\server.py", line 418, in handle
    self.handle_one_request()
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\http\server.py", line 406, in handle_one_request
    method()
  File "c:\users\zrc4\code\python-microscopy\PYME\cluster\HTTPDataServer.py", line 482, in do_GET
    f = self.send_head()
  File "c:\users\zrc4\code\python-microscopy\PYME\cluster\HTTPDataServer.py", line 569, in send_head
    return self.get_h5_part(path)
  File "c:\users\zrc4\code\python-microscopy\PYME\cluster\HTTPDataServer.py", line 669, in get_h5_part
    f, length = self._string_to_file(h5f.get_file(part))
  File "c:\users\zrc4\code\python-microscopy\PYME\IO\h5File.py", line 123, in get_file
    return self.get_frame(frame_num)
  File "c:\users\zrc4\code\python-microscopy\PYME\IO\h5File.py", line 99, in get_frame
    data = self.image_data[idx]
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\tables\vlarray.py", line 671, in __getitem__
    return self.read(start, stop, step)[0]
  File "C:\Users\zrc4\AppData\Local\Continuum\miniconda3\envs\pyme\lib\site-packages\tables\vlarray.py", line 811, in read
    listarr = self._read_array(start, stop, step)
  File "tables\hdf5extension.pyx", line 2106, in tables.hdf5extension.VLArray._read_array
ValueError: cannot set WRITEABLE flag to True of this array
----------------------------------------
```

zacsimile, 2020-05-20:
Whoops. OK, I tried spooling to file and got no errors following that approach. However, no analysis window launched, and no analysis tasks are listed on the server.

zacsimile, 2020-05-20:
Now every time I restart the server and pymeacquire, and try to spool, it tells me, "A file with that name exists," no matter what I change the file name to. Then it starts spooling, but I don't have control of the stop or analyze buttons, so I have to Ctrl+C. Restarting to see if it fixes this.

zacsimile, 2020-05-20:
Aside: it would be super helpful if we put that VisGUI context manager on PYMEAcquire as well.

zacsimile, 2020-05-20:
OK, restarted, ran through everything again (spooling to file). This time, the Stop button worked, the Analyze button did not work, and I received the same `ConnectionResetError` I received last time.

barentine, 2020-05-20:
> @barentine - We might be able to add a really simple landing page to the `RuleServer` and use that as the UI.

This seems reasonable. Maybe I'll benchmark with and without clusterUI.

zacsimile, 2020-05-20:
For kicks, I restarted one more time. I spooled to File (which is what you recommended in specifics for point 2, but is spool to cluster something we want to test? If so, I already did and got that writeable flag error.) Again, no luck clicking analyze from PYMEAcquire. I launched dh5view separately after stopping spooling and closing PYMEAcquire. I opened the file, hit Go on the analyser, and received the following traceback.

```
Traceback (most recent call last):
  File "c:\users\zrc4\code\python-microscopy\PYME\DSView\modules\LMAnalysis.py", line 277, in <lambda>
    bTestF.Bind(wx.EVT_BUTTON, lambda e : self.lmanal.OnTestFrame(e))
  File "c:\users\zrc4\code\python-microscopy\PYME\DSView\modules\LMAnalysis.py", line 628, in OnTestFrame
    ft, fr = self.testFrame()
  File "c:\users\zrc4\code\python-microscopy\PYME\DSView\modules\LMAnalysis.py", line 879, in testFrame
    res = ft(gui=gui,taskQueue=self.tq)
  File "c:\users\zrc4\code\python-microscopy\PYME\localization\remFitBuf.py", line 533, in __call__
    self.res[i] = fitFac.FromPoint(p.x, p.y)
  File "c:\users\zrc4\code\python-microscopy\PYME\localization\FitFactories\LatGaussFitFR.py", line 136, in FromPoint
    X, Y, data, background, sigma, xslice, yslice, zslice = self.getROIAtPoint(x, y, z, roiHalfSize, axialHalfSize)
  File "c:\users\zrc4\code\python-microscopy\PYME\localization\FitFactories\FFBase.py", line 86, in getROIAtPoint
    xslice, yslice, zslice, dataROI, sigma, bgROI = self._get_roi(x, y, z, roiHalfSize, axialHalfSize)
  File "c:\users\zrc4\code\python-microscopy\PYME\localization\FitFactories\FFBase.py", line 64, in _get_roi
    sigma = self.noiseSigma[xslice, yslice, zslice] if self.noiseSigma else None
ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()
Traceback (most recent call last):
  File "c:\users\zrc4\code\python-microscopy\PYME\DSView\modules\LMAnalysis.py", line 255, in <lambda>
    self.bGo.Bind(wx.EVT_BUTTON, lambda e : self.lmanal.OnGo(e))
  File "c:\users\zrc4\code\python-microscopy\PYME\DSView\modules\LMAnalysis.py", line 642, in OnGo
    self.analysisController.pushImages(self.image)
  File "c:\users\zrc4\code\python-microscopy\PYME\DSView\modules\LMAnalysis.py", line 301, in pushImages
    self._checkTQ()
  File "c:\users\zrc4\code\python-microscopy\PYME\DSView\modules\LMAnalysis.py", line 406, in _checkTQ
    import Pyro.core
ModuleNotFoundError: No module named 'Pyro'
```

barentine, 2020-05-20:
Thanks for the testing, @zacsimile. Maybe we have a quick zoom tomorrow?

For reference, I run PYMEAcquire, PYMEClusterOfOne, spooling to cluster works fine and opens dh5view on stop. Spooling to cluster and hitting analysis while spooling fails though - looks like a potential race condition or straight up bug where we aren't expecting the task list to keep growing
```
'http://127.0.0.1:60121/Andrew/2020_5_19/analysis/19_5_series_B.h5r.json'
DEBUG:root:Successfully created rule

DEBUG:root:task pusher poll loop started

DEBUG:root:Queue created

DEBUG:root:numTotalFrames: 3650, currentFrameNum: 10

DEBUG:root:we have unpublished frames - push them

WARNING:urllib3.connectionpool:Failed to parse headers (url=http://127.0.0.1:60123/release_rule_tasks?ruleID=000000-e018798bfdb74c6f93fc648e50fb176a&release_start=10&release_end=3649): [FirstHeaderLineIsContinuationDefect(' Traceback (most recent call last):\n',), FirstHeaderLineIsContinuationDefect('  File "/Users/Andrew/code/python-microscopy/PYME/util/webframework.py", line 285, in _process_request\n',), FirstHeaderLineIsContinuationDefect('    resp = handler(**kwargs)\n',), FirstHeaderLineIsContinuationDefect('  File "/Users/Andrew/code/python-microscopy/PYME/cluster/ruleserver.py", line 631, in release_rule_tasks\n',), FirstHeaderLineIsContinuationDefect('    rule.make_range_available(int(release_start), int(release_end))\n',), FirstHeaderLineIsContinuationDefect('  File "/Users/Andrew/code/python-microscopy/PYME/cluster/ruleserver.py", line 203, in make_range_available\n',), FirstHeaderLineIsContinuationDefect("    raise RuntimeError('Range (%d, %d) invalid with maxTasks=%d' % (start, end, self._task_info.size))\n",)], unparsed data: ''
Traceback (most recent call last):
  File "/Users/Andrew/miniconda3/lib/python3.6/site-packages/urllib3/connectionpool.py", line 441, in _make_request
    assert_header_parsing(httplib_response.msg)
  File "/Users/Andrew/miniconda3/lib/python3.6/site-packages/urllib3/util/response.py", line 71, in assert_header_parsing
    raise HeaderParsingError(defects=defects, unparsed_data=unparsed_data)
urllib3.exceptions.HeaderParsingError: [FirstHeaderLineIsContinuationDefect(' Traceback (most recent call last):\n',), FirstHeaderLineIsContinuationDefect('  File "/Users/Andrew/code/python-microscopy/PYME/util/webframework.py", line 285, in _process_request\n',), FirstHeaderLineIsContinuationDefect('    resp = handler(**kwargs)\n',), FirstHeaderLineIsContinuationDefect('  File "/Users/Andrew/code/python-microscopy/PYME/cluster/ruleserver.py", line 631, in release_rule_tasks\n',), FirstHeaderLineIsContinuationDefect('    rule.make_range_available(int(release_start), int(release_end))\n',), FirstHeaderLineIsContinuationDefect('  File "/Users/Andrew/code/python-microscopy/PYME/cluster/ruleserver.py", line 203, in make_range_available\n',), FirstHeaderLineIsContinuationDefect("    raise RuntimeError('Range (%d, %d) invalid with maxTasks=%d' % (start, end, self._task_info.size))\n",)], unparsed data: ''
ERROR:root:Failed on posting tasks with status code: 500
```

And the following in PYMEClusterOfOne terminal
```
DEBUG:nodeserver:Found 0 local tasks
DEBUG:nodeserver:Could not find enough local tasks, bidding on non-local
127.0.0.1 - - [19/May/2020 20:26:48] code 500, message Server Error
 Traceback (most recent call last):
  File "/Users/Andrew/code/python-microscopy/PYME/util/webframework.py", line 285, in _process_request
    resp = handler(**kwargs)
  File "/Users/Andrew/code/python-microscopy/PYME/cluster/ruleserver.py", line 631, in release_rule_tasks
    rule.make_range_available(int(release_start), int(release_end))
  File "/Users/Andrew/code/python-microscopy/PYME/cluster/ruleserver.py", line 203, in make_range_available
    raise RuntimeError('Range (%d, %d) invalid with maxTasks=%d' % (start, end, self._task_info.size))
RuntimeError: Range (10, 3649) invalid with maxTasks=1615

DEBUG:nodeserver:Found 0 local tasks
DEBUG:nodeserver:Could not find enough local tasks, bidding on non-local

```

David-Baddeley, 2020-05-20:
@zacsimile 
- The writable flag problem is caused by incompatible versions of pytables and numpy (i.e. a numpy version which is too high for the pytables version you are using). How are you installing packages (pip or conda), if conda, are you specifying both numpy= and numpy-base=? 
- The context manager is already in PYMEAcquire, but (as with VisGUI and dh5view) currently only used for actions which are started through the menus.
- spooling to file and then running the analysis will try and use `launchWorkers` by default which requires Pyro. Until we have ClusterOfOne working seemlessly that is the best / most compatible option (it's also the only option that will let you analyse files which are not in your cluster root), although we could and/or should test for py3 and default to the cluster code in that case.

@barentine 
- that's definitely a bug - we should probably be setting maxTasks higher when creating the queue if the original data file is not finalised.

zacsimile, 2020-05-20:
@David-Baddeley OK, so we want to test spooling to cluster. What version of pytables/numpy do you recommend? This did work with my version of pytables/numpy before (see a few comments back).

barentine, 2020-05-20:
If the spoolframe finds a cluster it should select spool to cluster automatically (i.e. if you run PYMEClusterOfOne first). 

The testing I just did was (apparently) numpy and numpy-base 1.18 and pytables 3.4.4, though as I've mentioned in issue #215  numpy 1.16 and pytables 3.5.2 works well and might be something we are comfortable with in the nearer future. I'd thought that writable flag was just present on pytables <3.5.2 but maybe you can get away with lower pytables with the latest numpy builds. I don't know what pytables works well with 1.14 - I am (usually) running 1.16 these days.

barentine, 2020-05-20:
Sorry for the non-answer - @zacsimile maybe try pytables 3.4.4? What version were you using earlier, 3.4.2?

zacsimile, 2020-05-20:
I already tried `numpy=1.14` and it prevented from launching `PYMEAcquire`. Something about `numpy.ufunc` backward compatibility problems. I tried `numpy=1.16` and `pytables=3.4.2` and got the `WRITEABLE` flag error. I'm trying to get `pytables=3.5.2` back, but the cluster keeps saying I'm spooling to files that already exist when I'm not.

zacsimile, 2020-05-20:
Yeah, I can't get anything to spool. It just says, "A file with that name already exists," starts spooling, because, of course, that file does not already exist, and locks everything. I've restarted 3 times. Maybe when the date changes I'll be able to do the next test.

David-Baddeley, 2020-05-20:
numpy=1.14.6 and pytables=3.4.4 works on my end. Suspect that the ufunc stuff is related to something like skimage or sklearn being at a really recent version. It's probably safest to try the different versions in a clean environment so that you get a consistent install (i.e. if you do a `conda env create` with numpy pinned it shouldn't install incompatible versions of some of the other libraries. If you have a mixed conda and pip environment all bets are off (to some extent even with a pure pip environment unless it was all installed with the one call to pip).

David-Baddeley, 2020-05-20:
I'm pretty sure pip doesn't check if it's going to break existing packages - ie if you have sklearn or scipy depending on numpy >=1.16 and you pip install numpy=1.14 it will go ahead and install anyway (I think it just checks what dependencies numpy has, not what depends on numpy). This mostly works if you are always increasing package versions (and packages are mostly backwards compatible), but can get really hairy when downgrading.

David-Baddeley, 2020-05-20:
but in short, numpy=1.14 and whatever version of pytables you need to work with that.

[26 more replies]

closed 2021-04-12 by David-Baddeley