python-microscopy/python-microscopy #164: Ability to turn off auto-apply of settings in visgui

lukasfue, 2020-02-27:
Adjusting things like alpha, point-sizes, point method, LUT, etc in visgui auto-applies any change in the setting. This becomes an issue when I have several millions of localizations and I want to adjust something like the alpha from 1.0 to 0.25. Visgui will automatically adjust the alpha to 0 when I type "0", then to 0.2 when I add the ".2", then finally 0.25 when I add the "5". With millions of points, it takes a significant amount of time to adjust the alpha, and in this example it will adjust it 3 times instead of just the 1 time I need it too at the final value of 0.25. It would be nice if the default was not to auto-apply adjustments and instead required you to press enter or a "go" button. Maybe this could be a setting one can toggle on/off.

David-Baddeley, 2020-02-27:
How many million points, which display mode, and what type of computer? Most of those changes should be pretty fast even with several million points (even on a 5 yr old mac with integrated intel graphics).

That said, I agree that the incremental updates when you are typing a float are annoying, It's unfortunately really hard to change because the behaviour is hard wired into traits / traitsui. I don't really want to add a "go" button if it can be avoided. One solution would be to change things so that they updated when the field lost focus, rather than on character entry, but I'm not sure how easy that would be to implement on top of traits. 

Another option option would be to limit the refresh rate - use the parameter change even to set a flag that things were stale, and then keep postponing the update while there had been a parameter change within the last, e.g. 0.5s. This would be really hacky to implement (and might well be a source of future bugs).

David-Baddeley, 2020-02-27:
Was actually **way** easier than expected to get traits to do something sensible.

You now need to either hit enter after typing the number or click away from the field to get the value to update. Applies to both alpha and point size.

David-Baddeley, 2020-02-27:
NB - in current repo head - hopefully this is good enough.

lukasfue, 2020-02-28:
David,

Thank you for the update!

To answer your questions, I was viewing about 16 million points using point
sprites. I was using a PC with Windows 10 that has a 5th generation i7 CPU,
64 GB of RAM, and a GTX 980 Ti GPU.

Best,
Lukas

closed 2020-03-12 by David-Baddeley