python-microscopy/python-microscopy #170: Gui manipulations take much longer (>10x) for color-channel outputs than for other outputs

lukasfue, 2020-03-04:
If I load data with 20 million points, changes to Alpha, point-size, LUT, etc. take about 6 seconds to complete. However, once I assign colors to channels (salvaged fluorescence color assignment) and display each color output in a separate layer, those same changes (alpha, point-size, etc) now take about 1 minute and 15 seconds. Each layer consists of less points than the total 20 million, so it seems strange that it takes more than 10x as long for these manipulations. I am happy to provide more info if it would be helpful.

David-Baddeley, 2020-03-04:
It looks like the time is spent getting the points, not their display.

The colour mapping is done on the fly, involves some calculation, and is not cached - it's therefore not super suprising that it takes a bit longer than non-colour mapped display. As it stands, the mapping computes a binary index for a given channel over all localizations. It re-computes this for each coordinate (x, y, z) etc .. access. The co-ordinates are recomputed every time you change alpha / point size etc ... there is therefore quite a lot of room for optimization, by either caching point positions within the layer so we don't need to update them when we change, e.g. alpha, or by caching the index in the colour filter. That said, I'm still surprised how slow it is - I've not run into update times of more that 1-2s in the past (which is why these optimisations haven't been done).

David-Baddeley, 2020-03-04:
It would be great if you could send be an example dataset - most of my examples only have 1-2M points.

lukasfue, 2020-03-04:
That's good news that there is room for optimization. I just noticed that it only takes this long for the original H5R file that the microscope PC saves the data to. If I assign color channels and save those localizations as an HDF and open that file in a new visgui window, it once again only takes ~6 seconds for each manipulation. I am happy to share a dataset with you, but the files are a couple GBs. Do you have access to the Bewersdorf lab-share? If so, I've uploaded both an H5R and an HDF file from the same dataset at the following file path: Z:\Lukas\Stuff_for_Zach\Cell7_Rnt4-Em-AF647_mCh-Sec61B-CF660C-ST...
If you don't have access to the B-lab labshare, I can find another way to share the data with you.

David-Baddeley, 2020-03-04:
No easy access from here to the Bewersdorf lab share, but I can probably do some profiling with the largest file I have. The fact that saved filtered version is that much faster points fairly clearly to where the computational cost is incurred (and to what is hopefully a reasonably simple, if slightly dirty, solution).

David-Baddeley, 2020-03-04:
Found a 10M localization ratiometric series here which should be good enough for testing (fingers crossed).

barentine, 2020-03-04:
I'll send you a google-drive link, David.

David-Baddeley, 2020-03-12:
Think we can safely close this - there is still some room for further optimisation, but I'm not sure it's warranted at this stage - performance seems to be similar for ratiometric and non-ratiometric data.

closed 2020-03-12 by David-Baddeley