python-microscopy/python-microscopy #174: cluster of one dataserver filter

barentine, 2020-03-06:
At the moment the dataserver filter used by the instance of clusterUI launched by `PYMEClusterOfOne` still respects the PYMEConfig `'dataserver-filter'`.

Is this the desired behavior? This makes it possible for the cluster-of-one launched clusterUI to browse the wrong cluster

David-Baddeley, 2020-03-10:
This is the desired behaviour, unless we can think of a compelling use case where this would be undesirable. The default for `dataserver-filter` is the machine name, so we should not have any issues with finding the wrong cluster. 

There is an inherent assumption here that a machine will not be both a cluster of one and simultaneously part of a larger cluster, but I think that is, for the most part, reasonable. 

I suspect you have run into this with testing the cluster of one on your desktop, which you have also set up to be part of the cluster. Can you think of any scenario (other than testing) in which you might want to run ClusterOfOne on a computer that is part of a proper cluster? Do you need `dataserver-filter` to be set on your pc in the first place? - you can still access data on the cluster by supplying the filter in the URL to e.g. dh5view.

Making everything use a different `dataserver-filter` when running cluster of one would be really hard (needing to somehow make PYMEAcquire etc ... use a different config mechanism when you were running clusterof one)

barentine, 2020-03-11:
As usual the use-cases I think of most are my own, which as you mention includes a fair amount of testing. From that perspective I would disagree that this is the desired behavior, but agree it'd be surprisingly painful to change.

I'll unset the filter on my desktop and keep this all in mind for the cluster of one docs. Thanks

closed 2020-03-11 by barentine