python-microscopy/python-microscopy #180: text trimming a bit odd in LMAnalysis panel

barentine, 2020-03-10:
ubuntu 16.04 LTS, wxpython 4.0.4, python 3.6.9, traits 6.1.0, traitsui 6.2.0

Text can be trimmed in an awkward way such maps which should be found automatically from metadata may appear to be blank
[image]
Though one can check by manually expanding the panel.
[image]
There's also an odd in-between spot where you get just a piece of the underscore for some reason
[image]

If you use the 'Set' button, then it tries to display the map name regardless of panel size:
[image]
[image]

David-Baddeley, 2020-03-10:
Not sure there is an easy fix to this - it's somewhat offset for practical use by the colour-coding so that you at least know if there is a map set (and can expand the panel to see the filename).

barentine, 2020-03-11:
Figured it wouldn't be worth the time investment to sort out but thought I'd raise it just in case. Thanks David.

closed 2022-12-14 by David-Baddeley