python-microscopy/python-microscopy #187: Memory allocation issue in pairwise distance histogram calculation on Windows 10, Python 3

zacsimile, 2020-03-12:
I am unable to run pairwise distance histogram calculations from VisGUI on a relatively small file (4 million localizations) in Windows 10, Python 3. I can calculate a pairwise distance histogram on the same file on my Mac just fine. Traceback below.

```
Traceback (most recent call last):                                                                                        
    File "c:\users\zrc4\code\python-microscopy\PYME\LMVis\Extras\clusterAnalysis.py", line 281, in OnPairwiseDistanceHistogram                                                                                                                 
        distances = distogram.execute()                                                                                       
    File "c:\users\zrc4\code\python-microscopy\PYME\recipes\base.py", line 519, in execute                                    
        m.execute(self.namespace)                                                                                             
    File "c:\users\zrc4\code\python-microscopy\PYME\recipes\measurement.py", line 399, in execute                             
        res = DistHist.distanceHistogram(pos0['x'], pos0['y'], pos1['x'], pos1['y'], self.nbins, self.binSize)              
RuntimeError: Error allocating output array
```

zacsimile, 2020-03-12:
Downgraded from numpy=1.18.1 to numpy=1.16.5 and it works again.

zacsimile, 2020-03-12:
Upgraded to numpy=1.17.0 and that also works. So numpy=1.18.1 is the problem.

David-Baddeley, 2020-03-12:
That's pretty weird - windows is pretty bad at memory allocation (you'll sometimes get allocation issues due to the lack of contiguous blocks when allocating a large array even if it seems like there should be enough memory for it), but the array being allocated here is small. 

What version of numpy is PYME compiled against in this case?

I suspect the issue is that this code is old, and uses `PyArray_FromDims()` to allocate the array, but this is on the deprecation path. Replacing with `PyArray_SimpleNew()` would probably do the trick.

Just looked at the numpy 1.18 release notes (https://numpy.org/devdocs/release/1.18.0-notes.html), and they would seem to confirm this - look under expired deprecations.

David-Baddeley, 2020-03-12:
Looks like this will effect a few other places too ...

```bash
DB3:python-microscopy david$ grep -n -r PyArray_FromDims .
./PYME/Analysis/points/astigmatism/astiglookup.c:57:    out_z = (PyArrayObject*) PyArray_FromDims(1,outDimensions,PyArray_INT);
./PYME/Analysis/points/astigmatism/astiglookup.c:64:    out_err = (PyArrayObject*) PyArray_FromDims(1,outDimensions,PyArray_FLOAT);
./PYME/Analysis/points/astigmatism/astiglookup.c:171:    out_z = (PyArrayObject*) PyArray_FromDims(1,outDimensions,PyArray_INT);
./PYME/Analysis/points/astigmatism/astiglookup.c:178:    out_err = (PyArrayObject*) PyArray_FromDims(1,outDimensions,PyArray_FLOAT);
./PYME/Analysis/points/DistHist/distHist.c:127:    out = (PyArrayObject*) PyArray_FromDims(1,outDimensions,PyArray_DOUBLE);
./PYME/Analysis/points/DistHist/distHist.c:303:    out = (PyArrayObject*) PyArray_FromDims(1,outDimensions,PyArray_DOUBLE);
./PYME/Analysis/points/DistHist/distHist.c:768:    out = (PyArrayObject*) PyArray_FromDims(1,outDimensions,PyArray_DOUBLE);
./PYME/Analysis/PSFGen/ps_app.c:162:    //out = (PyArrayObject*) PyArray_FromDims(3,size,PyArray_DOUBLE);
./PYME/Analysis/PSFGen/ps_app.c:405:    //out = (PyArrayObject*) PyArray_FromDims(3,size,PyArray_DOUBLE);
./PYME/Deconv/cpp_afunc/fft_test.c:103:    result = (PyArrayObject *)PyArray_FromDims(1, f->dimensions, PyArray_FLOAT); 
./PYME/Deconv/cpp_afunc/fft_test.c:200:    result = (PyArrayObject *)PyArray_FromDims(1, f->dimensions, PyArray_FLOAT); 
./PYME/localization/cModels/gauss_ap.c:82:    //out = (PyArrayObject*) PyArray_FromDims(2,size,NPY_DOUBLE);
./PYME/localization/cModels/gauss_ap.c:195:    //out = (PyArrayObject*) PyArray_FromDims(2,size,NPY_DOUBLE);
./PYME/localization/cModels/gauss_ap.c:329:    //out = (PyArrayObject*) PyArray_FromDims(2,size,NPY_DOUBLE);
./PYME/localization/cModels/gauss_ap.c:494:    //out = (PyArrayObject*) PyArray_FromDims(2,size,NPY_DOUBLE);
./PYME/localization/cModels/gauss_ap.c:628:    //out = (PyArrayObject*) PyArray_FromDims(2,size,NPY_DOUBLE);
./PYME/localization/cModels/gauss_ap.c:813:    //out = (PyArrayObject*) PyArray_FromDims(2,size,NPY_DOUBLE);
./PYME/localization/cModels/gauss_ap.c:1049:    //out = (PyArrayObject*) PyArray_FromDims(2,size,NPY_DOUBLE);
./PYME/localization/cModels/gauss_ap.c:1187:    out = (PyArrayObject*) PyArray_FromDims(3,size,NPY_DOUBLE);
./PYME/localization/cModels/gauss_ap.c:1294:    out = (PyArrayObject*) PyArray_FromDims(2,size,NPY_DOUBLE);
./PYME/localization/cModels/gauss_ap.c:1396:    out = (PyArrayObject*) PyArray_FromDims(3,size,NPY_DOUBLE);
./PYME/localization/cModels/gauss_ap.c:1510:    out = (PyArrayObject*) PyArray_FromDims(3,size,NPY_DOUBLE);
./PYME/localization/cModels/gauss_ap.c:1644:    out = (PyArrayObject*) PyArray_FromDims(3,size,NPY_DOUBLE);
./PYME/localization/cModels/gauss_ap.c:1761:    out = (PyArrayObject*) PyArray_FromDims(2,size,NPY_DOUBLE);
./PYME/localization/cModels/gauss_ap.c:1861:    out = (PyArrayObject*) PyArray_FromDims(2,size,NPY_DOUBLE);
./PYME/localization/lev_gfit/levmar_gfit.c:186:    outp = (PyArrayObject*) PyArray_FromDims(1,&m,PyArray_DOUBLE);
./PYME/localization/lev_gfit/levmar_gfit.c:295:    outcov = (PyArrayObject*) PyArray_FromDims(2,covSize,PyArray_DOUBLE);
```

David-Baddeley, 2020-03-12:
Looking at the commented out ones, it looks like I've tried `PyArray_SimpleNew()`, failed to get that to work, and then fallen back on `PyArray_New()`. Might need some experimentation ... `PyArray_SimpleNew()` is probably preferable if you can get it to work as it's cleaner and more readable.

zacsimile, 2020-03-12:
I'm looking into it now. Running into a different problem with windows compiler that I have sort out first :p. Will report back shortly.

zacsimile, 2020-03-12:
OK, swapping all `PyArray_FromDims()` for `PyArray_SimpleNew()` yields the same memory allocation error problem.

zacsimile, 2020-03-12:
OK, swapping all `PyArray_FromDims()` for `PyArray_SimpleNew()` and making sure the  `dims` passed to `PyArray_SimpleNew()` are all of type `npy_intp` instead of just `int` seems to fix it.

zacsimile, 2020-03-12:
I'm gonna go ahead and fix all of these and add it to that PR. Hold on.

zacsimile, 2020-03-12:
Fixed. See #188.

closed 2020-03-12 by zacsimile