python-microscopy/python-microscopy #192: gui/pipeline/recipes interfaces for splitter and color filter

barentine, 2020-03-19:
I'll preface this with the fact that perhaps some of my wishlist is implemented and I'm just not finding it, but regardless:
I am trying to help someone who wants to run analysis on separate channels in VisGUI, and also uses the splitter stuff, setting their ratios manually in the colourPanel. Would be great if they had that info in the metadata, and we’ll get there, but in general if we want the color filter stuff to be used rather than hacked around with table filters, I think it might be nice:
- if the upstream color filter could be set with the GUI, i.e. linked to the `Colour` panel.
- if the downstream color filter which sits at the end of the pipeline was _not_ linked to the `Colour` panel because it was actually just a color selector since there is now one present upstream by default (would need to watch out for it being deleted in the pipeline recipe panel or so)
- if said color selector was implemented as a separate recipe module, localizations.SelectColors
- if localizations.SelectColors had a CheckListEditor in the view which was populated using a property of `recipes.base.ModuleBase` similar to `_namespace_keys`.

David-Baddeley, 2020-03-20:
First two mostly describe the status quo. The ratio metric colour panel
adjusts the settings of the upstream (1st module on recipe) ProcessColour
module. With recent (2 wks ago) changes its a little more robust to changes
in the recipe, in that the link shouldn't break when you load recipes etc
(it takes the first ProcessColour module in the active recipe). There is a
slight wrinkle here, in that it also sets the cutoff probabilities used in
ColourFilter (ie it talks to both bits) - ProcessColour computes the
probability of a given localization being a particular colour and
ColourFilter thresholds the probabilities to just give you one colour.

As to point 3, I thought there was a recipe module for the ColourFilter (ie
selection) but that might not be the case - will double check when in front
of my laptop.

Re pt 4 - this sounds like you want to select multiple colours with the
colour filter. What is the use case here? This would require significant
re-working of ColourFilter and some slightly complicated design decisions -
e.g. if you chose both channels A and B which are next to each other in
ratio space, do you also include the localisations in between (which could
not unambiguously be assigned to either A or B and would therefore be
rejected when selecting a single colour).

David-Baddeley, 2020-03-27:
re pt 3 - the `localizations.ExtractTableChannel` module does this (admittedly the most obvious of names, but  named to match `ExtractChannel` as used for intensity images)

barentine, 2020-03-28:
Thanks for your replies, David! And sorry I missed the extraction module. How open would you be to a PR registering `localisations.ExtractTableChannel` as a legacy module by the same name with a register_moduleas `ExtractColor`? I appreciate trying to keep things consistent with intensity images, but in this case I think it'd be easier to find if it was consistent with `ProcessColour`. Modulo the spelling, because I'm biased and it's already hurts to type localisations ;) 

re: part 1-2, I'm glad to hear it's already linked. What might take this the last couple % polished would be if traits in the pipeline recipe would update on changes in the color panel (for instance, clicking 'guess' in the panel would auto deselect 'from metadata' and populate gFrac, etc. in the recipe display). Any thoughts on how desirable that would be?
[image]

David-Baddeley, 2020-03-28:
Just checked and things aren't updating as I'd expect - will see what I can
do to fix it - interestingly the updates are better in the 'Pipeline
Recipe' tab than on the vertical display - not quite sure why as they
should be synced. As it stands, ratios from metadata is not unset when you
change the ratios manually (which it should be). I'll fix that first and
then look at the syncing.

Slightly torn between deprecating one name, and keeping both active (as
aliases). Let me think a bit.

David-Baddeley, 2020-03-28:
GUI updating should now work (was a bit messy in that changing the ratios propagated the changes to the ProcessColour module, but didn't trigger the event needed to update the GUI). Should be fixed now. We now also turn off ratios_from_metadata if you change anything in the colour panel.

barentine, 2020-03-29:
Awesome - thank you for looking into it, and the fix @David-Baddeley!

closed 2020-04-19 by David-Baddeley