python-microscopy/python-microscopy #194: Make MATLAB loading CSV-style

zacsimile, 2020-03-20:
Convert the MATLAB loader so it has a single entry point and pops up with a GUI for mapping MATLAB workspace fields to PYME dataSource fields, just like we do when loading CSV files.

zacsimile, 2020-05-06:
First off, correcting myself, the MATLAB loader does have a single entry point.

Follow up on personal communications with @David-Baddeley:

We decided to have MATLAB automatically load without a GUI popup if it can find `x` and `y` variables. If there are only a couple of variables, and either `x` or `y` cannot be found, it will default to old-style MATLAB loading. Otherwise, if there are a lot of variables, and either `x` or `y` cannot be found, a CSV-style GUI variable mapping window will pop up.

zacsimile, 2020-05-07:
Addressed in #229

David-Baddeley, 2020-06-26:
now merged

closed 2020-06-26 by David-Baddeley