python-microscopy/python-microscopy #195: Splitter fitter shift interactions with image bounds

zacsimile, 2020-03-20:
Splitter fitters shift one of the channels by

```
DeltaX = self.metadata.chroma.dx.ev(x_, y_)
DeltaY = self.metadata.chroma.dy.ev(x_, y_)

dxp = int(DeltaX/vx)
dyp = int(DeltaY/vy)

Xr = Xg + DeltaX - vx*dxp
Yr = Yg + DeltaY - vy*dyp
```

(see PYME.localization.FitFactors.FFBase).

When we go to get the image bounds (`PYME.LMVis.pipeline.imageBounds`) for renderings (or anything else, I just noticed it in the renderings), assuming we have metadata `Camera.ROIWidth` (which we usually do), we grab the image bounds as

```
if 'Splitter' in mdh.getOrDefault('Analysis.FitModule', ''):
            if 'Splitter.Channel0ROI' in mdh.getEntryNames():
                rx0, ry0, rw, rh = mdh['Splitter.Channel0ROI']
                x1 = rw * 1e3 * mdh['voxelsize.x']
                x1 = rh * 1e3 * mdh['voxelsize.x']
            else:
                y1 = y1 / 2
```

in `PYME.IO.image.ImageBounds.extractFromMetadata()`. However, this does not account for the shift. (There are another two problems, I think, which are that we set `x1` twice, and these days we are usually splitting channels in the x-direction, so the `else` statement should contain `x1=x1/2`). 

This is hacked around in `PYME.LMVis.renderers.CurrentRenderer._get_image_bounds()` using

```
x0 = max(self.visFr.glCanvas.xmin, self.pipeline.imageBounds.x0)
y0 = max(self.visFr.glCanvas.ymin, self.pipeline.imageBounds.y0)
x1 = min(self.visFr.glCanvas.xmax, self.pipeline.imageBounds.x1)
y1 = min(self.visFr.glCanvas.ymax, self.pipeline.imageBounds.y1)
```

which is fine as long as we're generating renderings from the GUI. If we generate the renderings in any other way, e.g. from a Jupyter Notebook, the rendering is improperly shifted.

I think we should fix the image bounds calculation upstream in `PYME.IO.image.ImageBounds.extractFromMetadata()`. An alternative solution (though I believe still a hack) would be to check `pipeline.keys()` in `PYME.LMVis.renderers.CurrentRenderer._get_image_bounds()` instead of `pipeline.filterKeys()` on lines 114 and 118.

zacsimile, 2020-03-20:
This is potentially related to this older PR: https://bitbucket.org/david_baddeley/python-microscopy/pull-requests/223?atlOrigin=eyJpIjoiMWRiZjlmZjhkYmE3NDg0Mzk3NWI3ODZhZjczNGQyODQiLCJwIjoiYmItY2hhdHMtaW50ZWdyYXRpb24ifQ

David-Baddeley, 2020-03-20:
In general, I'm not sure that we should/need to be correcting for shift in the image bounds. As it stands, everything will be in the reference frame of Channel0 (i.e. corrected) - although for large shifts you might get some clipping of Channel1 (but as this corresponds to areas where Chan0 is not present this doesn't seem unreasonable). This is, I believe, the correct thing to do. What makes you believe that the rendering is improperly shifted?

the `_get_image_bounds()` method is not really relevant for pipelined analysis - it is simply saying that you should render what is on screen (or the data bounds if smaller). It has a lot of issues with respect to automated processing as the exact region you crop out is a function of either GUI zoom, or stochastic aspects of the localisation distribution - i.e. if you render 2 consecutive series through the GUI you don't get exactly the same ROI.

As to the smaller points:
- Setting `x1` twice is definitely a bug ... and should be fixed. 
- The `y1 = y1/2` else clause is to provide backwards compatibility for really old data (all new data should be defining `Channel0ROI` and `Channel1ROI`).

David-Baddeley, 2020-03-20:
Summary - renderings done using the recipe module directly are correct, renderings made through VisGUI may be arbitrarily shifted.

barentine, 2020-03-20:
Will renderings in different modes using the recipe module not still be sub-pixel shifted relative to each other, the subject of the old PR?

zacsimile, 2020-03-20:
I did not mean to imply the shifts were incorrect. More, the rendering clips one of the channels, so we lose a bit of data. This came up because I'm trying to create masks from the renderings and then apply these masks to real data and I was trying to figure out the shift between the mask and the data in multicolor images.

What method is used in the recipe? It seems like most everything I've found uses `_get_image_bounds()`. It's what's called by the `Generate()` method in the base class.

David-Baddeley, 2020-03-21:
It uses `_get_image_bounds()`, but returns on line 107 - i.e. the
unmodified imageBounds from the pipeline, which in turn should be the
camera ROI (if ROI metadata is defined).

I had thought that the sub-pixel shifts had been resolved, but maybe not -
I'll have another look at it.

David-Baddeley, 2020-03-21:
I've fixed the repeated x1= in extractFromMetadata(), so this should fix
some of the clipping issues.

That leaves any sub-pixel shifts, although these should be small (max 2.5
nm for a 5nm pixel size) and only (potentially) present if using different
rendering modalities for the two images.

By real data, do you mean the raw data? Probably the easiest thing to do is
to shift-correct the raw data using the same shift map - rendered images
will always be corrected to the first ROI frame of reference, so the raw
data from different ROIs should be definition not overlap (as it's not
corrected). You can do this shift correction with the splitting module in
dh5view.

zacsimile, 2020-03-22:
I think I may explained this poorly. What I'm trying to do is

- Create a rendering from localized data using `GaussianRenderer`
- Create a binary mask of the ER from this rendering (includes some dilation/erosion cleanup)
- Find the localizations that fall in the resulting mask
- Use these localizations for pairwise distance calculations

The second-to-last step has proven to be more difficult than I think is necessary for multicolor images. I think the fact that `_get_image_bounds()` returns on line 107 is what has been creating problems. Localizations in multicolor images are often shifted outside of the image bounds guessed from the metadata. Then, with the `GaussianRenderer`, these points are set to extremum in `PYME.LMVis.visHelpers.rendGauss` by the `argmin` calls. For example, an x-shift that pushes a localization outside of the original estimated bounds forces these points to show up in the bottom-most bin of the rendering, when this isn't really where they are. This is normally fixed when calling `_get_image_bounds()` from VisGUI, where the viewport sets the min/max of the point set based on what's displayed (after line 107). 

We could argue this is not a problem, but it means we have to use expensive `argmin` calls in `PYME.LMVis.visHelpers.rendGauss` and we have to use these calls again if we want to compare localizations to their position in an already-generated mask. I think if we simply expanded the bounds to something like `pipeline.imageBounds.x1 + pipeline.mdh.chroma.dx.ev(0,0)-int(pipeline.mdh.chroma.dx.ev(0,0)/vx)*vx`, etc. (following `PYME.localization.FitFactories.FFBase.getSplitROIAtPoint`), we could eliminate the `argmin` calls and make positions of localizations in the rendered images much more straightforward (`(pos + fuzz)/render_pixel_size`). What do you think?

zacsimile, 2020-03-24:
OK. This (silly way) of setting the bounds eliminates the need for `argmin` calls, gets the bounds right, and doesn't crop any of the shifted or unshifted data. (It also keeps the image square for ease-of-use).

```
minx = np.min(pipeline['x'])
miny = np.min(pipeline['y'])
pipeline.imageBounds.x0, pipeline.imageBounds.y0 = minx, miny
maxval = np.maximum(np.max(pipeline['x']),np.max(pipeline['y']))
maxdiff = np.maximum(maxval-minx, maxval-miny)
pipeline.imageBounds.x1, pipeline.imageBounds.y1 = (minx+maxdiff), (miny+maxdiff)
```

and then `xi =  int(round((x[i]-X[0])/pixelSize))` where `xi, x[i], X` and `pixelSize` are as in `PYME.LMVis.visHelpers.rendGauss`. All of this is easily extended to `yi` and other renderers. We can always crop the resulting image by more than `roiSize` to get back to the `mdh['Camera.ROIWidth']` by `mdh['Camera.ROIHeight']` size.

I think this may be a clearer (everything is binned as expected) and less expensive way of handling the ImageBounds and rendering. What do you think?

David-Baddeley, 2020-03-27:
Hmm, I'm fairly adamant that setting the image bounds to be the same as for the raw data is the only sensible thing to do when generating images automatically - i.e. the dimensions of automatically generated images must not depend on either the localization data itself or the localization precision (variable amount of fuzz). That's the 

There are faster ways of doing what `argmin` is doing in `rendGauss`, but I don't think this is limiting on speed at present - although `rendGauss` could use some optimization, it is tangential to the current discussion.  

Expanding the image bounds  to what you could get with shifts is never going to be sufficient/robust, as mis-fitted localisations can be well past that.

Reading data out from rendered images is pretty simple, and doesn't require any `argmin()`s - see   `PYME.Analysis.points.cluster_morphology.get_labels_from_image` and `PYME.recipes.localisations.LabelsFromImage`. `get_labels_from_image()` currently returns 0 for anything which falls out of bounds (which is appropriate for a labelled image/mask but not necessarily for some of the uses).

zacsimile, 2020-03-28:
I agree with you about setting the image bounds the same size as the raw data. I hadn't considered how far off a misfitted localization could be. I think the way this is handled in `PYME.Analysis.points.cluster_morphology.get_labels_from_image` is close to optimal, although it doesn't look like it accounts for the 2.5 pixel offset (for a 5 nm bin image). Can we write the pixel size into the generated image metadata and account for the offset using that?

David-Baddeley, 2020-03-28:
The half pixel (not 2.5 pixel) offset is something I'm still getting my head around how best to handle in a consistent way without breaking things - it's the difference between referencing to the centre of the 0,0th pixel and to it's top left corner. 

Ultimately I'll have to decide which imageBounds references and fix everything else. In general, you want to reference your grid to the pixel centres, but this doesn't work well across different pixel sizes, meaning that imageBounds should probably reference to the top-left. As it stands, the Gaussian renderer assumes that imageBounds references top left, `labels_form_image` assumes it references to the centre (could be changed by swapping the `np.round()`s on lines 55-57 to `np.floor()`), Histogram rendering references top-left, and I'd have to work out what other things do. 

The pixel size is always written to the metadata in any case.

David-Baddeley, 2020-03-28:
PS - localisations etc ... are referenced to pixel centres.

David-Baddeley, 2020-03-28:
Locations we'd need to address:

- renderers
- labels_from_image
- splitting / shiftmaps (although this might not be critical as if shift varies significantly over 1/2 a pixel all bets are off)
- tiling

David-Baddeley, 2020-03-28:
- simulation

David-Baddeley, 2020-03-28:
I've just pushed a commit which means that renderers and labels_from_image are consistent, which should fix your original workflow (render image, threshold, propagate mask back). These all now assume that things are referenced top-left. 

We still have an issue with localisation, and any of the measurement functions in scipy or skimage (e.g. as used by the Measure2D recipe module and/or conversion from pixels to nm), where we will have an offset of 0.5 of the original (raw) pixels.

zacsimile, 2020-03-29:
That's great! Thanks, David! Will keep an eye out for half-pixel shifts in the measurement functions.

closed 2020-03-29 by zacsimile