python-microscopy/python-microscopy #199: DBSCAN dialog not displaying most fields on Python 3, OSX

zacsimile, 2020-03-29:
[image]

`wxpython 4.04`
`traits 5.1.1`
`traitsui 6.1.1`

```
AttributeError: 'ImageControl' object has no attribute 'GetClientSizeTuple'
Traceback (most recent call last):
  File "/anaconda3/envs/pyme/lib/python3.6/site-packages/traitsui/wx/image_control.py", line 194, in _on_paint
    wdx, wdy = self.GetClientSizeTuple()
```

zacsimile, 2020-03-29:
This is what I get if I try to click in to the hidden fields. The same `AttributeError` just spools on the screen as I do this.

[image]

David-Baddeley, 2020-03-29:
This is a traitsui bug.

potential workarounds:

- downgrade wx to the lowest version you can on Py3
- write a custom view to force the use of a different editor for the trait (see comments on #200 )

closed 2020-04-19 by David-Baddeley