python-microscopy/python-microscopy #208: wx AssertionErrors abound

zacsimile, 2020-04-23:
Often, after closing a wx Widget in PYME, we get roughly the following traceback.

```
wx._core.wxAssertionError: C++ assertion "GetEventHandler() == this" failed at /opt/concourse/worker/volumes/live/b30868b5-2a7c-479c-4e29-27874bbc69a6/volume/wxpython_1547931175613/work/ext/wxWidgets/src/common/wincmn.cpp(478) in ~wxWindowBase(): any pushed event handlers must have been removed

The above exception was the direct cause of the following exception:

SystemError: <class 'wx._core.WindowDestroyEvent'> returned a result with an error set
wx._core.wxAssertionError: C++ assertion "GetEventHandler() == this" failed at /opt/concourse/worker/volumes/live/b30868b5-2a7c-479c-4e29-27874bbc69a6/volume/wxpython_1547931175613/work/ext/wxWidgets/src/common/wincmn.cpp(478) in ~wxWindowBase(): any pushed event handlers must have been removed

The above exception was the direct cause of the following exception:

SystemError: <class 'wx._core.WindowDestroyEvent'> returned a result with an error set
```

As best I can tell, this is a reference counting issue in wx Grids (see https://discuss.wxpython.org/t/howto-deal-with-reference-counts-in-grid-classes/34126, https://discuss.wxpython.org/t/bug-error-with-slight-change-to-gridhugetable-py/34529/4, https://github.com/wxWidgets/Phoenix/issues/627). 

One line where this traceback always gets thrown is line 65 of `PYME.recipes.vertical_recipe_display`: `self.fp.DestroyChildren()`. I think this is the due to the wx `GetChildren()` call (we need to find the children before we can destroy the children) increasing the refcount one more than is necessary (http://trac.wxwidgets.org/ticket/3563). I have tried, but am so far too stupid to figure out how to decrease the refcount properly.

David-Baddeley, 2020-04-24:
What wx version and platform?

David-Baddeley, 2020-04-24:
This really looks like a wx bug (that might be harmless) and not a bug with PYME (unless it is actually stopping something from working and we need a workaround)

barentine, 2020-04-24:
Just want to 'upvote' that it would be nice to close the floodgates a bit on stderr. Not fun to smooth over a bug in a dependency, but I think tidying up the console would go a long way to users paying attention to it.

David-Baddeley, 2020-05-12:
We can't do anything about reference counting from within python -  (reference counting happens in the compiled c code in python extensions and is not accessible from interpreted python code). Short of maintaining our own fork of wxpython (which I have zero desire to do) our only option is to wait for it to be fixed /submit a PR to the upstream wxpython and wait for it to propagate through.  Closing because this is not our bug.

@barentine on the more general topic of console verbosity/noise,  you'll be pleased to know that I've just found a separate bug that was dramatically increasing the volume of deprecation warning traffic we got on the console. The default behaviour of python is to print a DeprecationWarning the first time it is triggered for a given line of code and then shut up, one of our imports had messed with this so that they printed every time. This is now fixed.

closed 2020-05-12 by David-Baddeley