python-microscopy/python-microscopy #215: pytables version needs to be ratcheted up for py3 / numpy >=1.16

barentine, 2020-04-29:
pyme-depends pins pytables to <=3.4.2, which is buggy on later versions of numpy. See [here](https://github.com/PyTables/PyTables/issues/719#issuecomment-455612656).

We run into this when trying to load from vlarray hdf tables
```
Traceback (most recent call last):
  File "c:\users\pk466\code\nep-fitting\nep_fitting\dsviewer_modules\sted_psf_fitting.py", line 167, in <lambda>
    load_btn.Bind(wx.EVT_BUTTON, lambda e: self._on_load())
  File "c:\users\pk466\code\nep-fitting\nep_fitting\dsviewer_modules\sted_psf_fitting.py", line 484, in _on_load
    self._line_profile_handler.open_line_profiles(fpath)
  File "c:\users\pk466\code\nep-fitting\nep_fitting\core\handlers.py", line 224, in open_line_profiles
    p = profs[i]
  File "c:\users\pk466\code\python-microscopy\PYME\IO\ragged.py", line 127, in __getitem__
    return json.loads(self._data[item])
  File "C:\Users\pk466\Miniconda3\lib\site-packages\tables\vlarray.py", line 671, in __getitem__
    return self.read(start, stop, step)[0]
  File "C:\Users\pk466\Miniconda3\lib\site-packages\tables\vlarray.py", line 811, in read
    listarr = self._read_array(start, stop, step)
  File "tables\hdf5extension.pyx", line 2106, in tables.hdf5extension.VLArray._read_array
ValueError: cannot set WRITEABLE flag to True of this array
```

Why do we pin to pytables <=3.4.2?

A bit off topic:
I don't know our grand strategy on packaging, but with py2 being abandonware and people being forced to work from home at the moment, I think it would really behoove us to make installation on py3 more straightforward. The pip installation is actually pretty straightforward with the exception of needing to install cython and numpy first otherwise pymecompress can try to install before them in the queue and fails the whole thing. Are we considering making pip the default installation route? Should we make pypy pages for pyme and pymecompress?

David-Baddeley, 2020-04-29:
I've forgotten exactly what the issue was, but there was a fairly critical bug (i.e. one of spooling, saving fit results, and/or reading data back in didn't work at all) for a couple of the pytables releases post 3.4.2. It may well be that those bugs have since been fixed and we can change the pin to an exclusion of a certain range of pytables versions, but we would need to do some pretty thorough testing. The short answer is that I know that <=3.4.2 works, that 3.4.3 doesn't, and don't know if higher versions do.

I'll update the roadmap with a little more info, but my gut feeling is that we might be able to aim for 1 Sept as a target date for having the py3 version being the default install, and 1 Feb 2021 as when we hope to drop py2 and wx3 support. I don't think we're quite there yet on making py3 the default install -  (the h5rfile bug you found recently is an example of a pretty serious py3 issue which we've only just found). My absolute requirement that I won't compromise on is being able to smoothly spool and do analysis on a single windows machine. Making PYME pip-installable on py3 whilst keeping the conda package / installation docs py2 for now could be an attractive option in the interim (arguably it is already pip-installable if you give pip the github url), although there are a few issues to work through. We should, however, make pymecompress properly pip-installable first - this is really low hanging fruit and will give us a bit of experience with doing it for a simpler package.

Whether to aim for making the default install route via pip is a thorny issue. If we do go that route, we need to recognise that some of our users will not have a c compiler installed, and may not have the technical prowess (or potentially even the access rights - some users here are on centrally managed systems with no admin privileges) to install a c compiler. This means making binary wheels and ensuring that binary wheels are available for all our dependencies (a quick test of the pip installation route on linux confirmed that there are currently a fair number of packages which need to be compiled when pip-installing).

The other arguments for conda over pip are:

- conda links numpy against MKL rather than ATLAS, which can lead to a substantial performance improvement
- conda installs tend to be more self-contained (pip installs often assume the presence of various OS libraries - e.g. HDF5 and FFTW which may or may not be present in the right version)
- conda has a nice way of installing menu items/ shortcuts. If you 'conda install python-microscopy' on windows you now get links to the component programs in your start menu. This is not possible using pip.
- conda constructor offers a reasonably simple way of creating an executable installer for windows and OSX. 

Stacked against this is the pain that is conda dependency management. My gut feeling is to stick with conda as the default install route, but to offer pip as an option for people with a little move technical expertise.

David-Baddeley, 2020-05-04:
After testing a bunch of pytables and numpy (back to 1.11) versions I can't seem to replicate. I suspect this pin might date back to a really old numpy (or conda have removed and replaced some broken packages. Pin now removed.

closed 2020-05-04 by David-Baddeley