python-microscopy/python-microscopy #216: Naming inconsistencies in pyme modules

zacsimile, 2020-04-29:
We're introducing `pymeclusterofone` as yet another module command to launch elements of PYME. This is inconsistent with the other module names, which are all inconsistent with one another. To make entry points easier for new users, I propose the launch names be as follows.

```
pyme-acquire
pyme-visualize
pyme-image
pyme-cluster
```

If we could go a step further, I might even change `pyme-visualize` to `pyme-pointcloud` since both `pyme-visualize` and `dh5view` (`pyme-image`) are for visualization and analysis, just of different data types.

David-Baddeley, 2020-04-29:
It's a work in progress ... unfortunately hyphens in executable names aren't safe across platforms. 

All new entry points (including all the cluster ones) follow the pattern:
`PYMECommand`

`PYMEClusterOfOne` can't /shouldn't change to `PYMECluster` as we should be distinguishing between it and an actual cluster

`VisGUI` and `dh5view` definitely need to change (that's been on the TODO list for some time). The main reason it hasn't happened already is that I've been toying with the idea of combining them under a single entry point which would guess the data type and load the appropriate viewer. Do you have any feelings as to whether you think there is actually a need to two separate entry points here?

I'm not super interested in changing launchWorkers, fitMon, taskServerZC etc ... as I expect them to disappear.

David-Baddeley, 2020-04-29:
ps - I like `PYMEImage` for `dh5view`

zacsimile, 2020-04-29:
I agree that a single entry point makes the most sense.

I expect them to disappear, too. I was trying to make a 404 Not Found joke with that suggestion. I'll try to refrain from making deadpan jokes in issues from now on, but no promises.

zacsimile, 2020-07-08:
Addressed in #302.

closed 2020-07-08 by zacsimile