python-microscopy/python-microscopy #217: Segfaults / fatal error when testing on Ubuntu 18.04

David-Baddeley, 2020-04-30:
This is making our CI unit tests fail.

Have narrowed it down to something related to `test_spooler.py`, `test_pzf.py`, and `test_dataserver`. Interestingly multiple ones seem to need to run to trigger the error.

ie `pytest test_pzf.py` is fine, as are tests on each of the other modules in isolation, but, e.g. `pytest test_pzf.py test_dataserver.py` fails.

barentine, 2020-04-30:
This isn't a sure-fire debug strategy, but given how long we've struggled with finding this segfault it wouldn't be crazy to knock out any warnings: I wonder about our fromstring usages in pzf given the deprecation warning on later numpy versions.
```
tests/PYME/IO/test_pzf.py::test_PZFFormat_lossless_uint8
  /Users/Andrew/code/python-microscopy/PYME/IO/PZFFormat.py:354: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
    data = bcl.HuffmanDecompress(np.fromstring(data_s, 'u1'), outsize)
```
though this warning isn't triggered by the dataserver or spooler tests.

David-Baddeley, 2020-04-30:
none of the python code should generate segfaults - so the most likely candidate is some of the c code in pymecompress. I'm trying to get the tests running under valgrind now (it picks up a potential leak, but is somehow not finding the debug symbols so I don't know where it is).

David-Baddeley, 2020-05-01:
Finally making some progress after having to install valgrind from scratch ... - looks like it might be in the quantization code.

David-Baddeley, 2020-05-01:
this was down to writing over the end of the array using the 16 byte wide AVX instructions in PYMECompress quantization, now fixed

closed 2020-05-01 by David-Baddeley

barentine, 2020-05-01:
Good sleuthing. Thanks David

zacsimile, 2020-05-01:
This is thrilling. Thanks, David!