python-microscopy/python-microscopy #219: Ripley's analysis is not working

lukasfue, 2020-05-05:
I tried testing the new Ripley's analysis using the default parameter values and got this traceback:

Traceback (most recent call last):
  File "c:\users\laf62\python-microscopy\PYME\LMVis\Extras\clusterAnalysis.py", line 411, in OnRipleys
    result = recipe.apply(inputPositions=pipeline.selectedDataSource, inputMask=mask)
AttributeError: 'ModuleCollection' object has no attribute 'apply'

barentine, 2020-05-05:
Looks like that `apply()` should be an `execute()`. Thanks @lukasfue.

barentine, 2020-05-05:
oops, `recipe` should be `r`. Anyway, we'll take care of it

zacsimile, 2020-05-05:
I'm taking care of it now. Almost there, I think. There were a few other slight bugs after fixing `apply`.

David-Baddeley, 2020-05-05:
Apply is correct, not execute

zacsimile, 2020-05-05:
'Apply' is correct for the recipe, not for the `ModuleCollection`. I switched to using `apply` on the recipe (it was previously on the `ModuleCollection`). Fixes are coming shortly. I'm just testing them on real data right now to make sure they work.

zacsimile, 2020-05-05:
I pushed fixes in #220 that make this run now (tested on 2D and 3D data/masks). I'm not 100% sure the graphs coming out make sense. Will need to test on some more data structures/more masks to confirm. We are passing tests for `PYME.analysis.points.ripleys.ripleys_k_from_mask_points()`, so at least that part should be correct.

zacsimile, 2020-05-08:
This was fixed with the merge of #220. I think we can close.

closed 2020-05-12 by David-Baddeley