python-microscopy/python-microscopy #223: Animation improvements

zacsimile, 2020-05-06:
#222 restores the animation panel to its former glory, but I think we can do more. Ideally we'd be able to 

- [ ] set the number of frames between each view
- [ ] set the frame rate of the animation
- [ ] set the output pixel size (or at least scale factor)
- [ ] choose the output file type (JPG, PNG, TIFF)
- [ ] fix the coloring, particularly on the LUT bar in exported animations
- [ ] combine the "make" and "save" buttons so you can choose to export as JSON, JPG, PNG, or TIFF

It looks like Michael Graff started implementing some of these ideas in `PYME.LMVis.Extras.animation.EditPanel` and `EditDialog`. Did we make a decision not to use the `EditPanel`? Or was it never finished? Are there additional features we'd like to add to our animation module?

David-Baddeley, 2020-05-07:
- I'd favour keeping 'Save' and 'Make' as separate buttons (ideally renaming 'Make' to 'Export') so that 'Save' refers to the key frames and other bits (something you can reload on other data) and Export gives you an option between JPG, PNG, and AVI (if you have openCV installed).
- setting the number of frames between views and setting the frame rate are a bit redundant when exporting individual images.
- what is wrong with the colouring?

The `EditDialog` gets displayed when you double click on one of the views. It would probably make more sense to have the duration as an additional column.

zacsimile, 2020-05-07:
- Good point, we'll drop frame rate. 
- I agree duration should be an additional column rather than a popup.
- The coloring is very...psychedelic. If you export an animation, you'll see what I mean.

zacsimile, 2020-05-11:
Can we add single TIFFs as an export option (not stacks)?

zacsimile, 2020-05-13:
There were requests to add `time` and `alpha` changes to the animation. I think we'd just need to put these in `PYME.LMVis.views.View`. Doesn't seem terrible.

David-Baddeley, 2020-05-22:
Was tidying and found the animation UI mockup I originally drew for Michael (with a duration column :p) 

[image]

zacsimile, 2020-05-22:
Cool! We can build this, we're already fairly close. I wonder if a duration column makes sense. The duration is associated with the transition from one view to the next, so I think it should show up in between views (maybe as an arrow with a time next to it), rather than as a column associated with a view, since this looks ambiguous. However, I think this would require making a non-standard window. I think, in this case, it may be necessary.

David-Baddeley, 2020-05-22:
Although the duration is technically associated with the transition, I don't think it's awful to have it in the GUI on the same row as the view you are transitioning from - it's certainly not worth writing a custom UI component.

zacsimile, 2020-06-20:
All of these are addressed in #282.

closed 2020-06-25 by zacsimile