python-microscopy/python-microscopy #224: Where do we interact with snapshot and animation scale factors?

zacsimile, 2020-05-06:
Should the option to change the scale factor (pixel size) always pop up when generating a screenshot or animation, or should this be set at a default value and changed elsewhere, either in the shell or a different GUI settings panel?

My vote is for pop up every time we generate a screenshot so users know it's an option, and have it listed as an option in the animation panel.

David-Baddeley, 2020-05-06:
The ideal in terms of UI (at least for a shapshot) would be be to to display it as a changeable option within the save dialog (think of most of the Adobe save dialogs for an example) - this would give us the best of both worlds - could normally be ignored, but is there if you need it. The downside to this is that we would need a custom save dialog (which are invariably gross - see the adobe ones for example - and not themed to match the OS) rather than the default OS ones.

For the animation we could have it as part of the animation panel - maybe under a collapsable 'export settings section' or similar. This would be better than an extra pop-up.

Part 2 of this should be how do we present the scale factor to the user. Do we present it as a pixel size in nm, or as a ppi value (with the default of 72 for screen resolution). If you want to measure from the resulting image, the nm value is potentially useful, and is also useful for getting 2 differnt images to the same scale. Measuring from images, however, is problematic for orthographic 3D projections and useless in perspective 3D so I'd like to heavily discourage measurement from snapshots or animations. If you are taking the snapshot to put on a poster or in a figure, ppi is probably the more relevant quality metric, but this isn't much use for making different images the same scale (you'd have to do this before export - which might actually be the right approach - expose setting the pixel size for the live display so you can match zooms and then have a ppi setting for export).

yongdengzhang, 2021-11-09:
I personally really want this feature. It would be great to output the current 3D snapshot as high-resolution images for making nice figures, not necessarily for measurements. As this depends on random orientations, it is not easy to do by other means. At the moment, I would mostly use PYME for generating high-resolution images for figures and making nice movies. It will be a bump to me if I can only output the snapshots limited by the resolution of the screen.

lukasfue, 2021-12-06:
I too really want this feature. Especially with 3D surfaces and such. There's a lot that can be visualized with PYME that can't be included in a high-resolution generated image. So the option of having high-resolution screenshots would be excellent.

David-Baddeley, 2021-12-06:
Snapshot aspect should be fixed in #1162, should probably revisit the animation stuff to make the UI similar / check that it works.

closed 2021-12-06 by David-Baddeley

David-Baddeley, 2021-12-06:
Just for documentation:

## View->Save snapshot

Now brings up the following dialog:

[image]

Final image size can be specified in inches+ dpi/ppi for print figures, and in pixels for e.g. webpages. To make things work with 3D, the images cannot be assumed to have a fixed pixel size, as this will change with projection and perspective (depth into the scene).

*note:* Failure to specify a filename will result in silent failure (akin to cancel) - we should potentially add a log message here

## Animation

[image]

You can now specify the width in pixels in the export settings (-1 means current window size).