python-microscopy/python-microscopy #225: Add orthographic projection view to PYMEVisualize

zacsimile, 2020-05-06:
Give our point clouds a little perspective.

zacsimile, 2020-05-06:
@David-Baddeley where is this implemented in the backend?

David-Baddeley, 2020-05-06:
`glCanvas.displayMode='3DPersp'` - see lines 364-367 in gl_render_3D_shaders.py

the amount of perspective (given by front and back planes in the `glFrustrum()` call) might need tweaking.

David-Baddeley, 2020-05-06:
There used to be a GUI radio button for this, but it was dropped to save screen real estate, with the plan to resurrect when replacing the current radio buttons in the toolbar with icons.

David-Baddeley, 2020-05-07:
Orthographic describes the current default ... where lines are parallel and scaling (mostly) preserved)

David-Baddeley, 2020-05-07:
current perspective viewport is super trippy - definitely needs work on the `glFrustrum()` call.

David-Baddeley, 2020-05-07:
I think the vanishing point is behind you (at least when compared to how the mouse rotates the object)

zacsimile, 2020-05-07:
Some of this is easy to fix. One thing I'm not sure how to handle is the LUT and scale bar. As our architecture is set up now, I don't see a way to not apply the `glFrustrum()` call to them as well. It's kind of weird that they shrink slightly along with our point cloud perspective.

zacsimile, 2020-05-07:
OK, I'm lost on this one. When the math is done out (followed this Cornell lecture: http://www.cs.cornell.edu/courses/cs4620/2011fa/lectures/practicum02.pdf), this should work

```
near = 1 + eye
far = 100.0
fovy = np.pi/2
top = near*np.tan(fovy)
bottom = -top
right = aspect_ratio*top
left = -right
glFrustum(left, right, bottom, top, near, far)
```

And we can debate what near and far should be, but generally this should be correct. However, it seems that the perspective cone is inverted, so that I'm looking from the bottom of the pyramid instead of the top. What am I missing?

zacsimile, 2020-05-11:
Re: personal communication with @David-Baddeley, we have this back. However, the shift in perspective from the front to the back of the viewing cone seems a bit too gradual. This is particularly noticeable with the bounding box on. I think we should make the projection more dramatic.