python-microscopy/python-microscopy #227: Convert numbers in PYMEVisualize View Clipping to editable text boxes

zacsimile, 2020-05-06:
Feature request to allow users to set the min/max manually as well as with the slider. This would be in addition to double-clicking to set Clipping range, we still want to keep that.

David-Baddeley, 2020-05-06:
I'm concerned about the amount of screen real estate this would take if we were to add standard text boxes. It feels like this should potentially be something like an 'advanced clipping settings' popout or similar - not sure how to make it easy to use and discoverable. Alternative is to further develop the custom slider control so that, e.g. you could double click on the numbers (but developing custom UI widgets is a questionable use of resources at the best of times).

zacsimile, 2020-05-06:
I think it will only add a little more width to the left-hand bar, which would be fine. We can safely assume in this day and age people will use PYME full-screen, in which case a little extra screen real estate in exchange for easy clipping control will be welcome. I think the ease-of-use benefits outweigh the screen real estate loss.

David-Baddeley, 2020-05-06:
I never use it full screen :p. I also have a habit of putting a VisGUI window and a rendered image window side by side (meaning that you only have half the screen for the VisGUI window). 

If we use standard controls, we'd end up with around 50% of the width of the side bar being used by the text boxes and we'd probably double or triple the height of the view clipping stuff (the controls are is particularly large on linux, which has to do with the default font sizes, but as we need 5 figures for each of the upper and lower bounds in a typical image they will be large on any platform).

zacsimile, 2020-05-07:
I think there's a way to make this work. Vutara seems to have this functionality without taking up too much screen real estate. Maybe I'll just try it and see what happens. I bet there's a way to shrink the font/box size, which would help a lot. It looks like `wx.TextCtrl` has a `wxSize` property where you can set the width and height of the box and a `style` argument, which you can use to set the font size. Both of these arguments are available in wx3 and wx4. Those should be a good start.

David-Baddeley, 2020-05-07:
It'll be very hard to do whilst maintaining some semblance of consistency / having things work across platforms. I'd strongly advise against making the vertical size anything other than -1 (auto) - changing the font size should change the vertical size to match. I doubt it'll be OK in terms of screen real estate / general look and feel with the cropping in it's current location whatever you do.

The two solutions I had thought of were:

- make a text field pop up for editing when you clicked one of the bounds (there would be a bit of fiddling to make this work - potentially including moving the limit displays to the ends of the bars rather than having them track the limits, but there are a few examples of similar things in the wx demos)
- moving the clipping to the right hand side in a pop out tab (like those used for display settings dh5view) where it could potentially be a bit larger. This would align with the long term plan of moving all the display settings to pop-outs here to give us some UI consistency between VisGUI and dh5view.

David-Baddeley, 2020-05-07:
You could even pop up the text field on hover, rather than needing a click (potentially more intuitive)

zacsimile, 2020-06-30:
Addressed in #233.

closed 2020-06-30 by zacsimile