python-microscopy/python-microscopy #230: MATLAB localization file not loading into Visgui

lukasfue, 2020-05-07:
I used Yongdeng's MATLAB code to localize some single-particle tracking data, but I get an error when I try to load the localization file into Visgui. Here is a screenshot of the 'handles' structure.

[image]

Here is the Traceback:

Traceback (most recent call last):
  File "c:\users\laf62\python-microscopy\PYME\LMVis\visCore.py", line 402, in OnOpenFile
    self.OpenFile(filename)
  File "c:\users\laf62\python-microscopy\PYME\LMVis\visCore.py", line 713, in OpenFile
    args = self._populate_open_args(filename)
  File "c:\users\laf62\python-microscopy\PYME\LMVis\visCore.py", line 674, in _populate_open_args
    mf = loadmat(filename)
  File "C:\Users\laf62\.conda\envs\pyme\lib\site-packages\scipy\io\matlab\mio.py", line 141, in loadmat
    MR, file_opened = mat_reader_factory(file_name, appendmat, **kwargs)
  File "C:\Users\laf62\.conda\envs\pyme\lib\site-packages\scipy\io\matlab\mio.py", line 71, in mat_reader_factory
    raise NotImplementedError('Please use HDF reader for matlab v7.3 files')
NotImplementedError: Please use HDF reader for matlab v7.3 files

zacsimile, 2020-05-07:
Thanks for posting this! I've run into this issue before. I'm not sure if the old-style MATLAB loader (`PYME.IO.tabular.MatfileSource`) is supposed to handle this kind of file and we just need to update to using `h5py`, or if we never supported this style (likely).

One quick-and-dirty solution is to export the columns of `fitInfo_dc` as three variables, `x`, `y`, and `z` and save that as a new `.mat` and open that with PYME.

zacsimile, 2020-05-07:
Worth noting that this is the Zeiss, not the 4pi MATLAB code.

barentine, 2020-05-07:
Good gravy - what version of matlab was that saved with, @lukasfue? Maybe the dirty(er) solution is just to open the .mat in a more recent matlab and resave it, then open in visgui.

barentine, 2020-05-07:
Please excuse my ignorance - maybe not a matlab version issue, but I wonder if @lukasfue needs to save to a [different .mat version](https://www.mathworks.com/help/matlab/import_export/mat-file-versions.html), on top of @zacsimile 's suggestion to change the contents too.

zacsimile, 2020-05-07:
Ideally yes, but this is the default coming out of our Zeiss setup. From a user perspective, it just doesn't work in PYME, but does work in MATLAB.

lukasfue, 2020-05-07:
Fair question @barentine, its MATLAB 2011b which we use for running Yongdengs code because newer versions don't work and nobody has bothered to update the code

zacsimile, 2020-05-07:
I think a different version would be ideal. I'm not sure why v7.3 is the default for the Zeiss code.

zacsimile, 2020-05-08:
Since Lukas was able to save the MATLAB file in a different style and then load it into PYME, and since the Zeiss is soon to be decommissioned, I propose we do not support this type of file. 

We could put a `try`/`except` in `_populate_open_args` that provides a suggestion to save the MATLAB with one vector per variable.

zacsimile, 2020-06-30:
I think we can close this. #229 plus upgrades in the lab should deprecate this issue.

closed 2020-07-21 by David-Baddeley