python-microscopy/python-microscopy #231: You can only track once

zacsimile, 2020-05-07:
If you run `Analysis>Tracking>Track single molecule trajectories` more than once, you get a cyclic dependencies traceback, I believe due to using the same output name. To fix this, we should use `PYME.LMVis.pipeline.new_ds_name()` on `tracks` and `with_tracks`. Thanks to @lukasfue for pointing this out.

```
Traceback (most recent call last):
  File "/Users/zachcm/Code/python-microscopy/PYME/LMVis/Extras/particleTracking.py", line 107, in OnTrackMolecules
    recipe.execute()
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/base.py", line 584, in execute
    exec_order = self.resolveDependencies()
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/base.py", line 565, in resolveDependencies
    return toposort.toposort_flatten(dg, sort=False)
  File "/anaconda3/envs/pyme/lib/python3.6/site-packages/toposort.py", line 80, in toposort_flatten
    for d in toposort(data):
  File "/anaconda3/envs/pyme/lib/python3.6/site-packages/toposort.py", line 71, in toposort
    raise ValueError('Cyclic dependencies exist among these items: {}'.format(', '.join(repr(x) for x in data.items())))
ValueError: Cyclic dependencies exist among these items: ('tracks', {<PYME.recipes.tracking.FindClumps object at 0x1a3121b938>}), ('with_tracks', {<PYME.recipes.tracking.FindClumps object at 0x1a3121b938>}), (<PYME.recipes.tracking.FindClumps object at 0x1a3121b938>, {'with_tracks'})
```

zacsimile, 2020-05-07:
It would also probably be nice if we alerted the user if no tracks are found rather than fail fairly quietly only within the terminal.

```
Traceback (most recent call last):
  File "/Users/zachcm/Code/python-microscopy/PYME/LMVis/Extras/particleTracking.py", line 110, in OnTrackMolecules
    layer = TrackRenderLayer(pipeline, dsname=tracking_module.outputClumps, method='tracks')
  File "/Users/zachcm/Code/python-microscopy/PYME/LMVis/layers/tracks.py", line 97, in __init__
    self._set_method()
  File "/Users/zachcm/Code/python-microscopy/PYME/LMVis/layers/tracks.py", line 114, in _set_method
    self.update()
  File "/Users/zachcm/Code/python-microscopy/PYME/LMVis/layers/tracks.py", line 148, in update
    self.update_from_datasource(self.datasource)
  File "/Users/zachcm/Code/python-microscopy/PYME/LMVis/layers/tracks.py", line 232, in update_from_datasource
    self._bbox = np.array([x.min(), y.min(), z.min(), x.max(), y.max(), z.max()])
  File "/anaconda3/envs/pyme/lib/python3.6/site-packages/numpy/core/_methods.py", line 34, in _amin
    return umr_minimum(a, axis, None, out, keepdims, initial, where)
ValueError: zero-size array to reduction operation minimum which has no identity
```

We could have a dialog pop up and say, "hey, try different parameters".

We also probably shouldn't create `tracks` and `with_tracks` data sources if no tracks are found.

zacsimile, 2020-05-08:
Added incremental naming of `with_tracks` and `tracks` in #232 to partially fix this. I think this is sufficient for now. There's still no alert to the user that they choose bad parameters, and I don't prevent `with_tracks` and `tracks` data sources from being generated if there are no tracks. Reasons are as follows.

- Nowhere else (AFAIK) do we alert users to poor parameter choices, and I'm not sure this is the place to start.
- In order to prevent `with_tracks` and `tracks` from being generated if no tracks were found, we'd have to, in these instances, make the recipe `PYME.recipes.tracking.FindClumps` return no output, which feels like a bad move.

David-Baddeley, 2020-05-12:
I've changed the display to use the `with_tracks` object which partially resolves the poor parameter choice problem - this is mostly the minClumpSize parameter which is biting us. Will change things further such that a `tracks` output is not provided by default (at least until track reporting is up and running), and potentially also change how the minClumpSize is displayed in the UI.

David-Baddeley, 2020-05-12:
Actually, I now regret merging #232 - tracking twice is almost always an error and should be treated as such. You should modify the exisiting tracking module rather than adding a new one. The right behaviour would be to test to see if `with_tracks` is in the namespace, and if so pop up a dialog telling the user to edit the parameters of the existing tracking module instead of adding a new one. We could also be nice and pop this up for them (would probably require holding a reference to the tracking module we create the first time we call the tracking function).

The only valid use case for tracking twice is if you are doing multi-colour tracking, but this is both relatively unlikely and sufficiently specialised and complex that you probably want to set it up in the recipe editor rather than using the menu functions.

David-Baddeley, 2020-05-12:
Now fixed - tracking more than once displays an error dialog and re-opens the settings for the initial failed tracking.

closed 2020-05-12 by David-Baddeley