python-microscopy/python-microscopy #234: Set raw data as background broken in VisGUI

zacsimile, 2020-05-10:
Output from `File>Open Raw/Prebleach Data` followed by selecting `Chan0` and then `View>Set as visualization background` below. Selecting `Slices` and `GLComp` also does not work, but there's no traceback.

```
Traceback (most recent call last):
  File "/Users/zachcm/Code/python-microscopy/PYME/DSView/modules/visgui.py", line 45, in OnViewBackground
    img = numpy.minimum(255.*(ivp.image.img - ivp.clim[0])/(ivp.clim[1] - ivp.clim[0]), 255).astype('uint8')
AttributeError: 'ImageStack' object has no attribute 'img'
```

zacsimile, 2020-05-10:
Whoops. Looking at the code, I misunderstood what this was supposed to do. It looks like it changes the background in the DSViewer canvas to the current image, not the VisGUI background. However, the image is already displayed in DSViewer. I assume this must add the image as a background to a glCanvas if we render a surface from the image and then want to display the original image along with the surface.

David-Baddeley, 2020-05-10:
Definitely broken - the background image display disappeared with the move to shaders. A replacement is almost ready - using the `PYME.LMVis.layers.image_layer.ImageRenderLayer` class, but is not plumbed into the GUI.

Accessible on from the shell using the following:

```python
from PYME.LMVis.layers import image_layer
l = image_layer.ImageRenderLayer({'im': ImageStack(...)}, 
                                 dsname='im', 
                                 channel=i, # colour channel
                                 slice=0, # z / t slice
                                 context=glCanvas.gl_context))
glCanvas.add_layer(l)
```

Will see if I can fix the GUI.

David-Baddeley, 2020-05-10:
A bit more context for that particular bit of code. The use case (currently broken) is a follows:

- Open some raw data in VisGUI
- Render an image (this pops up a new dh5view window with the image and the `visgui` module loaded)
- Set the background of the VisGUI window to the rendered image 

It should also work with images loaded via the `File->Open raw/prebleach data` menu item in a similar fashion. 

Nothing to do with the surface rendering ...

zacsimile, 2020-05-11:
What do we select (when it works) in the GUI to set the contents of the
dh5view window to the background of the PYMEVisualize window?

David-Baddeley, 2020-05-11:
`View->Set as visualisation background`. Have almost got things fixed, but not quite - running into shader issues (which might be due to the shader being allocated to the wrong GL context).

David-Baddeley, 2020-05-11:
Just pushed where I've got to with the fix to keep you posted (and as I should probably drop this for the day to work on other stuff). Still broken, but less broken than it was (should be close to where it needs to be, once we find out what is going wrong with the shader).

zacsimile, 2020-05-11:
Thanks, David! I may look into this for a second if you're dropping it, but probably not for too long as it's getting late here. 

Looking through the code, it looks like `DSViewFrame`s `glCanvas` variable is used exclusively for connecting to PYMEVisualize. Or have a missed a case?

David-Baddeley, 2020-05-11:
pushed on a little and now fixed  (a little gross as we need an explicit `SetContext()` call to ensure that the right context is active when we initially load the shader) - we might need to revisit our shader architecture at some point. 

You are right on `glCanvas` being used exclusively as a reference to the the VisGUI display in dsviewer instances launched within VisGUI.

David-Baddeley, 2020-05-11:
Now fixed / back to where it was pre shaders breakages. Still lacks good support for z (and display sync with image view).

closed 2020-05-11 by David-Baddeley

zacsimile, 2020-05-11:
Thanks so much, David!

It might be worth throwing a comment in on line 78 of `PYME.DSView.dsviewer` indicating that's what `glCanvas` is for. It took me a second to figure that out and `dsviewer.glCanvas` is all over the place, giving the initial impression that there's another glCanvas for `dh5view` (I originally assumed this was the canvas you used for 3D renderings coming from `dh5view`).

zacsimile, 2020-05-11:
Hi David,

I gave it a go. The code is no longer throwing errors, but it seems like the pixel values aren't getting mapped quite right. This is what it looks like after I try to map a TIFF behind some tracks. I tried with a few different tiffs just to be sure the problem is consistent. 

[image]

David-Baddeley, 2020-05-11:
yep - some comments and or renaming would be very useful - it predates any of the surface stuff.

As to the scaling, if you go to the im layer (should be layer 0) you should be able to change the scaling. We should really try and copy it across from the other viewer (at least to start with) - might be as simple as calling `sync_display_opts()` (modulo spelling) on the new layer..

zacsimile, 2020-05-11:
Oh, I see. Thanks!