python-microscopy/python-microscopy #235: Showing shortcut help window freezes PYMEVisualize

zacsimile, 2020-05-11:
If you go to `Help>Show shortcuts` and then click `OK`, the whole VisGUI window is frozen and you have to restart to do anything. No traceback. Best guess is `Help>Show shortcuts` retains the focus after closing.

David-Baddeley, 2020-05-11:
The issue here is that the OK button for a modal dialog should have `id=wx.ID_OK` and should (probably) not be bound to `OnClose()`. It could also that calling `Destroy()` in the OK button handler before the dialog has actually closed is an issue.

The proposed solution makes the dialog non-modal, but doesn't fix the underlying issue. In general you should always destroy a dialog after showing it. I guess this is commented out in your code as `.Show()` returns straight away (as contrasted to `ShowModal()` which waits until you close the dialog), and you would destroy the dialog before you saw it otherwise.

zacsimile, 2020-05-11:
I'm fairly sure the dialog gets destroyed by its `OnClose()` method. Some Googling confirmed that both the object and the dialog are destroyed by that call. The proposed solution seems to work on my machines. It's not on yours?

David-Baddeley, 2020-05-11:
That's not the case - the dialog sticks around so that you can, e.g. read out the values of text fields after control returns to the calling code (there is a chance this might be specific to ShowModal), but in any case, you should probably be calling `self.Close()` rather than `self.Destroy()` in the handler for the OK method. It's probably better to not call it `OnClose` either as this has the potential to be a special name that you are over-riding (for context this was originally a contrib from Christian when he was just learning wx). 

I was going to follow up a bit more on the first post, but you beat me to it - my main concern is that the dialog as it stands is pretty broken, and if we are fixing it we may as well do it properly so it doesn't bite us again. 

Your solution fundamentally changes how the shortcuts dialog behaves (i.e. modal vs non-modal). This is something which should be considered - ie why do you think that it should be non-modal?I'm not saying that it is wrong, just that it should be considered. As I see it, the potential advantage to non-modal is that you can leave the dialog open for reference. The disadvantage is that it potentially never gets closed, "littering" your screen.

zacsimile, 2020-05-11:
Maybe I'm misunderstanding your first paragraph, but the dialog is destroyed by the `OnClose()` method as it's currently written if you use `Show()` instead of `ShowModal()`. If you press `OK`, the dialog is definitely gone. I have tested it. 

You are right that this `OnClose()` overwrites a special name. It would probably be better to not do that. 

I think leaving this dialog up for reference is a good idea (switching `ShowModal` to `Show`). We could certainly do it more cleanly (write a dialog with `id=wx.ID_OK`).

zacsimile, 2020-05-11:
The more I look at it, the more I think Christian got the custom `wx.Dialog` mostly right. I've re-written it to work with both `Show()` and `ShowModal()`, and I've switched it back to `ShowModal()` so I don't change it's behavior.

zacsimile, 2020-05-12:
Ah, I realize I did misunderstand your first paragraph, sorry. It's possible the `Destroy()` command wasn't actually destroying the dialog, but it appeared as though it was. The updates I've made, however, should actually destroy it.

closed 2020-05-12 by David-Baddeley