python-microscopy/python-microscopy #236: 3D bounding box

zacsimile, 2020-05-11:
Optional display of a 3D bounding box.

barentine, 2020-05-11:
I believe we have this in `PYME/LMVis/layers/ScaleBoxOverlayLayer`, though it'll probably need to be gone over/tested again as well as added back to the UI

zacsimile, 2020-05-11:
Looks like it's loaded every time, just hidden. If you go to the shell and type `glCanvas.underlays[0]._show = True`, it pops up. Pretty neat!

zacsimile, 2020-05-11:
Presumably we should just make a button to show/hide this and leave it hidden by default @David-Baddeley?

barentine, 2020-05-11:
If I'm not crazy, once upon a time there was a UI to toggle whether there was a grid or just the edges, whether box sizes of each dimension were shown or only one, etc.

barentine, 2020-05-11:
PYME/LMVis/Extras/scaleBox.py

I think Michael made this pretty complete after sitting down with Yongdeng.

zacsimile, 2020-05-11:
Oh I see. Under `Extras>Scale Box`. I just have to update the `GridSizer` argument and we're good to go. PR to follow.

zacsimile, 2020-05-11:
I still think it would be nice to have a button at the top of the GUI to toggle this on and off, but the controls are back.

David-Baddeley, 2020-05-11:
Reworking the UI for this to put a show/hide toggle on the toolbar (and make it slightly less ugly) has been on the TODO list for ages.

I know you guys keep saying that everything works on wx4, but this is another example of something which doesn't. I'm pretty sure there are still lots of similar ones that we haven't found yet - we're slowly ticking them off, but this is why I want to keep wx3 compatibility for now - so I know we have a version where everything works that we can compare to.

David-Baddeley, 2020-05-11:
To elaborate on slightly less ugly - I think the settings - colour, axis flipping etc probably belong in a settings dialog, and not something that is visible the whole time - the activating the scale box from the menu is a bit of a crummy workaround in the meantime.

zacsimile, 2020-05-11:
It works on wx4 now! And I searched the rest of the code and I think those were the last of the deprecated `GridSizer` calls. 

I agree with your proposed UI changes.

barentine, 2020-05-11:
@David-Baddeley I agree that there're probably similar wx4 migration errors lurking, but they're probably similarly trivial. Is it reckless to call the past releases the ones we know worked with wx3 and take a small wx4 plunge with the source?

David-Baddeley, 2020-05-12:
Not sure whether to close this, or to mark as an enhancement (re adding to toolbar etc..)

zacsimile, 2020-05-12:
I'd say enhancement (re-adding to toolbar).