python-microscopy/python-microscopy #237: LUT names

zacsimile, 2020-05-11:
Can we have

- Fewer LUT options by default
- Human-readable names for these LUT options (e.g. 'C' should read 'Cyan')

barentine, 2020-05-11:
Probably isn't super annoying to leave everything there if we just add a list of ~10 or 20 human-readable at the start of the dropdown (even if they're repeated by their pyplot names farther down in the cmap abyss)

David-Baddeley, 2020-05-11:
I'm keen to keep all the matplotlib ones - they can be surprisingly useful. There is a good argument, however, for grouping 'r', 'g', 'b', 'c', 'm', 'y' along with their upper case -solid colour equivalents together, and making it more obvious (especially for the solid colour ones) what they are.

zacsimile, 2020-12-21:
Discussion moved to #700.

closed 2020-12-21 by zacsimile