python-microscopy/python-microscopy #241: Generate isosurface from Delaunay tessellation is broken

zacsimile, 2020-05-11:
I need to revisit it at some point. 

We should also probably think about the pipeline from a user's perspective. If you select `View > Create Delaunay Tesselation` (sp. tessellation), and then `View > Generate Isosurface from Delaunay Tesselation`, it won't work. It expects the field `dn` (density) from `Estimate density` or `Estimate density (circumcenters)`, which also create, but do not display, a tessellation. It's just a bit confusing.

Also these should probably not be in the `View` menu. I suggest we move most of it to `Analysis > Density` and the isosurface generation from Delaunay tessellation to `Mesh`.

David-Baddeley, 2020-05-11:
Agree that it's a bit of a mess - At the moment I consider this pretty experimental, so there might be an argument for letting the use cases firm up somewhat before trying to do get the UI nailed. Maybe we need an "alpha" menu in VisGUI to put stuff that is in development.

zacsimile, 2020-05-11:
Not a bad idea. Maybe `Extras>Alpha`?

Use cases would be helpful. However, I should be able to fix the marching tetrahedra without use-cases, I think.

PR #522 (2020-12-08): Marching Tetrahedra Updates
Addresses issue #241.

**Is this a bugfix or an enhancement?**
Both

**Proposed changes:**
- Update calls in `DelaunayMarchingTetrahedra` (formerly `MarchingTetrahedra`, sorry, I thought the rename made things clearer, I know I'm not supposed to do this) recipe to be modern
- Make `View>Generate Isosurface from Delaunay Tesselation` auto-call `DelaunayTesselation` and `LocalPointDensity`
- Implement a raster version of MarchingTetrahedra as a double-check
- Add generation of test data at command line launch to `PYMEVis` (`pymevis -t`) for debugging purposes

All of  [... cut]
closed 2020-12-08 by David-Baddeley