python-microscopy/python-microscopy #246: Data pipeline disappears when there is more than one starting point for the pipeline

zacsimile, 2020-05-14:
[image]

zacsimile, 2020-05-14:
[image]

barentine, 2020-05-14:
or more than one output
[image]

David-Baddeley, 2020-05-18:
Issue is not present on py2, wx3. 

best guess, it's related to this traceback:
```
wx._core.wxAssertionError: C++ assertion "GetEventHandler() == this" failed at /opt/concourse/worker/volumes/live/b30868b5-2a7c-479c-4e29-27874bbc69a6/volume/wxpython_1547931175613/work/ext/wxWidgets/src/common/wincmn.cpp(478) in ~wxWindowBase(): any pushed event handlers must have been removed

The above exception was the direct cause of the following exception:

SystemError: <class 'wx._core.UpdateUIEvent'> returned a result with an error set

```

The implication of which is that it's probably traits not cleaning up after itself properly when we destroy the old recipe display.

Any punters to try on py2, wx4 to confirm this is a wx4 issue?

zacsimile, 2020-07-17:
Not a wx4 issue. Tried on python 2. Works with wx3 and wx4. Still receive this error with py2/wx4, but the pipeline doesn't disappear.

```
wx._core.wxAssertionError: C++ assertion "GetEventHandler() == this" failed at /opt/concourse/worker/volumes/live/80490b2f-c578-4407-562a-dbbbe365123b/volume/wxpython_1547931284446/work/ext/wxWidgets/src/common/wincmn.cpp(478) in ~wxWindowBase(): any pushed event handlers must have been removed
```

I tried matching the versions of `wxpython`, `traits`, `traitsui`, and `matplotlib` that work on python 2 in the python 3 install. No dice. Exactly the same package sets don't have this error on python 2, but do on python 3.

David-Baddeley, 2020-07-17:
ugh. I doubt if it's a matplotlib issue (pipeline display on left doesn't use matplotlib). Seeing as it would appear to be py3 rather than wx4 I wonder if it's an iterator vs list issue ...

zacsimile, 2020-07-17:
Is pipeline display on the left defined in `PYME.recipes.vertical_recipe_display`? I'm not getting any iterator issues, but it could be in a `try` block...

David-Baddeley, 2020-07-17:
yep - it's vertical recipe display

zacsimile, 2020-07-17:
Cool. It does use `matplotlib` in two places, but I don't think that's the issue. All the dictionary iterators seem to have been updated. Will keep investigating.

zacsimile, 2020-07-17:
OK, the issue is in this section at the beginning of `_layout`. 

```
if self.fp:
             self.fp.elements = []
             self.fp.DestroyChildren()
        self.fp = None
        #print('destroyed fold panel children')
        
        self.DestroyChildren()
        #print('destroyed children')
```

It looks like `fp` is destroyed multiple times in py3, but just the once in py2. Not sure why yet.

zacsimile, 2020-07-17:
It was a red herring. Fix is in #334. `sorted()` on line 204 of `PYME.recipes.recipeLayout is sometimes called on a list of objects, e.g.`[<PYME.recipes.localisations.ProcessColour object at 0x1c24c57308>, <PYME.recipes.surface_fitting.DualMarchingCubes object at 0x1c25dcf410>]`. This is no problem on py2, but `sorted()` needs a key to sort on objects in py3.

zacsimile, 2020-07-19:
Fixed in #334.

closed 2020-07-19 by zacsimile