python-microscopy/python-microscopy #249: Freeform selection crop in dh5view not working as expected in py3

zacsimile, 2020-05-17:
I generated a Gaussian mask. Then, if I square crop the mask, everything is fine. If I freeform crop, the `GLComp` window shows up either dark or as a coarse gradient (attached), which is not what I cropped. If I switch to the `Image` window, I get the following traceback.

```
Traceback (most recent call last):
  File "/Users/zachcm/Code/python-microscopy/PYME/LMVis/imageView2.py", line 136, in OnPaint
    self.DoPaint(MemDC)
  File "/Users/zachcm/Code/python-microscopy/PYME/LMVis/imageView2.py", line 322, in DoPaint
    imw.Rescale(imw.GetWidth() * sc, imw.GetHeight() * sc)
wx._core.wxAssertionError: C++ assertion "IsOk()" failed at /opt/concourse/worker/volumes/live/b30868b5-2a7c-479c-4e29-27874bbc69a6/volume/wxpython_1547931175613/work/ext/wxWidgets/src/common/image.cpp(1739) in GetWidth(): invalid image
```

[image]

David-Baddeley, 2020-05-17:
How are you doing the freeform crop? I don't think it's currently possible to do a freeform crop (you can get there with a lot of messing around and combining a freeform mask with a rectangular crop, but it would be convoluted and artificial. I'm guessing you are doing the following.

- Change to freeform drawing tool
- Draw a region
- `Processing->Crop` from the menu

`Processing->Crop` always does a rectangular crop, using the current rectangular selection. If you've drawn a freeform loop, this will be a small rectangle spanning the start and end points of the loop (the corners of the selection rectangle are always defined by the start and end points of your mask drag). You can see what your crop region is by switching back to the rectangle selection tool before cropping.

I guess we could break this down into:

- a feature request for a freeform crop function
- making the crop behaviour less confusing if you have the freeform tool selected. The simplest approach here would be to disable crop if you have the freeform tool selected, but we could also look at, e.g., making the rectangular selection the bounding box of the freeform line.

Currently to get a freeform / masked crop you would need to:

- crop to the desired rectangular ROI
- load the `manualSegmentation` module (this should generate a new window with an empty mask)
- draw your freeform shape in the original window.
- choose `Segmentation->Fill selection` from the menu
- choose `Segmentation->Multiply image with mask`

This workflow is flexible in that you can mask multiple regions within the one image, but has a little more overhead than simply drawing a selection and hitting crop.

Ultimately the functions of the `annotation` and `manualSegmentation` modules need to be combined and tidied up. I have also have a feeling that `manualSegmentation currently requires shapely (so is OSX specific / requires a fair bit of manual installation pain on other platforms).

David-Baddeley, 2020-05-17:
PS - what is the use case for a freeform crop? Is this specifically targeted at the visualisation feature request or is there also an analysis use case? For visualisation (and most analysis requirements) as mask (rather than a crop) is likely sufficient or even preferable.

zacsimile, 2020-05-18:
I was trying to clean up a mask and thought, 'oh, how convenient, there's a freeform polygon tool and a crop function`, so I tried to use it. It's fine that it doesn't work, and we don't need it, I just assumed it was supposed to work, so I made this issue.

I did try it exactly as you guessed, it seemed intuitive. I think the main problem with the segmentation pipeline you described is it's less intuitive than just lasso + crop.

David-Baddeley, 2020-05-18:
choosing crop with the freeform selection dialog now displays an error and directs the user to the manualSegment module for freeform masking.

zacsimile, 2020-05-18:
I think this is a sufficient solution for the moment. We can fold this into #226.

closed 2020-05-18 by zacsimile