python-microscopy/python-microscopy #252: Tidy up annotation tools (and make VisGUI compatible)

David-Baddeley, 2020-05-18:
There are currently 3 distinct but overlapping annotation modules in PYME (potentially a 4th if the NEP-Fitting line profile selection is included)

`manualSegment` - allows creation of a binary mask based on free-form selections which are filled. Requires shapely which can be tricky to install on windows. Annotations are stored as a binary image.
`svmLabel` - allows the creation of a label mask (with user chosen integer labels). Supports both free-form filled and line annotations, filled annotations do not use shapely, line annotations do. Annotations stored as an integer-valued label image.
`annotation` - vector rather than raster annotation, with annotations being stored as a list of paths with associated labels (allows displaying annotation as an overlay on image rather than in a separate window, and in principle, editing/deletion of annotations). Currently supports line and freeform curve annotation, with optional 'morphological snake' fitting (i.e. line can snap to edges or ridges). Does not depend on shapely. Can generate a raster label image (like that produced by `svmLabel`) as needed. Serialises to json.

The `svmLabel` and `annotation` modules are also used to train and run some simple ML based pixel classifiers.

My thoughts are to make the `annotation` module do everything (specifically by adding free-form filled and possibly point/ rectangle annotation modes). It would make sense to use the same logic for mask generation (`manualSegment`), but it is possible that the full annotation UI will be a bit unwieldily for that task.

Also on the TODO list is to add annotation selection and deletion.

zacsimile, 2020-05-18:
I like this path forward. 

I took at look at what we use from `shapely` and I think we could quickly write our own versions if we need to. On the other hand, it seems `shapely` install works fine on windows for `pip >= 8` and `python >= 3.5` (https://gis.stackexchange.com/questions/62925/shapely-not-installing-correctly), so this problem may disappear in the future.

David-Baddeley, 2020-05-18:
`skimage.draw` will also do a lot of what we were using shapely for (at least when writing into masks)

David-Baddeley, 2021-11-11:
See also #226

David-Baddeley, 2021-11-11:
I did not realise how far I'd got on the dh5view end - we should probably just go ahead and remove `manualSegment` and `svmLabel`