python-microscopy/python-microscopy #255: Request for a Units class

zacsimile, 2020-05-19:
Low priority: I think PYME could benefit from a general `Unit` class that stores arbitrary (pixels, bytes, meters) and allows you specify if it’s `giga`, `mega`, `kilo`, `hecto`, `deca`, `1.0`, `deci`, `centi`, `milli`, `micro`, `nano`, etc. Consider it a generalization of `voxelsize_nm` where we can pass numeric values as a `Unit` with `as_nm()` methods. I think this would save a lot of `*1e3`, `/1e9`, etc. in the code base.

zacsimile, 2020-05-19:
This could be extended to `DistanceUnit`, `StorageUnit`, `ADUnit`, etc. to satisfy the physicists among us.

David-Baddeley, 2020-05-19:
- I strongly suspect there is already a python library for this (if we don't need to reinvent the wheel we shouldn't)
- multiplying by 1e3, 1e9 etc is already fairly readable
- making everything an instance of some `Unit` (or `HasUnits`) class introduces a lot of programming overhead for what is not obviously a huge gain. Specifically, none of the API functions we call (e.g. to get free space) return a `HasUnits` instance, so we are going to need to wrap each call in a `HasUnits()` instantiation. 
- To be useful, we'd need to handle numpy arrays, and probably also lists and tuples as well as constants. We could potentially subclass ndarray, but doing this right is going to be really hard - there are loads of edge cases that would get us into trouble (e.g. structured arrays with mixed units).

What I think would ultimately be more useful is some way of annotating what units something has so that it's clear when reading the code. By far the simplest way of doing this is probably by adopting a variable naming convention where we always append the units to a variable name if the quantity has units - e.g. 

```python
free_space_bytes = get_free_space(dir)
free_space_GB=free_space_bytes/1e9
``` 

A naming convention is also likely to be the most robust  (for file size conversions it will also generate more easily readable code, as there are 2 possible definitions of the prefixes - e.g. MB can either be 1000 bytes or 1024 bytes and we would otherwise need to go digging in the source of the units module to work out which was used).

That said, I think we could potentially consider whether there is a (reasonably) simple way of adding some form of unit annotations to `Tabular` objects (as these can get serialised to disk, and/or passed around in ways which loose variable naming conventions).

I also think there could be a strong case for adding `pixel_coords_to_nm()` and `nm_coords_to_pixels()` methods to either the `ImageStack` class or `MetaDataHandlerBase`.

David-Baddeley, 2020-05-19:
I think we should also just define a standard set of units:
- distances are in nm
- times are in s
- file sizes are in bytes

zacsimile, 2020-05-19:
I agree, standardization and better commenting would fix this problem. I thought a class might be a good method of standardization. If we can adhere to the standard set of units across PYME, that would also work. 

Right now, it's not always obvious, even with newer functions. For example, the new Ripley's stuff uses `.voxelsize` instead of `.voxelsize_nm` because `.voxelsize` is in nm for masks. But, you have to know that. To figure that out, you have sort of the same problem as having to look for what was used to define MB (1000 or 1024). In the case of a `Units` class, though, there would always be one place to look.

David-Baddeley, 2020-05-19:
At the moment, everything is in nm except where it isn't :p. `ImageStack.voxelsize` is **always** in nm  (but is just a backwards compatible alias for `voxelsize_nm` in any case).  I'll add a deprecation warning for `ImageStack.voxelsize` and push people to the more explicit version.

I'm still undecided about how to handle some of the current inconsistent defaults, and whether we should be, e.g., changing all the column names in VisGUI.

zacsimile, 2020-05-19:
I think we're close on the nm side of the things. Incidentally, we should probably write `All units are in nm (except where they aren't)` somewhere early in the PYME documentation. Not sure what state we're in for tracking other units (e.g. what are the units in the `PYME.Acquire.Hardware.Simulator.fakeCam.NoiseMaker` class?). We may want to define a few more standards.

What column names might we need to change in VisGUI?

David-Baddeley, 2020-05-19:
As far as VisGUI goes, `x`, `y`, and `z` are in nm, but not explicitly so, `t` is in frames (not, e.g. seconds), but it seems a bit hard to change these.  `A` is a mess - it's in ADUs, but variably refers to the peak intensity or the area under the curve. We do have `nPhotons` as a semi-standardised derived amplitude variable, although that is currently not present for all fits.

I'm tempted to add a `t_s` column where the metadata for camera cycle time it is present, along with a `frame` column as an alias of the current `tIndex`/`t` columns.

barentine, 2020-05-20:
Everything is in nanometers except for PYMEAcquire, and some other things ;) We currently default e.g. all piezo's to micrometers, and voxelsize in metadata (rather than ImageStack attribute) is also in um
[image].

Intensity units in fit factories certainly vary, as David alludes.