python-microscopy/python-microscopy #256: wx version vision

barentine, 2020-05-20:
Having yet to fully step into wx4, I'm sure this will sit for quite some time, but wanted to mention that as long as we are sticking with wx, 4.1 includes check boxes in listctrl which would be nice for PYME's annotations, nep-fitting's line profile list, chained recipe ui, etc., so I would suggest we target that at some point.

David-Baddeley, 2020-05-20:
You've always been able to do check boxes in list controls in wx ... from the wx3 demos:

[image]

The particle tracking UI in dh5view used to use them, although it seems to be commented out at present.

barentine, 2020-05-20:
Weird, docs say `EnableCheckBoxes` is new in 4.1, but I should have dug deeper. Thanks @David-Baddeley

closed 2020-05-20 by barentine

David-Baddeley, 2020-05-20:
It used to be done through a mixin (multi-inheritance with `ListCtrl` and `CheckListMixin`), so it's possible that 4.1 has moved support to the core `ListCtrl`. I'm pretty sure things could be made a lot cleaner than the way they were supported in wx3 - when I was playing with them I think there were a few annoying restrictions (I seem to recall only being able to have checkboxes in the first column).

I'm not sure I see the use case in annotations and recipe UI.