python-microscopy/python-microscopy #259: pyface/traitsui bug

zacsimile, 2020-05-22:
Just did a fresh miniconda3 install on my Mac. Per our recent testing with cluster, I created a `python=3.6`, `numpy=`1.4`, `wxpython=4.0.4` environment. I ran `visgui` and tried to open a file and received the following error. The strange thing is, this set of versions of `pyface`/`traits`/`traitsui` is listed as working in some of our other issues.

<details>
<summary>OSX packages</summary>
<pre>
# Name                    Version                   Build  Channel
alabaster                 0.7.12                   py36_0  
appnope                   0.1.0            py36hf537a9a_0  
attrs                     19.3.0                     py_0  
babel                     2.8.0                      py_0  
backcall                  0.1.0                    py36_0  
backports                 1.0                        py_2  
backports.lzma            0.0.14           py36hd0afa68_0  
beautifulsoup4            4.9.0                    py36_0  
blas                      1.0                         mkl  
bleach                    3.1.4                      py_0  
blosc                     1.15.0               hd9629dc_0  
bzip2                     1.0.8                h1de35cc_0  
ca-certificates           2020.1.1                      0  
cairo                     1.14.12              hc4e6be7_4  
certifi                   2020.4.5.1               py36_0  
cffi                      1.14.0           py36hc512035_1  
chardet                   3.0.4                 py36_1003  
cheroot                   8.3.0                    py36_0  
cherrypy                  18.6.0                   py36_0  
cloudpickle               1.4.1                      py_0  
conda                     4.8.3                    py36_0  
conda-build               3.18.11                  py36_0  
conda-package-handling    1.6.1            py36h1de35cc_0  
cryptography              2.9.2            py36ha12b0ac_0  
cycler                    0.10.0           py36hfc81398_0  
cython                    0.29.17          py36h0a44026_0  
cytoolz                   0.10.1           py36h1de35cc_0  
dask-core                 2.16.0                     py_0  
dbus                      1.13.14              h517e14e_0  
decorator                 4.4.2                      py_0  
defusedxml                0.6.0                      py_0  
dispatch                  1.0                      py36_0    david_baddeley
django                    1.11.8           py36hd476221_0  
docutils                  0.16                     py36_0  
entrypoints               0.3                      py36_0  
enum-compat               0.0.2            py36h14911fc_0    david_baddeley
expat                     2.2.6                h0a44026_0  
fftw                      3.3.4                         0    david_baddeley
filelock                  3.0.12                     py_0  
fontconfig                2.13.0               h5d5b041_1  
freetype                  2.9.1                hb4e5f40_0  
future                    0.18.2                   py36_0  
gettext                   0.19.8.1             hb0f4f8b_2  
glib                      2.63.1               h70d4741_1  
glob2                     0.7                        py_0  
hdf5                      1.10.4               hfa1e0ec_0  
icu                       58.2                 h0a44026_3  
idna                      2.9                        py_1  
imageio                   2.8.0                      py_0  
imagesize                 1.2.0                      py_0  
importlib_metadata        1.5.0                    py36_0  
importlib_resources       1.4.0                    py36_0  
intel-openmp              2019.4                      233  
ipykernel                 5.1.4            py36h39e3cac_0  
ipython                   7.13.0           py36h5ca1d4c_0  
ipython_genutils          0.2.0                    py36_0  
ipywidgets                7.5.1                      py_0  
jaraco.classes            3.1.0                      py_0  
jaraco.collections        3.0.0                      py_0  
jaraco.functools          3.0.1                      py_0  
jaraco.text               3.2.0                    py36_0  
jedi                      0.17.0                   py36_0  
jinja2                    2.11.2                     py_0  
joblib                    0.15.1                     py_0  
[... 163 lines cut ...]
    import traitsui.api as tui
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traitsui/api.py", line 36, in <module>
    from .editors.api import ArrayEditor
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traitsui/editors/__init__.py", line 23, in <module>
    from .api import ArrayEditor
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traitsui/editors/api.py", line 24, in <module>
    from .code_editor import CodeEditor
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traitsui/editors/code_editor.py", line 37, in <module>
    class ToolkitEditorFactory(EditorFactory):
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traitsui/editors/code_editor.py", line 49, in ToolkitEditorFactory
    mark_color = Color(0xECE9D8)
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traits/trait_factory.py", line 40, in __call__
    return self.maker_function(*args, **metadata)
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traits/traits.py", line 717, in Color
    return ColorTrait(*args, **metadata)
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traitsui/toolkit_traits.py", line 8, in ColorTrait
    return toolkit().color_trait(*args, **traits)
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traitsui/toolkit.py", line 109, in toolkit
    _toolkit = find_toolkit('traitsui.toolkits')
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/pyface/base_toolkit.py", line 263, in find_toolkit
    return import_toolkit(ETSConfig.toolkit, entry_point)
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/pyface/base_toolkit.py", line 227, in import_toolkit
    raise RuntimeError(msg)
RuntimeError: No traitsui.toolkits plugin could be loaded for wx
```

David-Baddeley, 2020-05-22:
I suspect that the traitsui builds might be different on different platforms :(. This is going to be a right pain.

to summarise, the issue is `traitsui==6.1.3` on OSX. This is pointing more and more to at least needing to maintain our own packages for traitsui rather than relying on conda to do it properly.

zacsimile, 2020-05-23:
Ah, man, I already went through this once in #92. Sorry. Downgrading to `pyface=6.0.0` and `traits=5.1.1` fixed it. As long as I do this, I can leave `traitsui=6.1.3` and everything works.

zacsimile, 2020-05-23:
Maybe we should pin to these versions in `pyme-depends`?

barentine, 2020-05-24:
I was able to leave pyface at 6.1.2, and traitsui at 6.1.3, but downgrading traits from 6.0.0 -> 5.1.1 as you suggested fixed things.

zacsimile, 2020-06-20:
Fixed in python-microscopy/pyme-conda-recipes#5

closed 2020-06-20 by zacsimile