python-microscopy/python-microscopy #268: Implicit scipy<=1.3 dependency

zacsimile, 2020-06-07:
We are implicitly dependent on `scipy<=1.3`. I installed `scipy=1.4`. I received the following traceback.

```
/Users/zachcm/Code/python-microscopy/PYME/Acquire/acquiremainframe.py:109: wxPyDeprecationWarning: Call to deprecated item __call__. Use :meth:`EvtHandler.Bind` instead.
  wx.EVT_CLOSE(self, self.OnCloseWindow)
/Users/zachcm/Code/python-microscopy/PYME/Acquire/ui/splashScreen.py:49: wxPyDeprecationWarning: Call to deprecated item __call__. Use :meth:`EvtHandler.Bind` instead.
  wx.EVT_PAINT(self, self.OnPaint)
/Users/zachcm/Code/python-microscopy/PYME/Acquire/ui/splashScreen.py:124: wxPyDeprecationWarning: Call to deprecated item EmptyBitmap. Use :class:`wx.Bitmap` instead
  MemBitmap = wx.EmptyBitmap(s.GetWidth(), s.GetHeight())
DEBUG:PYME.Acquire.acquiremainframe:Init run, waiting on background threads
ERROR:PYME.Acquire.ExecTools:Error running background init task Fake Camera
Traceback (most recent call last):
  File "/Users/zachcm/Code/python-microscopy/PYME/misc/fftw_compat.py", line 6, in <module>
    from fftw3f import Plan, create_aligned_array
ModuleNotFoundError: No module named 'fftw3f'

Traceback (most recent call last):
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/Users/zachcm/Code/python-microscopy/PYME/Acquire/ExecTools.py", line 149, in _bginit
    raise e
  File "/Users/zachcm/Code/python-microscopy/PYME/Acquire/ExecTools.py", line 140, in _bginit
    self.codeObj(defLocals['scope'])
  File "/Users/zachcm/Code/python-microscopy/PYME/Acquire/Scripts/init.py", line 55, in cm
    from PYME.Acquire.Hardware.Simulator import fakeCam
  File "/Users/zachcm/Code/python-microscopy/PYME/Acquire/Hardware/Simulator/fakeCam.py", line 24, in <module>
    from . import rend_im
  File "/Users/zachcm/Code/python-microscopy/PYME/Acquire/Hardware/Simulator/rend_im.py", line 42, in <module>
    from PYME.Deconv.wiener import resizePSF
  File "/Users/zachcm/Code/python-microscopy/PYME/Deconv/wiener.py", line 29, in <module>
    import PYME.misc.fftw_compat as fftw3f
  File "/Users/zachcm/Code/python-microscopy/PYME/misc/fftw_compat.py", line 8, in <module>
    import pyfftw
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/pyfftw/__init__.py", line 43, in <module>
    from . import interfaces
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/pyfftw/interfaces/__init__.py", line 245, in <module>
    from . import scipy_fftpack
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/pyfftw/interfaces/scipy_fftpack.py", line 65, in <module>
    from scipy.fftpack import (dct, idct, dst, idst, diff, tilbert, itilbert,
ImportError: cannot import name '_fftpack'
```

Downgrading to `scipy=1.3` fixed this. Scipy changed its semi-internal interface to fftpack in 1.4: https://github.com/labsyspharm/ashlar/issues/59.

David-Baddeley, 2020-06-16:
I guess we have 2 options - pin scipy in pyme, or pin it in the conda package for pyfftw which we build (or possibly update pyfftw if they have fixed this in the meantime),

barentine, 2020-06-16:
I can't find where I posted this issue before, but yes pyfftw has already fixed it. Don't know whether someone's hosting it on conda at this point or just in source right now.

zacsimile, 2020-06-16:
We'll need to find that version of pyfftw, then, and pin to that. Default install from conda channels (not conda-forge) on python 3 is busted. Need to downgrade scipy to 1.13 to get it to work (or maybe upgrade pyfftw, who knows?).

barentine, 2020-06-17:
I knew I'd written it somewhere! I mentioned it in what has become the new py3 super-issue: #160 
Anyway, relinking the pyfftw issue [here](https://github.com/pyFFTW/pyFFTW/issues/279), which was [fixed with their release 0.12.0 ](https://github.com/pyFFTW/pyFFTW/issues/280). As advertised it is on pypi and conda-forge, though I'm not seeing it in any non conda-forge channel.

I've found pip installing from our requirements.txt on py3 quite smooth, but David has mentioned some points in favor of sticking with conda for ~releases. I suspect some of conda version pain will go away when we can be py3 or bust and don't have to straddle quite so far back.

closed 2020-11-07 by zacsimile