python-microscopy/python-microscopy #274: Cannot properly visualize tracked data filtered by track length

zacsimile, 2020-06-15:
Since we switched from `ClumpManager` to a tabular datasource `with_tracks` for tracks layers, we display all tracks, regardless of what we enter for `Min clump size` in `Analysis>Tracking>Track single molecule trajectories`. The `Output Filter` only adjusts the output layer, which is a points layer. To visualize tracks of length `Min clump size` or greater, one has to add a `FilterTable` on `clumpSize` to the pipeline and then add a tracks layer referencing this output. It should be easier to visualize only tracks of length `Min clump size`. The easiest thing to do would be to re-instate `ClumpManager` output from `Analysis>Tracking>Track single molecule trajectories`. We should explore other ideas as well (e.g. auto-recast `output` to the proper layer type).

David-Baddeley, 2020-06-16:
The tracks data source is a points source, so it's entirely valid to select it as the current data source and make the tracks layer also point to output (which passes through the output filter). This would be a 1:1 reimplementation of the way tracks were implemented previously. That said, I can see an argument for not having to make the tracks the output. 

I dislike the `ClumpManager` solution for visualisation for a number of reasons:
- changing the minimum clump size forces everything to be re-tracked (which has the potential to be a major performance problem)
- in a dense tracking scenario we will create and store a large number of track objects (potentially also a performance issue)
- the `with_clumps` format (i.e. raw data with a column containing track assignments) is easier to export / save and a closer match to what you get from 3rd party tracking software - e.g. ImageJ, whereas `ClumpManager` currently has no obvious serialisation.

I think there are a few potential solutions, but I think that it's imperative that any filtering happens after the tracking and without needing to re-run the tracking module:

1) make the UI automatically add a `FilterTable` module after the tracking module
2) `selectDataSource()` with the tracking output and make the tracks layer point to `output` (this would be a temporary solution as the output filter will probably go away at some point). 
3) make the minimum track size a property of the track visualisation layer.
4) re-write `ClumpManager` to be lazy, just maintaining an index into a `with_tracks` tabular object, owning a`clump_size` parameter (or taking it as a parameter to a `get_clumps()` call), and returning clump objects on the fly.

zacsimile, 2020-06-16:
5. Automatically add a `FilterTable` after `with_tracks` and create a second datasource called `filtered_tracks` that points to the output of this filter and is passed to the tracks layer. Sort of a mix of 1 and 4.

2 is a good option, although not as good if `output` is going away. I dislike 3 because I prefer to keep the filtering and layer logic separate. I think 4 is a good option, but we could just do option 5 and do away with clump manager entirely.

David-Baddeley, 2020-06-16:
number 5 is what I meant with number 1 (I just didn't write it out in full). Probably what I'd implement as a first step.

The clump manager is currently used in the dh5view tracking workflow where the track objects make sense for the way HTML tracking reports are generated, but I don't think it's super useful in the VisGUI case (at least until we are routinely doing MSD plots etc).

closed 2020-06-24 by zacsimile