python-microscopy/python-microscopy #275: dh5view has no uncheck/remove module functionality

zacsimile, 2020-06-16:
Some of the code is there, but the way it's written we'll have to write a destroyer function, `Unplug`, for every module. It should be possible to uncheck a module from the `Module` menu and have it disappear.

David-Baddeley, 2020-06-16:
How important is this as a functionality? Trying to remove any of the GUI stuff is going to be a major headache. When I initially wrote the module stuff I though about adding a remove function but when thinking about what that would involve I stopped and asked if there was really a situation where you would **need** to unload a module. The main consideration would be to unclutter the panel on the left of the image, but that is at least partially facilitated by having the panels foldable.

The UE is admittedly a bit off (based on how it's presented, you might expect to be able to uncheck a module), but I'm not sure how much we should care. The rationale for having modules manually loaded rather than loading everything off the bat was to facilitate a faster launch (which is a bit moot at the moment due to the clusterIO import), and, to some extent, to avoid un-necessarily cluttering the UI. I guess there are 3 questions:

- Do we actually need manual loading or should we load everything automatically?
- Is module unloading a desirable feature?
- Can we make the module menu UE a bit less confusing?

zacsimile, 2020-06-16:
It's not terribly important. It's a little confusing that loaded modules are checkable, but not uncheckable. I don't think we should be able to load everything by default. What we have now seems fine, I just noticed that there clearly was code present with the intent of unloading/unchecking modules. I would expect the ability to load/unload modules at will, since that's a standard feature in programs like ImageJ, Photoshop, etc., but it's not required or high-priority.

David-Baddeley, 2020-06-16:
ImageJ loads everything at launch time, as does photoshop. I think what you are referring to is toggling visibility (which might actually be easier than removing bits from the UI). To improve the UE, we could potentially disable a modules menu entry once it had been loaded.

zacsimile, 2020-06-16:
Ah, I didn't realize that. Yes, I think toggling visibility is the thing. I looked into it briefly earlier today. It seemed it would be rather difficult to toggle the visibility of the menus that get added, but I suspect we could toggle visibility of the panes no problem.

zacsimile, 2020-06-16:
Sorry, went off on a bit of a tangent there. Disabling the menu entry might work, although I'm not opposed to leaving it as-is. I like the look of the check.

David-Baddeley, 2020-06-16:
[image]

Disabling when loaded looks like it might work, although it removes some developer functionality. At present, selecting a loaded module re-loads it so you can edit module code and reload it on the fly whilst debugging new modules.

zacsimile, 2020-06-16:
That does look more user-friendly. Maybe we could add a `-d` (developer mode) flag to `dh5view` that still allows module reload when laucnhed?

David-Baddeley, 2020-06-16:
or put a `Reload` submenu down the bottom of the modules menu.

zacsimile, 2020-06-16:
Yeah, that sounds good.