python-microscopy/python-microscopy #277: Quitting dh5view is difficult

zacsimile, 2020-06-16:
Both `PYMEVisualize` and `PYMEAcquire` will exit on a `Command+Q` or pressing the `X` for exit. `dh5view`, by contrast, can only be closed with a `Ctrl+C` in the shell.

David-Baddeley, 2020-06-16:
Which platform, and how was dh5view launched? - `Command+Q` and/or `Alt-F4` have always worked for me.

zacsimile, 2020-06-16:
OSX. Launched from the command line. Python 3. I had opened a bioformats file. I repeatedly tested to make sure this was a consistent issue.

David-Baddeley, 2020-06-16:
My guess is that it's a bioformats issue and has to do with not disposing of the jvm properly - `dh5view -t` quits just fine on OSX and py3.

zacsimile, 2020-06-16:
Yes, that works for me, too.

David-Baddeley, 2020-06-26:
Should hopefully be fixed in 44e3c44d77b3deb8c359b8e87394e6e2f890d2f3 . Error would seem to not be bioformats related, but was rather a failure to call `dlg.Destroy()` on the voxel size dialog which pops up if voxel size is not in image metadata. After bugging you about remembering this with the message dialog, we've finally found somewhere where it really matters and it was a bug in my use of dialogs.

For future reference:
- wx event loop does not finish while there are still top level windows (anything which has `parent=None` and any dialogs)
- ordinary frames are destroyed when they are closed
- dialogs are not (unless you explicitly destroy them) - there might be some performance reason behind this - e.g. permitting lightweight reuse of a regularly used dialog, but it doesn't really matter.
- debugging insight was to call `wx.GetTopLevelWindows()` in the shell for the bioformats and `dh5view -t` cases. The voxel size dialog popped up in this list in the bioformats case (even after it had been closed).

zacsimile, 2020-06-26:
Thanks, @David-Baddeley! And thanks for the live demo of why we should care about proper handling of message dialogs!

The Java VM still hangs the program even after this fix. The `BioformatsFile` never gets garbage collected for some reason, so we never properly kill the VM. I have a fix, which I'll push in a second. I thought it didn't work when I tried it yesterday, but the program wasn't closing after the fix because of this dialog issue. The program now exits no problem.