python-microscopy/python-microscopy #278: editor improvement for recipe modules with dict inputs

barentine, 2020-06-17:
Something like a `DictInputModule(ModuleBase)` would be helpful for things like  which takes a dict input rather than the standard str, e.g. all modules with `self.inputVariables = DictStrStr()` like `output.HDFOutput`.

While one could think of potential choices to populate a drop down, e.g. [json.dumps({k:k}) for k in namespace.keys()], the current `CBEditor` is just a convenient way to throw an error from the gui:
[image]
[image]

David-Baddeley, 2020-06-18:
The dictionary input on HDFOutput is a bit of an aberration, and not something I'd want to encourage as a pattern. We only get away with it because it's an output module and handled rather differently than standard modules. Rather than subclassing ModuleBase, I think the two reasonable options are:

1) re-jig HDFOutput so that it doesn't need a dictionary input
2) write a custom editor and view specifically for HDFOutput

barentine, 2020-07-23:
On windows 10, wx 4.1.0, py3.6, traits 6.1.0, traitsui 7.0.0 we currently get the following when trying to add an HDFOutput module in bakeshop. I can't remember if I'd also hit it on my laptop above and just 'fixed' it by casting self.values to str first or not.
```
Traceback (most recent call last):
  File "c:\users\bergamot\code\python-microscopy\PYME\recipes\recipeGui.py", line 520, in OnAddModule
    if c.configure_traits(kind='modal'):
  File "C:\Users\Bergamot\Anaconda2\envs\pyme\lib\site-packages\traits\has_traits.py", line 2015, in configure_traits
    args,
  File "C:\Users\Bergamot\Anaconda2\envs\pyme\lib\site-packages\traitsui\wx\toolkit.py", line 219, in view_application
    context, view, kind, handler, id, scrollable, args
  File "C:\Users\Bergamot\Anaconda2\envs\pyme\lib\site-packages\traitsui\wx\view_application.py", line 101, in view_application
    args=args,
  File "C:\Users\Bergamot\Anaconda2\envs\pyme\lib\site-packages\traitsui\view.py", line 462, in ui
    ui.ui(parent, kind)
  File "C:\Users\Bergamot\Anaconda2\envs\pyme\lib\site-packages\traitsui\ui.py", line 246, in ui
    self.rebuild(self, parent)
  File "C:\Users\Bergamot\Anaconda2\envs\pyme\lib\site-packages\traitsui\wx\toolkit.py", line 133, in ui_modal
    ui_modal.ui_modal(ui, parent)
  File "C:\Users\Bergamot\Anaconda2\envs\pyme\lib\site-packages\traitsui\wx\ui_modal.py", line 44, in ui_modal
    _ui_dialog(ui, parent, BaseDialog.MODAL)
  File "C:\Users\Bergamot\Anaconda2\envs\pyme\lib\site-packages\traitsui\wx\ui_modal.py", line 63, in _ui_dialog
    BaseDialog.display_ui(ui, parent, style)
  File "C:\Users\Bergamot\Anaconda2\envs\pyme\lib\site-packages\traitsui\wx\ui_base.py", line 71, in display_ui
    ui.owner.init(ui, parent, style)
  File "C:\Users\Bergamot\Anaconda2\envs\pyme\lib\site-packages\traitsui\wx\ui_modal.py", line 137, in init
    sw = panel(ui, window)
  File "C:\Users\Bergamot\Anaconda2\envs\pyme\lib\site-packages\traitsui\wx\ui_panel.py", line 272, in panel
    panel, content[0], ui
  File "C:\Users\Bergamot\Anaconda2\envs\pyme\lib\site-packages\traitsui\wx\ui_panel.py", line 431, in fill_panel_for_group
    panel, group, ui, suppress_label, is_dock_window, create_panel
  File "C:\Users\Bergamot\Anaconda2\envs\pyme\lib\site-packages\traitsui\wx\ui_panel.py", line 587, in __init__
    self.add_items(content, panel, self.sizer)
  File "C:\Users\Bergamot\Anaconda2\envs\pyme\lib\site-packages\traitsui\wx\ui_panel.py", line 909, in add_items
    editor.prepare(item_panel)
  File "C:\Users\Bergamot\Anaconda2\envs\pyme\lib\site-packages\traitsui\editor.py", line 240, in prepare
    self.init(parent)
  File "c:\users\bergamot\code\python-microscopy\PYME\ui\custom_traits_editors.py", line 24, in init
    style = wx.CB_DROPDOWN, value=self.value, choices=self.factory.choices)
TypeError: ComboBox(): arguments did not match any overloaded call:
  overload 1: too many arguments
  overload 2: argument 'value' has unexpected type 'TraitDictObject'
```

barentine, 2020-07-23:
changing line 24 in init to `value=str(self.value)` indeed lets us add the module, and gets us back to the error I'd reported at the top of this thread

David-Baddeley, 2020-08-18:
Editor fix in #405 gets us most of the way, but is a bit clunkier/slower than the default string based editor that was working fine on py2. Suggest we leave this issue open for now. Ultimately transitioning to an `EditListCtrl` or similar would likely fix this for good.

barentine, 2020-08-19:
Would note that the str/dict type error within the CBEditor was py2/3, but even the old editor was broken on py2 if you used the dropdown to select an input (`str` vs `DictStrStr`), still, `EditListCtrl` probably makes more sense long term