python-microscopy/python-microscopy #279: hdf5 file mode/lock issue in chaining recipe after localization

barentine, 2020-06-18:
opening an hdf5 file in read mode for a recipe immediately after it's been created/opened in append mode for localization fails 
```
DEBUG:ruleserver:adding chained rule
DEBUG:ruleserver:chained rule ID 000007-83b5df21d72a40fd9a862e0a7a85ae79
DEBUG:ruleserver:on completion present: False
DEBUG:ruleserver:000004-7e13c4b0cc864a2780a65a9ec5dd505e, n_max: 1, n_completed: 0, n_assigned: 0, n_failed: 1
DEBUG:ruleserver:000005-0623b7e4630047bb887397f4f6c52729, n_max: 1000000, n_completed: 0, n_assigned: 0, n_failed: 0
DEBUG:ruleserver:000006-0e31f41ec2c34e0d9767193b070211ef, n_max: 150, n_completed: 66, n_assigned: 84, n_failed: 0
HDF5-DIAG: Error detected in HDF5 (1.10.4) thread 123145318404096:
  #000: H5F.c line 509 in H5Fopen(): unable to open file
    major: File accessibilty
    minor: Unable to open file
  #001: H5Fint.c line 1400 in H5F__open(): unable to open file
    major: File accessibilty
    minor: Unable to open file
  #002: H5Fint.c line 1615 in H5F_open(): unable to lock the file
    major: File accessibilty
    minor: Unable to open file
  #003: H5FD.c line 1640 in H5FD_lock(): driver lock request failed
    major: Virtual File Layer
    minor: Can't update object
  #004: H5FDsec2.c line 941 in H5FD_sec2_lock(): unable to lock file, errno = 35, error message = 'Resource temporarily unavailable'
    major: File accessibilty
    minor: Bad file ID accessed
Traceback (most recent call last):
  File "/Users/Andrew/code/python-microscopy/PYME/cluster/taskWorkerHTTP.py", line 337, in computeLoop
    recipe.loadInput(url, key)
  File "/Users/Andrew/code/python-microscopy/PYME/recipes/base.py", line 897, in loadInput
    with tables.open_file(fn, mode='r') as h5f:
  File "/opt/miniconda3/lib/python3.6/site-packages/tables/file.py", line 320, in open_file
    return File(filename, mode, title, root_uep, filters, **kwargs)
  File "/opt/miniconda3/lib/python3.6/site-packages/tables/file.py", line 784, in __init__
    self._g_new(filename, mode, **params)
  File "tables/hdf5extension.pyx", line 492, in tables.hdf5extension.File._g_new
tables.exceptions.HDF5ExtError: HDF5 error back trace

  File "H5F.c", line 509, in H5Fopen
    unable to open file
  File "H5Fint.c", line 1400, in H5F__open
    unable to open file
  File "H5Fint.c", line 1615, in H5F_open
    unable to lock the file
  File "H5FD.c", line 1640, in H5FD_lock
    driver lock request failed
  File "H5FDsec2.c", line 941, in H5FD_sec2_lock
    unable to lock file, errno = 35, error message = 'Resource temporarily unavailable'

End of HDF5 error back trace

Unable to open/create file '/Users/Andrew/PYMEData/Andrew/2020_6_18/analysis/18_6_series_BC.h5r'
```

David-Baddeley, 2020-06-18:
hmm, probably best worked around by making `ModuleCollection.loadInput()` use `h5rfile` rather than pytables directly.

barentine, 2020-06-18:
Didn't go loads (pun) better that time
```
HDF5-DIAG: Error detected in HDF5 (1.10.4) thread 123145550778368:
  #000: H5F.c line 509 in H5Fopen(): unable to open file
    major: File accessibilty
    minor: Unable to open file
  #001: H5Fint.c line 1400 in H5F__open(): unable to open file
    major: File accessibilty
    minor: Unable to open file
  #002: H5Fint.c line 1615 in H5F_open(): unable to lock the file
    major: File accessibilty
    minor: Unable to open file
  #003: H5FD.c line 1640 in H5FD_lock(): driver lock request failed
    major: Virtual File Layer
    minor: Can't update object
  #004: H5FDsec2.c line 941 in H5FD_sec2_lock(): unable to lock file, errno = 35, error message = 'Resource temporarily unavailable'
    major: File accessibilty
    minor: Bad file ID accessed
Traceback (most recent call last):
  File "/Users/Andrew/code/python-microscopy/PYME/cluster/taskWorkerHTTP.py", line 337, in computeLoop
    recipe.loadInput(url, key)
  File "/Users/Andrew/code/python-microscopy/PYME/recipes/base.py", line 893, in loadInput
    with h5rFile.openH5R(fn, mode='r')._h5file as h5f:
  File "/Users/Andrew/code/python-microscopy/PYME/IO/h5rFile.py", line 31, in openH5R
    file_cache[key] = H5RFile(filename, mode)
  File "/Users/Andrew/code/python-microscopy/PYME/IO/h5rFile.py", line 45, in __init__
    self._h5file = tables.open_file(filename, mode)
  File "/opt/miniconda3/lib/python3.6/site-packages/tables/file.py", line 320, in open_file
    return File(filename, mode, title, root_uep, filters, **kwargs)
  File "/opt/miniconda3/lib/python3.6/site-packages/tables/file.py", line 784, in __init__
    self._g_new(filename, mode, **params)
  File "tables/hdf5extension.pyx", line 492, in tables.hdf5extension.File._g_new
tables.exceptions.HDF5ExtError: HDF5 error back trace

  File "H5F.c", line 509, in H5Fopen
    unable to open file
  File "H5Fint.c", line 1400, in H5F__open
    unable to open file
  File "H5Fint.c", line 1615, in H5F_open
    unable to lock the file
  File "H5FD.c", line 1640, in H5FD_lock
    driver lock request failed
  File "H5FDsec2.c", line 941, in H5FD_sec2_lock
    unable to lock file, errno = 35, error message = 'Resource temporarily unavailable'

End of HDF5 error back trace

```

I'm a bit confused since we use h5rFile in the dataserver, but will look into it.

David-Baddeley, 2020-06-18:
might not be the error I thought it was (I thought the file was open for writing in the same process, which `h5rfile` would solve) - it looks like this is the standard problem of opening from another process which sometimes works and sometimes doesn't depending on how lucky you are (i.e. you can't be writing to the file when you try and open it). Two possible solutions:

- add a small delay to ensure that writing is finished before loading the file (potentially safer, likely better performance on the single computer scenario, and clearer how to deal with metadata, events, etc ...)
- use the h5r part endpoint on the dataserver

barentine, 2020-06-19:
Looks like things take longer than we'd like. An os.path.getmtime before the open call in h5rfile shows we can repeatedly hit the error even 5 seconds after the last-modified time (note that I'm working with tiny h5r files at the moment to boot).

In h5rfile I added a second retry with a full keep-alive time (20 s) sleep. It works, but seems excessive. Trying lower times, however, like 5 s (which is still kind of long) hits the same error. 

As you mention the tabular-part endpoint should work without delay (just maybe slower) since that's the process that opened the file last. Alternatively, could we try `unifiedIO` first, and if that fails with the `tables.exceptions.HDF5ExtError` we try a `clusterIO.get_file()`?

barentine, 2020-07-23:
fixed in #281 - as noted by David's comments in the merge it could be a little less weird, but OK for now

closed 2020-07-23 by barentine