python-microscopy/python-microscopy #283: pixel_size argument for PYME.LMVIs.gl_render3D_shaders incorrectly crops viewport

zacsimile, 2020-06-20:
The obvious fix is to change the zoom factor when we change the image bounds, but it feels a little hacky. We should restore the viewport to its original state after taking the snapshot.

David-Baddeley, 2020-06-23:
This needs a little more info - I assume it refers to the `getIm()` function. `getIm()` is a bit of a mess having been hacked about multiple times. As it stands, the zoom (and hence pixel size) is set by the `setView()` call on line 925, although we don't currently restore it after calling `getSnapshot()` which is a little silly.

David-Baddeley, 2020-06-23:
The caveat here is that, as we use the 2D bounding box to set the view extent (see line 919), all bets are off for 3D rotated views.

zacsimile, 2020-06-24:
Ah, fair enough. The main reason I want this to work is I want the option for the user to scale the size of the output video in animation. Maybe an easier move would be to pass a user-specified `scale` and use `scipy.ndimage.zoom` on the output we already get from `getSnapshot` and call that good for now.

zacsimile, 2020-06-24:
Is this something we want to fix for 3D rotated views? Do we need it? You already pointed that we can just resize the viewing window for animations (see #282). Maybe we could do the same thing for scaled snapshots and then just scale the viewing window back. The downside to this is we can't scale the window any larger than the screen. I think what Meshlab does is scale the GL canvas up to whatever scale factor you choose (past the screen width, if need be), snapshot, and then scale the canvas back down to size. We could try that. In this case, I'd move away from using the `pixel_size` argument to `scale` argument.

David-Baddeley, 2021-11-11:
This (or a related problem) has recently been flagged on #224 - definitely something we should revisit

PR #1162 (2021-12-06): Snapshot scaling
An initial GUI for setting scaling (dpi) for snapshots, should fix the snapshot aspect of #224. Fixes #283.Animation stuff hopefully to follow ...
closed 2021-12-06 by David-Baddeley