python-microscopy/python-microscopy #291: z stepping (and maybe some other stuff) currently disabled

barentine, 2020-06-25:
I suspect the spool controller settings.update refactor missed a bit
[image]

`stack` is a startspool input parameter and defaults to `False`:
[image]

barentine, 2020-06-25:
as mentioned in #292 there is also a bug for the compression setting.

Not mentioned there is that spool controller's z_dwell attribute will now override whatever you have in the protocol. I'm actually a huge fan of being able to override the protocol z_dwell, but we should probably make it optional and controllable through the gui

David-Baddeley, 2020-06-26:
I'm not too worried about over-riding the protocol z_dwell - solution is probably just to remove/deprecate the z_dwell argument to the protocol.

barentine, 2020-06-30:
Would it be OK to ~complete the settings-dict refactor on `SpoolController` and replace all the kwargs in `start_spooling` with a single `settings` parameter (dict) that gets passed to `update_settings` first thing in `start_spooling`? To ensure it's fast we might also want to change `update_settings` doesn't pop / call set methods unless it actually needs to

barentine, 2020-07-23:
Fixed with merges of #292, #327 , and #285

closed 2020-07-23 by barentine