python-microscopy/python-microscopy #296: Clean up the logo

zacsimile, 2020-06-28:
The PYME logo, stored at `PYME/resources/icons/pymeLogo.png`, is a bit messy. There is a tiny orange blob in the lower-right corner. The drop shadow on the bottom left is unceremoniously cut off. I think that short-term we should extract the contours of the microscope and snake using Illustrator (or comparable software), and create a clean, vector-version of the logo. I'm happy to do this. Long term, we should find a graphic designer to update the logo.

David-Baddeley, 2020-06-28:
I should have a vector version of this floating around somewhere ...

David-Baddeley, 2020-06-28:
but can't easily find it - I suspect that it pre-dates my current computer.