python-microscopy/python-microscopy #300: [ENHANCEMENT] Does this change how users interact with the software? How will these changes be communicated?

zacsimile, 2020-06-30:
**Is your feature request related to a problem? Please describe.**
This PR bullet point sounds to me more like a general problem to solve. I think we should move this out of the PR and create a permanent solution.

**Describe the solution you'd like**
Ship PYME with a "What's New?" dialog and/or build notes.

**Describe alternatives you've considered**
Develop PYMEy the snake, an animated cartoon wizard that points out new features with it's tail and says, "sssee this?"

**Additional context**

**Are you prepared to work on it?**
Sure.

David-Baddeley, 2020-06-30:
Or document the rationale behind it a bit better in the contributing docs (on my TODO list). I put it in the PR template is just as a mental have you thought about this? checkbox. It was partially in response to some proposed changes of Andrews which would have dramatically changed the acquisition workflow and required retraining of around 20 users across 4 labs. That's a bit of a non-starter for me unless there is a) an incredibly strong argument for change and b) some way of leading people gently into it.

The solutions can and probably should vary depending on what the PR is. For most PRs (e.g. bugfixes, new functionality, etc ... ) a valid response is that you've thought about it and this doesn't materially change existing workflows. For things which do represent major changes to existing workflows, some combination of the following are valid approaches:
- new workflow is strictly opt-in with old workflow as default
- new workflow is well documented with extensive end user documentation (ideally written at a level that the worst intern/undergrad in the lab can understand - not everyone is as lucky with the calibre of students etc as you are at Yale)
- build notes, what's new dialogs or similar with links to the new documentation - but remember that the average user does not read release notes, so this cannot be the only solution.

In general I'm not so sure that the checkboxes on the PRs should be checkboxes, but I do think that some form of reminder of what likely sticking points / areas which need justification are is useful.

zacsimile, 2020-06-30:
It's definitely work thinking about. Maybe it's something that belongs in `Contributing.rst`? There's no harm in leaving it in the PRs, it just may often not be checked off.

closed 2022-05-11 by zacsimile