python-microscopy/python-microscopy #304: [ENHANCEMENT] Automatically detect bioformats TIFF

zacsimile, 2020-06-30:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when `.tif` defaults to a `TiffDataSource`, but, because it's a bioformats TIFF, `getSlice()` fails. Changing to `.tiff` fixes this. (However, it's worth noting that Bioformats sometimes has a problem detecting multicolor TIFFs, see https://trello.com/c/kDDb9pEu/365-multicolor-tiff-issues-with-new-olympus, and I've run into this exporting multicolor TIFFs from old versions of ImageJ). I think we should be able to automatically detect the TIFF type.

**Describe the solution you'd like**
Automatically decide if a `.tif` or `.tiff` is bioformats (use `BioformatsDataSource`) or otherwise (use `TiffDataSource`).

**Describe alternatives you've considered**
We could leave it as-is and document `.tif` vs `.tiff` usage. I think the only "documentation" we have on this is the logger warning and dialog boxes that come up starting around line 1057 of `PYME.IO.image.py`.

**Additional context**

**Are you prepared to work on it?**
Sure.

David-Baddeley, 2020-06-30:
What specifically is the issue? Where did the data come from? Could you give a traceback?

TiffDataSource should handle OME TIFF just fine, and is significantly faster and more robust than bioformats. We should really only be using bioformats for things which are saved nominally as TIFF, but not following any of the common standards (i.e. OME etc ...) but for which bioformats has some special case logic. These will be very hard to detect automagically, and I don't think that manually having to re-name these is the end of the world. 

If it's a more general problem with loading e.g. multichannel OME then I think we should fix out loader. I'm really keen on keeping bioformats as an option of last resort.

David-Baddeley, 2020-06-30:
or an example of an offending file ...

zacsimile, 2020-06-30:
Hi David. Yes, sorry for not providing an example. My main point with this issue is I think it would be nice if we could automatically decide which loader to use regardless of whether the TIFF extension is `.tif` or `.tiff`. Here is a traceback I get when I try to hit play in `dh5view` after loading a `.tif` stack created from exported images from our animation software in ImageJ. If I change the extension to `.tiff`, the bioformats loader takes over and pressing play works just fine. Github doesn't support TIFF file uploads, so I'll just email you the file.

<details>
<summary>Traceback</summary>
<pre>
Traceback (most recent call last):
  File "/Users/zachcm/Code/python-microscopy/PYME/DSView/scrolledImagePanel.py", line 58, in OnPaint
    self.renderer(MemDC);
  File "/Users/zachcm/Code/python-microscopy/PYME/DSView/arrayViewPanel.py", line 497, in DoPaint
    im = self.Render(fullImage)
  File "/Users/zachcm/Code/python-microscopy/PYME/DSView/arrayViewPanel.py", line 1107, in Render
    seg = self.do.ds[x0_:(x0_+sX_):step,y0_:(y0_+sY_):step,int(self.do.zp), chan].squeeze().T
  File "/Users/zachcm/Code/python-microscopy/PYME/IO/DataSources/BaseDataSource.py", line 137, in __getitem__
    r.append(np.concatenate([np.atleast_2d(self.getSlice(i)[keys[0], keys[1]])[:,:,None] for i in indices], 2))
  File "/Users/zachcm/Code/python-microscopy/PYME/IO/DataSources/BaseDataSource.py", line 137, in <listcomp>
    r.append(np.concatenate([np.atleast_2d(self.getSlice(i)[keys[0], keys[1]])[:,:,None] for i in indices], 2))
  File "/Users/zachcm/Code/python-microscopy/PYME/IO/DataSources/BufferedDataSource.py", line 52, in getSlice
    sl = self.dataSource.getSlice(ind)
  File "/Users/zachcm/Code/python-microscopy/PYME/IO/DataSources/TiffDataSource.py", line 99, in getSlice
    return self.im[0].asarray(False, False)[0, 0, :,:,ind].squeeze()
IndexError: index 3 is out of bounds for axis 4 with size 3
Traceback (most recent call last):
  File "/Users/zachcm/Code/python-microscopy/PYME/DSView/scrolledImagePanel.py", line 58, in OnPaint
    self.renderer(MemDC);
  File "/Users/zachcm/Code/python-microscopy/PYME/DSView/arrayViewPanel.py", line 497, in DoPaint
    im = self.Render(fullImage)
  File "/Users/zachcm/Code/python-microscopy/PYME/DSView/arrayViewPanel.py", line 1107, in Render
    seg = self.do.ds[x0_:(x0_+sX_):step,y0_:(y0_+sY_):step,int(self.do.zp), chan].squeeze().T
  File "/Users/zachcm/Code/python-microscopy/PYME/IO/DataSources/BaseDataSource.py", line 137, in __getitem__
    r.append(np.concatenate([np.atleast_2d(self.getSlice(i)[keys[0], keys[1]])[:,:,None] for i in indices], 2))
  File "/Users/zachcm/Code/python-microscopy/PYME/IO/DataSources/BaseDataSource.py", line 137, in <listcomp>
    r.append(np.concatenate([np.atleast_2d(self.getSlice(i)[keys[0], keys[1]])[:,:,None] for i in indices], 2))
  File "/Users/zachcm/Code/python-microscopy/PYME/IO/DataSources/BufferedDataSource.py", line 52, in getSlice
    sl = self.dataSource.getSlice(ind)
  File "/Users/zachcm/Code/python-microscopy/PYME/IO/DataSources/TiffDataSource.py", line 99, in getSlice
    return self.im[0].asarray(False, False)[0, 0, :,:,ind].squeeze()
IndexError: index 6 is out of bounds for axis 4 with size 3
Traceback (most recent call last):
  File "/Users/zachcm/Code/python-microscopy/PYME/DSView/scrolledImagePanel.py", line 58, in OnPaint
    self.renderer(MemDC);
  File "/Users/zachcm/Code/python-microscopy/PYME/DSView/arrayViewPanel.py", line 497, in DoPaint
    im = self.Render(fullImage)
  File "/Users/zachcm/Code/python-microscopy/PYME/DSView/arrayViewPanel.py", line 1107, in Render
    seg = self.do.ds[x0_:(x0_+sX_):step,y0_:(y0_+sY_):step,int(self.do.zp), chan].squeeze().T
  File "/Users/zachcm/Code/python-microscopy/PYME/IO/DataSources/BaseDataSource.py", line 137, in __getitem__
    r.append(np.concatenate([np.atleast_2d(self.getSlice(i)[keys[0], keys[1]])[:,:,None] for i in indices], 2))
  File "/Users/zachcm/Code/python-microscopy/PYME/IO/DataSources/BaseDataSource.py", line 137, in <listcomp>
    r.append(np.concatenate([np.atleast_2d(self.getSlice(i)[keys[0], keys[1]])[:,:,None] for i in indices], 2))
  File "/Users/zachcm/Code/python-microscopy/PYME/IO/DataSources/BufferedDataSource.py", line 52, in getSlice
    sl = self.dataSource.getSlice(ind)
  File "/Users/zachcm/Code/python-microscopy/PYME/IO/DataSources/TiffDataSource.py", line 99, in getSlice
    return self.im[0].asarray(False, False)[0, 0, :,:,ind].squeeze()
IndexError: index 9 is out of bounds for axis 4 with size 3
Traceback (most recent call last):
  File "/Users/zachcm/Code/python-microscopy/PYME/DSView/scrolledImagePanel.py", line 58, in OnPaint
    self.renderer(MemDC);
  File "/Users/zachcm/Code/python-microscopy/PYME/DSView/arrayViewPanel.py", line 497, in DoPaint
    im = self.Render(fullImage)
  File "/Users/zachcm/Code/python-microscopy/PYME/DSView/arrayViewPanel.py", line 1107, in Render
    seg = self.do.ds[x0_:(x0_+sX_):step,y0_:(y0_+sY_):step,int(self.do.zp), chan].squeeze().T
  File "/Users/zachcm/Code/python-microscopy/PYME/IO/DataSources/BaseDataSource.py", line 137, in __getitem__
    r.append(np.concatenate([np.atleast_2d(self.getSlice(i)[keys[0], keys[1]])[:,:,None] for i in indices], 2))
  File "/Users/zachcm/Code/python-microscopy/PYME/IO/DataSources/BaseDataSource.py", line 137, in <listcomp>
    r.append(np.concatenate([np.atleast_2d(self.getSlice(i)[keys[0], keys[1]])[:,:,None] for i in indices], 2))
  File "/Users/zachcm/Code/python-microscopy/PYME/IO/DataSources/BufferedDataSource.py", line 52, in getSlice
    sl = self.dataSource.getSlice(ind)
  File "/Users/zachcm/Code/python-microscopy/PYME/IO/DataSources/TiffDataSource.py", line 99, in getSlice
    return self.im[0].asarray(False, False)[0, 0, :,:,ind].squeeze()
IndexError: index 12 is out of bounds for axis 4 with size 3
</pre>
</details>

David-Baddeley, 2020-07-01:
Hmm - that's super odd. The example file you sent loads perfectly on this end. 

This is what I get on the console ...
<pre>
Loading data
filename == Downloads/anim.tif
Downloads/anim.tif
(90, 645, 365, 3)
Detected RGB TIFF
[645, 365, 90, 3]
[645, 365, 90, 3]
0
</pre>

That said it's an RGB TIFF, which generally implies not data, (i.e. has been colourmapped and is not suitable for quantitative measurement) and as a consequence is super low priority for loading as you should really be going back to the person who provided it and asking them for the raw data instead - we should probably make the RGB warning a little bit more shouty and visible (maybe pop up a warning dialog).

If I had to guess what is going wrong on your end, I'd bet on a py3 compatibility issue with tifffile. We might need to look at the string handling and/or try and merge our modifications with the most recent version of tifffile.

David-Baddeley, 2020-07-01:
nope, it's not python3 - example dataset works fine on py3 for me ...

David-Baddeley, 2020-07-01:
what python and numpy versions?

David-Baddeley, 2020-07-01:
Actually, I misspoke - 1st frame loads fine, subsequent frames don't. I still maintain that loading RGB TIFF is usually an error.

David-Baddeley, 2020-07-01:
This is also the case on python=2.7 and numpy=1.14, so not a compatibility issue. Updated title to reflect status.
*NOTE* - first page loads fine.

David-Baddeley, 2020-07-01:
fixed in 2e1f3d6a8fa01986b70b500562e5db7ce8ee091e

closed 2020-07-01 by David-Baddeley