python-microscopy/python-microscopy #310: [ENHANCEMENT] plugin API changes - how to deal with for external packages

csoeller, 2020-07-07:
Recent changes related to the plugin API to work arount reference counting issues have brought into focus a question that I have. How do external packages/repos best deal with API changes? Note that this is not an issue for plugins internal to the core PYME dist as these will by definition in sync.

**Describe the solution you'd like**
I have no actual solution at this point. Any solution should report to an installing user as clearly as possible what the issue is, e.g. "these plugins require a PYME >= version XX that implements API version YYY" or similar. I am looking for suggestions how to best deal with the situation, however, also recognise that currently this may be an N=1 problem (PYME-extra). Is this as simple as requiring a particular version of PYME in the requirements? A runtime check could be ok (for those who update in place with a PYME-extra develop install) but that requires that any plugin API change bumps the PYME version.

**Describe alternatives you've considered**
No solution yet hence no alternatives to consider.

**Additional context**
Recent PYME commit 468002ec5ed2e321e9cb4d87e4502821cbec9124 has brought this to my attention. If I fix my plugins to the new architecture then somebody installing PYME-extra with an older PYME install will run into issues - I think.

**Are you prepared to work on it?**
Definitely will testdrive any proposed solution. For now just looking for ideas how to approach.

David-Baddeley, 2020-07-07:
Pretty exhaustive response here to have this on record ...

I don't think it's just you. There are a couple of other plugins that could be effected. the new changes should be backwards compatible, however, in that old plugins should still work (albeit with a warning).

If you are using features that are only present in a new version (e.g. the `Plugin` base class) in a plugin I think that compatibility checking should be left to the plugin (this matches how standard python packages, e.g. numpy, scipy, etc handles it). This can be handled by a simple check on `PYME.version`, or by appropriate version restrictions on python-microscopy in the modules conda recipe or pip info.  I'd suggest pinning to a git release tag rather than an intermediate version, but the date you last pulled before updating the API calls is probably good enough (especially when we don't really have formal releases). I don't think it's the end of the world for a plugin to say I need at least version x of PYME (and to be broken on earlier versions). I'm not particularly interested in versioning the API itself (rather than just relying on PYME versions) as I can't see this doing anything other than adding overhead. The argument for API versioning was if we were going to make massive backwards incompatible changes to the API and we wanted to maintain long term support for old plugins which could request to use the old API instead.  This sounds like a maintenance nightmare - it's better to either maintain backwards compatibility or just deprecate stuff (with reasonable notice) and expect people to adapt their code if they want it to run on newer versions of PYME. The exception to this is file formats which need long term backwards compatibility - we already version PZF, and we should version our metadata and HDF5 formats.

On the PYME end we should: 
- endeavour to maintain backwards compatibility on the API for a reasonable length of time so that old modules still work
- if planing on deprecating the old api raise a warning when it is used, with a timeframe to deprecation if known/imminent
- change the documentation to indicate at which version the new features became available.

There could be an argument for having more formal release notes in the future if we move to more structured releases, but to me the argument for structured releases and separate development and stable branches etc ... is currently poor and probably doesn't justify the significant overhead it would entail (particularly the amount of testing prior to each release - if we had dedicated stable and dev branches I'd want to thoroughly test everything before each stable release (conservatively 1-2 weeks worth of work for each release - particularly given how bad our unit tests are). As it stands, I'm not sure anything in the code can really be classed as "stable". I'd say we currently have 2 camps of users - those who are happy living on the bleeding edge, and those who update only when there is a new feature or bugfix they want. The current continuous alpha approach works quite well for this case.

barentine, 2020-07-07:
Chiming in on the release notes bit:
I'm happy on the bleeding edge, but not always overjoyed reading through commits and their messages to get a feel for what's going on and the rational behind - especially with the nonlinear nature of progress. Obviously I can/do diff, etc., but I wonder if we could improve our communication a bit (and double it as note-taking to look back to).

What would you think @David-Baddeley of submitting/merging a PR when you're doing something non-trivial? Doesn't have to change governance, but I think it'd be an easy way to keep (release-ish)notes/rational/tracebacks/associated issues, etc. all together with the net diff, which could help keep all of us on the same page at a low overhead. I've been trying to do the same with nep-fitting / warp-drive, though with their low traffic the motivation has been more for note keeping so far.

zacsimile, 2020-07-07:
I agree with @barentine. It would be awesome if this explanation @David-Baddeley just posted was linked with commit 468002e and other associated commits (4847f0dfbadb9d424895821f21eff34efe8f5467 and dcef9cfb0c9848040f665be92edfb92acb6ee74d?).

csoeller, 2020-07-07:
Hi all, thanks for feedback, sounds all sensible. Quick check, how do I check at runtime the pyme version/checkout date of the running pyme is after the date of a parcticular commit?

zacsimile, 2020-07-07:
Off the top of my head, `git log -1` will return the latest commit/date you have in your repository, but then you'll need to do a manual comparison. https://stackoverflow.com/questions/14222510/find-out-if-a-git-commit-was-checked-in-before-or-after-another-commit-given-the/29984371 seems to have a few more ideas.

David-Baddeley, 2020-07-07:
No need to check git tags etc... You can get the current version you are
running as PYME.version.version and there are a few python tools for
comparing versions. distutils.version is in the std. library and the
packaging module as a more up to date alternative should be pretty
ubiquitous as it's a dependency of setup tools.

See
https://stackoverflow.com/questions/11887762/how-do-i-compare-version-numbers-in-python

David-Baddeley, 2020-07-07:
Short of full release notes, I think some form of changelog might be worth looking at - it's a little more admin overhead than the status quo, but not a huge amount (the difficulty will be remembering to actually use it).

I've been meaning to adapt my workflow to use branches more, but I'm not sure that this one would have been captured  even with a branched workflow - it started out as a fairly minor bugfix.

David-Baddeley, 2020-07-07:
see https://github.com/python-microscopy/python-microscopy/blob/master/CHANGELOG.md

the challenge will now be keeping this up to date. To be useful the change version should be the date of incorporation into main, so this will probably require a separate PR on CHANGELOG after an initial PR, unless I update CHANGELOG for everything). We could potentially think about ways of automating this - potentially including a changelog update in the PR but using `FIXVERSION` (or some else suitably unique in the version field). This would be pretty easy for me to edit in the PR immediately prior to merging (and could potentially be automated in the future with a script which did string substitution).

csoeller, 2020-07-08:
Ok, with that version checking runs fine, something like this:

import PYME.version as PYMEv
print("latest PYME version tag - %s" % PYMEv.version)

from packaging import version as pv
if pv.parse(PYMEv.version) > pv.parse('20.7.1'):
    print("latest PYME version is post July 2020")
else:
    print("latest PYME version is pre July 2020")

David-Baddeley, 2020-07-08:
It's a bit safer to use 2 digits (a leading 0) for the month and day bits, although packaging.version should do the right thing.

David-Baddeley, 2020-07-08:
Can we close this now?

closed 2020-07-09 by csoeller