python-microscopy/python-microscopy #314: [BUG] taskWorkerHTTP clusterIO.get_file errors, possible windows/sockets/webframework issues?

barentine, 2020-07-08:
**Describe the bug**
The htsms microscope currently runs a little clunky/freezes up when we are running through queued series in the ActionManager with auto analysis. After several successfully localized series, the localization tasks start failing with `clusterIO.get_file` errors. There are plenty of windows socket errors on the instrument computer, coming it seems both from the offsetpiezo and from spooling.

Dataserver performance is additionally turf'd. If I open a series of 10k frames via `dh5view pyme-cluster://stuff` and try to slice the datasource through `t` it will choke and eventually fail on retries in `clusterIO.get_file`.

**To Reproduce**
Steps to reproduce the behavior:
0. pause action queue
1. queue up 10+ series, of 10k+ frames
2. check 'save analysis settings to metadata', with the GPU fit
3. resume the action queue
4. watch everything fall a part

-- alternative --
1. open a PYME cluster series in dh5view
2. try to run through slices either with fast playback or programmatically in the shell.
3. watch a bunch of clusterIO retry messages slowly appear before hitting one too many retries and giving up.

**Expected behavior**
smooth running, analysis keeping up, etc., ability to open a pcs datasource in dh5view and use things normally.

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: [e.g. Win10] Win10 on the instrument computer, ubuntu14.04 on the cluster
 - Python version: 3.6 on all
 - numpy version: 1.16 on all
 - wxpython version [if a UI bug]:
4.1.0 on insturment computer(I pip reinstalled recently and didn't catch this. Could be relevant due to wx call-afters in frameWrangler, etc.), n/a on cluster

**Additional context**
There've been some webframework reworks in the last couple months while the microscope has been down. The webframework is being used here by my offsetpiezo, which is hitting socket errors.

I haven't cleared the rule logs for quite some time; without wading through which 00000-hash is the most recent. The logs are like this one from a slightly earlier run
```
===========================================================================
Error in rule: 000008-83f22ef62f8c4fd78a015a4d1187d127 running task: 4501 on stromberg:3535

taskDescr:
----------
{'id': '000008-83f22ef62f8c4fd78a015a4d1187d127~4501', 'type': 'localization', 'taskdef': {'frameIndex': '4501', 'metadata': 'PYME-CLUSTER://villains/Bergamot/2020_7_3/<super-sectret>/000/analysis/3_7_series_00005.h5r.json'}, 'inputs': {'frames': 'pyme-cluster://villains/Bergamot/2020_7_3/<super-sectret>/000/3_7_series_00005.pcs'}, 'outputs': {'fitResults': 'http://10.150.7.3:15348/__aggregate_h5r/Bergamot/2020_7_3/<super-sectret>/000/analysis/3_7_series_00005.h5r/FitResults', 'driftResults': 'http://10.150.7.3:15348/__aggregate_h5r/Bergamot/2020_7_3/<super-sectret>/000/analysis/3_7_series_00005.h5r/DriftResults'}}

Traceback:
----------
Traceback (most recent call last):
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/PYME/cluster/taskWorkerHTTP.py", line 308, in computeLoop
    res = task()
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/PYME/localization/remFitBuf.py", line 427, in __call__
    bufferManager.bBuffer.calc_background(self.bgindices)
  File "/home/ubuntu/code/pyme-warp-drive/warpdrive/buffers.py", line 213, in calc_background
    self.update_buffer(bg_indices)
  File "/home/ubuntu/code/pyme-warp-drive/warpdrive/buffers.py", line 176, in update_buffer
    frame_data = np.ascontiguousarray(self.data_buffer.getSlice(frame), dtype=np.float32)
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/PYME/IO/buffers.py", line 28, in getSlice
    sl = self.dataSource.getSlice(ind)
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/PYME/IO/DataSources/ClusterPZFDataSource.py", line 71, in getSlice
    sl = PZFFormat.loads(clusterIO.get_file(frameName, self.clusterfilter))[0]
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/PYME/IO/clusterIO.py", line 788, in get_file
    raise IOError("Specified file could not be found: %s" % filename)
OSError: Specified file could not be found: Bergamot/2020_7_3/<super-sectret>/000/3_7_series_00005.pcs/frame04480.pzf
```

Typical dataserver logs are pretty boring, most interesting I've seen is:
```
========================================
PYMEDataServer, running on python 3.6.9 |Anaconda, Inc.| (default, Jul 30 2019, 19:07:31) 
[GCC 7.3.0]

Serving from directory: /data/PYMEData/
Serving HTTP on 10.150.7.4 port 15348 ...
10.150.7.131 - - [07/Jul/2020 18:48:39] code 405, message File already exists /data/PYMEData/Bergamot/2020_7_7/testing_with_old_sample/000/7_7_series_C.pcs/frame04058.pzf

[... 2751 lines cut ...]
    self.run()
  File "C:\Users\Bergamot\Anaconda2\envs\pyme\lib\threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\HTTPSpooler.py", line 197, in _queuePoll
    clusterIO.put_files(files, serverfilter=self.clusterFilter)
  File "c:\users\bergamot\code\python-microscopy\PYME\IO\clusterIO.py", line 1117, in put_files
    name, info = _chooseServer(serverfilter)
  File "c:\users\bergamot\code\python-microscopy\PYME\IO\clusterIO.py", line 871, in _chooseServer
    name, info = serv_candidates[np.argmin(costs)]
  File "<__array_function__ internals>", line 6, in argmin
  File "C:\Users\Bergamot\Anaconda2\envs\pyme\lib\site-packages\numpy\core\fromnumeric.py", line 1269, in argmin
    return _wrapfunc(a, 'argmin', axis=axis, out=out)
  File "C:\Users\Bergamot\Anaconda2\envs\pyme\lib\site-packages\numpy\core\fromnumeric.py", line 55, in _wrapfunc
    return _wrapit(obj, method, *args, **kwds)
  File "C:\Users\Bergamot\Anaconda2\envs\pyme\lib\site-packages\numpy\core\fromnumeric.py", line 44, in _wrapit
    result = getattr(asarray(obj), method)(*args, **kwds)
ValueError: attempt to get argmin of an empty sequence

ERROR:root:Exception whilst putting files
Traceback (most recent call last):
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\HTTPSpooler.py", line 197, in _queuePoll
    clusterIO.put_files(files, serverfilter=self.clusterFilter)
  File "c:\users\bergamot\code\python-microscopy\PYME\IO\clusterIO.py", line 1117, in put_files
    name, info = _chooseServer(serverfilter)
  File "c:\users\bergamot\code\py

barentine, 2020-07-08:
Conversation with David:
- try clearing the logs
- after that, update PYME

barentine, 2020-07-13:
Update:
- clearing the logs did not fix
- updating the cluster nodes _may_ have helped (not hitting clusterIO.get_file errors on a dummy run remotely) but dataserver performance appears to be poor (CPU usage is railed batching analysis of non-blinking data, while GPU usage is quite low, and tasks take a while to tick off)

I notice that while looking at clusterUI with analysis queued, in the column showing node CPU/GPU usage nodes will disappear / reappear, which may be indicative of what's bogging us down at the moment. On top of that, I've noticed ocassional errors the first couple tasks after restarting cluster processes related to getting the nameserver:

<details>
<summary> rule log </summary>

```
===========================================================================
Error in rule: 000000-4b03d94dd95f48fe9ce0a9213fb19ab1 running task: 2500 on lechiffre:4903

taskDescr:
----------
{'id': '000000-4b03d94dd95f48fe9ce0a9213fb19ab1~2500', 'type': 'localization', 'taskdef': {'frameIndex': '2500', 'metadata': 'PYME-CLUSTER://villains/Bergamot/2020_7_10/000/analysis/10_7_series_00001.h5r.json'}, 'inputs': {'frames': 'pyme-cluster://villains/Bergamot/2020_7_10/000/10_7_series_00001.pcs'}, 'outputs': {'fitResults': 'http://10.150.7.8:15348/__aggregate_h5r/Bergamot/2020_7_10/000/analysis/10_7_series_00001.h5r/FitResults', 'driftResults': 'http://10.150.7.8:15348/__aggregate_h5r/Bergamot/2020_7_10/000/analysis/10_7_series_00001.h5r/DriftResults'}}

Traceback:
----------
Traceback (most recent call last):
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/PYME/IO/clusterIO.py", line 284, in locate_file
    locs, t = _locateCache[cache_key]
KeyError: 'villains::Bergamot/2020_7_10/000/analysis/10_7_series_00001.h5r.json'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/PYME/misc/sqlite_ns.py", line 100, in getNS
    ns = nsd[(protocol, thread_id)]
KeyError: ('_pyme-http', 140270871553792)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/PYME/cluster/taskWorkerHTTP.py", line 307, in computeLoop
    task = remFitBuf.createFitTaskFromTaskDef(taskDescr)
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/PYME/localization/remFitBuf.py", line 292, in createFitTaskFromTaskDef
    mdh.update(json.loads(unifiedIO.read(md)))
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/PYME/IO/unifiedIO.py", line 226, in read
    s = clusterIO.get_file(sequenceName, clusterfilter)
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/PYME/IO/clusterIO.py", line 777, in get_file
    locs = locate_file(filename, serverfilter, return_first_hit=True)
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/PYME/IO/clusterIO.py", line 299, in locate_file
    services = get_ns().get_advertised_services()
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/PYME/misc/hybrid_ns.py", line 31, in get_advertised_services
    services = dict(self._sqlite_ns.get_advertised_services())
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/PYME/misc/hybrid_ns.py", line 16, in _sqlite_ns
    return sqlite_ns.getNS(self._protocol)
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/PYME/misc/sqlite_ns.py", line 102, in getNS
    ns = SQLiteNS(protocol)
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/PYME/misc/sqlite_ns.py", line 41, in __init__
    self._conn.execute("CREATE TABLE dns (name TEXT, address TEXT, port INTEGER, creation_time FLOAT, URI TEXT)")
sqlite3.OperationalError: table dns already exists

===========================================================================
Error in rule: 000000-4b03d94dd95f48fe9ce0a9213fb19ab1 running task: 100 on lechiffre:4891

taskDescr:
----------
{'id': '000000-4b03d94dd95f48fe9ce0a9213fb19ab1~100', 'type': 'localization', 'taskdef': {'frameIndex': '100', 'metadata': 'PYME-CLUSTER://villains/Bergamot/2020_7_10/000/analysis/10_7_series_00001.h5r.json'}, 'inputs': {'frames': 'pyme-cluster://villains/Bergamot/2020_7_10/000/10_7_series_00001.pcs'}, 'outputs': {'fitResults': 'http://10.150.7.8:15348/__aggregate_h5r/Bergamot/2020_7_10/000/analysis/10_7_series_00001.h5r/FitResults', 'driftResults': 'http://10.150.7.8:15348/__aggregate_h5r/Bergamot/2020_7_10/000/analysis/10_7_series_00001.h5r/DriftResults'}}

Traceback:
----------
Traceback (most recent call last):
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/PYME/IO/clusterIO.py", line 284, in locate_file
    locs, t = _locateCache[cache_key]
KeyError: 'villains::Bergamot/2020_7_10/000/analysis/10_7_series_00001.h5r.json'

During handling of the above exception, another exception occurred:

[... 12 lines cut ...]
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/PYME/IO/unifiedIO.py", line 226, in read
    s = clusterIO.get_file(sequenceName, clusterfilter)
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/PYME/IO/clusterIO.py", line 777, in get_file
    locs = locate_file(filename, serverfilter, return_first_hit=True)
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/PYME/IO/clusterIO.py", line 299, in locate_file
    services = get_ns().get_advertised_services()
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/PYME/misc/hybrid_ns.py", line 31, in get_advertised_services
[... cut ...] tween-node transfers will be substantially reduced.

2. When we get frames to calculate the background, we are looping over `ClusterPZFDataSource.getSlice` method, which serially calls `clusterIO.get_file`. If we added a `clusterIO.get_files` function which used a new endpoint on the dataservers we could return all of the files that node has within the slice in one go, which should reduce a fair amount of network overhead (not sure how this would play if someone was running cluster-duplication).

barentine, 2020-07-15:
After some profiling it looks like the issue could be due to performance of `PYME.misc.HybridNS` as opposed to `pyme_zeroconf.ZeroConfNS` (particularly the `get_advertised_services` methods), which we switched to on February 24.

This would also explain why the node status' flake out now

mProfile reports from opening a pcs in dh5view and pressing play for a bit more than 500 frames.
[PZFFormat.py.pdf](https://github.com/python-microscopy/python-microscopy/files/4922041/PZFFormat.py.pdf)
[clusterIO.py.pdf](https://github.com/python-microscopy/python-microscopy/files/4922042/clusterIO.py.pdf)
[ClusterPZFDataSource.py.pdf](https://github.com/python-microscopy/python-microscopy/files/4922043/ClusterPZFDataSource.py.pdf)

mProfile reports from a dataserver while doing some similar slicing on a pcs in dh5view
[clusterListing.py.pdf](https://github.com/python-microscopy/python-microscopy/files/4922063/clusterListing.py.pdf)
[HTTPDataServer.py.pdf](https://github.com/python-microscopy/python-microscopy/files/4922064/HTTPDataServer.py.pdf)

David-Baddeley, 2020-07-15:
Hmm - looking at those logs, it would not appear that there is an issue with `hybrid_ns`, although it's a bit hard to be sure. The profiling here is super hard to interpret because the first call to `get_ns()` is much slower than subsequent calls due to  all the sleeps associated with making zeroconf play nice.

Splitting the `get_ns()` and `get_advertised_services()` calls onto separate lines would help with debugging, but it appears that the times within `get_ns()` already add up to most of the time in `locate_file()` (L777). Most of this is spent in either in the `time.sleep()`s required for zeroconf which only occur on the first call, or waiting on the lock. The fact that so much time is spent waiting on the lock suggests a minor issue as implies that you are getting a bunch of hits on `get_ns()` that are concurrent with the initial call (this should really be done and dusted when loading clusterIO), but I don't think that it is important in any case (all of this will contribute to initial latency when loading a file, but should not have any effect on actually getting frame data - i.e. will not cause timeouts when connecting to the dataserver).

The line which is most interesting in debugging the problem on the client side is line 801 of clusterIO - ie the actual requests.get() call as this is indicative of server performance and is where timeout errors etc arise. This suggests that the average round-trip time for a request to the dataserver is 0.02 seconds, which is an order of magnitude faster than our timeout.

David-Baddeley, 2020-07-15:
Re: hybrid_ns related error in rule traceback 2 posts up: this points to a race condition in sqlite_ns. I'll add some locking and see if it fixes that.

**NOTE:** any ns performance issues won't cause timeout or connection errors. Worst case, they slow the client down a bit, but I don't see any evidence of that at this point.

barentine, 2020-07-15:
Hi @David-Baddeley - thanks loads for looking into this with me. I should have been clear in my original post about the `hybrid_ns` that while a good chunk of time is spent on the sleeps with the first `get_ns`, the `get_advertised_services` calls themselves are responsible for 13.98 - 8.89 = 5.09 s, or 6.65 ms per call, which is higher than I would have (naively) expected at 1/3 the `requests.get` time. Does that seem normal to you?

In-line with your observation about the timeout speeds, increasing the `get_file` timeout lead to equivalent mProfile reports. I agree that `request.get` profiles (including the pzf stuff and HTTPDataServer) indicate that the DataServers are fine.

Good point that my blaming the ns for timeout/connection errors was fairly presumptuous.

barentine, 2020-07-15:
oh, and thank you for adding the locking in `sqlite_ns` - it's a bit minor compared to the general 'why is this all slower now' question, but once we solve that it will be nice not to fail a couple frames from the first series each time.

barentine, 2020-07-16:
commented out the hybrid_ns call and uncommented the old pzc call in clusterIO. Running the same test gets us 3.8 ms per call to `get_advertised_services`. 

I'll add the config option to use pzc in clusterIO rather than hybrid, as discussed.

[clusterIO_pzcns.pdf](https://github.com/python-microscopy/python-microscopy/files/4934908/clusterIO_pzcns.pdf)

David-Baddeley, 2020-07-16:
Config option should already be there - did it yesterday while we were
talking.

barentine, 2020-07-16:
You sure did! Sorry I missed that and thank you! I'll close this.

closed 2020-07-16 by barentine